Ysurac/openmptcprouter #1698: Invalid signature

gnosoz, 2021-03-26:
apt update with no gpg errors

## Current Behavior
apt update shows the following signature error 
```
Err:4 https://repo.openmptcprouter.com stretch InRelease
  The following signatures were invalid: EXPKEYSIG 608FD17B2B24D936 OpenMPTCProuter <contact@openmptcprouter.com>
Reading package lists... Done
Building dependency tree       
Reading state information... Done
All packages are up to date.
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://repo.openmptcprouter.com stretch InRelease: The following signatures were invalid: EXPKEYSIG 608FD17B2B24D936 OpenMPTCProuter <contact@openmptcprouter.com>
W: Failed to fetch https://repo.openmptcprouter.com/dists/stretch/InRelease  The following signatures were invalid: EXPKEYSIG 608FD17B2B24D936 OpenMPTCProuter <contact@openmptcprouter.com>
W: Some index files failed to download. They have been ignored, or old ones used instead.
```

## Possible Solution
verify that keys didn't change

## Steps to Reproduce the Problem
apt update 
this was tested with same results on both Ubuntu 20 and Debian 10 with same signature failure result

## Specifications

  - OpenMPTCProuter VPS version: VPS 0.1022  
  - OpenMPTCProuter platform:

Ysurac, 2021-03-26:
The key changed but server script was updated to download new key.
Else you can do ```wget -O - http://repo.openmptcprouter.com/openmptcprouter.gpg.key | apt-key add -```

gnosoz, 2021-03-26:
thank you very much Yannick :)

closed 2021-03-26 by gnosoz