Ysurac/openmptcprouter #1699: Seems not add 3 connexions

jcpamart, 2021-03-26:
Hi, I have 3 connexions : 2 ADSL approx 6 Mbits & 1 wifimax approx 30 Mbits.
Now I try to download a debian iso, and the download stil at 350 Kb/s
The OMR is on a rasp B3+
But I can't find how to solve why the download is so slow......
How can I test the adding connexion ?

Thanks for your help

Ysurac, 2021-03-26:
Please use correct issue category and use template. I always need versions and arch.
To test connections speed, use ```omr-test-speed wan1``` via SSH or Services->Terminal (Ctrl+C to stop after at least 2 minutes)

jcpamart, 2021-03-26:
Thanks
Impossible to connect via putty. Connexion refused...
In terminal, I have nothing appear, so can't used it...
Sorry for the template and category, I'm new here.....

Thanks for your help....

Just precise, when I connect my PC in my wifimax, all download are more speed, and just with only this one connexion....

Ysurac, 2021-03-27:
Connection via SSH must work. Port 22 on the router, user root and no password by default.

jcpamart, 2021-03-27:
Hi, it's OK, this is the result....

root@OpenMPTCProuter:~# omr-test-speed wan1
Select best test server...
host: speedtest.frankfurt.linode.com - ping: 148
host: speedtest.tokyo2.linode.com - ping: 400
host: speedtest.singapore.linode.com - ping: 295
host: speedtest.newark.linode.com - ping: 246
host: speedtest.atlanta.linode.com - ping: 282
host: speedtest.dallas.linode.com - ping: 279
host: speedtest.fremont.linode.com - ping: 320
host: speedtest.tele2.net - ping: 192
host: www.ovh.net - ping:
host: speed.hetzner.de - ping: 177
host: ipv4.bouygues.testdebit.info - ping: 185
host: par.download.datapacket.com - ping: 84
host: nyc.download.datapacket.com - ping: 156
host: ams.download.datapacket.com - ping: 159
host: fra.download.datapacket.com - ping: 154
host: lon.download.datapacket.com - ping: 191
host: mad.download.datapacket.com - ping: 214
host: prg.download.datapacket.com - ping: 234
host: sto.download.datapacket.com - ping: 236
host: vie.download.datapacket.com - ping: 244
host: war.download.datapacket.com - ping: 247
host: atl.download.datapacket.com - ping: 317
host: chi.download.datapacket.com - ping: 299
host: lax.download.datapacket.com - ping: 335
host: mia.download.datapacket.com - ping: 324
host: nyc.download.datapacket.com - ping: 336
Best server is http://par.download.datapacket.com/10000mb.bin, running test:
ipset v7.6: Syntax error: cannot parse 185.93.2.9185.93.2.25185.93.2.16: resolving to IPv4 address failed
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:46 --:--:--     0
curl: (28) Failed to connect to par.download.datapacket.com port 80: Operation timed out

ipset v7.6: Syntax error: cannot parse 185.93.2.9185.93.2.25185.93.2.16: resolving to IPv4 address failed
root@OpenMPTCProuter:~# omr-test-speed wan2
Select best test server...
host: speedtest.frankfurt.linode.com - ping: 44
host: speedtest.tokyo2.linode.com - ping: 282
host: speedtest.singapore.linode.com - ping: 185
host: speedtest.newark.linode.com - ping: 109
host: speedtest.atlanta.linode.com - ping: 141
host: speedtest.dallas.linode.com - ping: 150
host: speedtest.fremont.linode.com - ping: 177
host: speedtest.tele2.net - ping: 43
host: www.ovh.net - ping:
host: speed.hetzner.de - ping: 45
host: ipv4.bouygues.testdebit.info - ping: 32
host: par.download.datapacket.com - ping: 32
host: nyc.download.datapacket.com - ping: 122
host: ams.download.datapacket.com - ping: 40
host: fra.download.datapacket.com - ping: 48
host: lon.download.datapacket.com - ping: 45
host: mad.download.datapacket.com - ping: 49
host: prg.download.datapacket.com - ping: 56
host: sto.download.datapacket.com - ping: 63
host: vie.download.datapacket.com - ping: 53
host: war.download.datapacket.com - ping: 60
host: atl.download.datapacket.com - ping: 128
host: chi.download.datapacket.com - ping: 123
host: lax.download.datapacket.com - ping: 178
host: mia.download.datapacket.com - ping: 142
host: nyc.download.datapacket.com - ping: 105
Best server is http://ipv4.bouygues.testdebit.info/10G.iso, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 9536M    0 25.5M    0     0   349k      0  7:45:13  0:01:14  7:43:59  332k^C
root@OpenMPTCProuter:~# omr-test-speed wan3
Select best test server...
host: speedtest.frankfurt.linode.com - ping: 37
host: speedtest.tokyo2.linode.com - ping: 267
host: speedtest.singapore.linode.com - ping: 188
host: speedtest.newark.linode.com - ping: 107
host: speedtest.atlanta.linode.com - ping: 133
host: speedtest.dallas.linode.com - ping: 164
host: speedtest.fremont.linode.com - ping: 167
host: speedtest.tele2.net - ping: 38
host: www.ovh.net - ping:
host: speed.hetzner.de - ping: 39
host: ipv4.bouygues.testdebit.info - ping: 28
host: par.download.datapacket.com - ping: 26
host: nyc.download.datapacket.com - ping: 133
host: ams.download.datapacket.com - ping: 38
host: fra.download.datapacket.com - ping: 35
host: lon.download.datapacket.com - ping: 31
host: mad.download.datapacket.com - ping: 42
host: prg.download.datapacket.com - ping: 42
host: sto.download.datapacket.com - ping: 61
host: vie.download.datapacket.com - ping: 51
host: war.download.datapacket.com - ping: 92
host: atl.download.datapacket.com - ping: 120
host: chi.download.datapacket.com - ping: 132
host: lax.download.datapacket.com - ping: 169
host: mia.download.datapacket.com - ping: 136
host: nyc.download.datapacket.com - ping: 115
Best server is http://par.download.datapacket.com/10000mb.bin, running test:
ipset v7.6: Syntax error: cannot parse 185.93.2.16185.93.2.9185.93.2.25: resolving to IPv4 address failed
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:15 --:--:--     0^C
  0     0    0     0    0     0      0      0 --:--:--  0:00:18 --:--:--     0^C
ipset v7.6: Syntax error: cannot parse 185.93.2.16185.93.2.9185.93.2.25: resolving to IPv4 address failed
root@OpenMPTCProuter:~#

Seems a little bit bad that no ?

Ysurac, 2021-03-27:
There is an issue with current script that appear on site server.
Do that: ```wget -O /bin/omr-test-speed https://github.com/Ysurac/openmptcprouter-feeds/raw/develop/openmptcpr
outer/files/bin/omr-test-speed``` (this fix the script) and try again.

jcpamart, 2021-03-27:
Hi, the problem seems to be fixed ! Thanks you very much :)

jcpamart, 2021-03-27:
Do you know if any routeur exist in hardware to configure an OMR on it ?
Thanks !

jcpamart, 2021-03-27:
But now I have that always remaining...

jcpamart, 2021-03-27:
[image]

jcpamart, 2021-03-27:
nop !!!!
After restart, download debian stay at the adsl connexion, and don't add the second in wifimax.....
[image]

Ysurac, 2021-03-27:
I still don't have omr-test-speed for each wan result ;)

jcpamart, 2021-03-28:
Sorry, I forgot it !

root@OpenMPTCProuter:~# omr-test-speed wan1
Select best test server...
host: speedtest.frankfurt.linode.com - ping: 37
host: speedtest.tokyo2.linode.com - ping: 254
host: speedtest.singapore.linode.com - ping: 189
host: speedtest.newark.linode.com - ping: 107
host: speedtest.atlanta.linode.com - ping: 133
host: speedtest.dallas.linode.com - ping: 159
host: speedtest.fremont.linode.com - ping: 165
host: speedtest.tele2.net - ping: 39
host: www.ovh.net - ping:
host: speed.hetzner.de - ping:
host: ipv4.bouygues.testdebit.info - ping: 27
host: par.download.datapacket.com - ping: 31
host: nyc.download.datapacket.com - ping: 115
host: ams.download.datapacket.com - ping: 38
host: fra.download.datapacket.com - ping: 36
host: lon.download.datapacket.com - ping: 32
host: mad.download.datapacket.com - ping: 43
host: prg.download.datapacket.com - ping: 45
host: sto.download.datapacket.com - ping: 55
host: vie.download.datapacket.com - ping: 48
host: war.download.datapacket.com - ping: 56
host: atl.download.datapacket.com - ping: 121
host: chi.download.datapacket.com - ping: 132
host: lax.download.datapacket.com - ping: 164
host: mia.download.datapacket.com - ping: 137
host: nyc.download.datapacket.com - ping: 114
Best server is http://ipv4.bouygues.testdebit.info/10G.iso, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  3 9536M    3  376M    0     0  5167k      0  0:31:29  0:01:14  0:30:15 4714k^C
root@OpenMPTCProuter:~# omr-test-speed wan2
Select best test server...
host: speedtest.frankfurt.linode.com - ping: 45
host: speedtest.tokyo2.linode.com - ping: 280
host: speedtest.singapore.linode.com - ping: 188
host: speedtest.newark.linode.com - ping: 107
host: speedtest.atlanta.linode.com - ping: 131
host: speedtest.dallas.linode.com - ping: 147
host: speedtest.fremont.linode.com - ping: 178
host: speedtest.tele2.net - ping: 46
host: www.ovh.net - ping:
host: speed.hetzner.de - ping: 47
host: ipv4.bouygues.testdebit.info - ping: 36
host: par.download.datapacket.com - ping: 36
host: nyc.download.datapacket.com - ping: 107
host: ams.download.datapacket.com - ping: 44
host: fra.download.datapacket.com - ping: 51
host: lon.download.datapacket.com - ping: 49
host: mad.download.datapacket.com - ping: 52
host: prg.download.datapacket.com - ping: 59
host: sto.download.datapacket.com - ping: 67
host: vie.download.datapacket.com - ping: 56
host: war.download.datapacket.com - ping: 64
host: atl.download.datapacket.com - ping: 131
host: chi.download.datapacket.com - ping: 125
host: lax.download.datapacket.com - ping: 176
host: mia.download.datapacket.com - ping: 140
host: nyc.download.datapacket.com - ping: 107
Best server is http://ipv4.bouygues.testdebit.info/10G.iso, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 9536M    0 8373k    0     0   376k      0  7:12:32  0:00:22  7:12:10  370k^C

Seems working like it !

jcpamart, 2021-03-28:
Speak French ?

closed 2021-07-01 by github-actions[bot]