Ysurac/openmptcprouter #1702: VPN is not running, and Can't Contact Server Admin Script

NoShitSherlockium, 2021-03-28:
## Expected Behavior

VPN should be working, and it should be able to contac
## Current Behavior

It refuses to start any of the VPNs (I checked the keys), and says it cannot contact the server admin script, in the log it shows Connection refused on all of the vpns
## Specifications

  - OpenMPTCProuter version: 0.57.3 
  - OpenMPTCProuter VPS version: 0.1024 
  - OpenMPTCProuter platform: x86_64

Ysurac, 2021-03-29:
Did you reboot VPS after install ?
What is the reason indicated in log for the server admin script contact problem ?

blade5502, 2021-03-29:
Same issue here. Running latest server on Debian 10 and latest x86_64 client.

Logread shows omrvpn as connected
1daemon.notice netifd: Network device 'tun0' link is up
daemon.notice netifd: Interface 'omrvpn' has link connectivity
daemon.notice netifd: Interface 'omrvpn' is setting up now
daemon.notice netifd: Interface 'omrvpn' is now up

Bonding seems to work also, but webinterface states VPN not connected
[image]

closed 2021-07-02 by github-actions[bot]