Ysurac/openmptcprouter #1705: Readonly filesystem after violent reboots

PierreGranger, 2021-03-29:
## Expected Behavior
I need to change parameters on my config (old VPS was in OVH/SBG2, RIP) but filesystem is readonly

## Current Behavior
After multiple reboots, in /admin/system/openmptcprouter/status i have the message :
"Système de fichier en lecture seule" (Filesystem is readonly i guess).
I've tried to reboot property in Sytem > Reboot but it does not make a file system check on boot.
As seen here it looks like there should be an option to force file system ckeck : 
https://github.com/Ysurac/openmptcprouter/issues/1541#issuecomment-770303931

But i can't find it, how can i force a fsck ? (i guess i can do it on ssh but i have no idea of the user/password i need to connect to the raspberry)

Thanks a lot for your help !

## Specifications

  - OpenMPTCProuter version: v0.57.3 r0+15225-bfc433efd4 / LuCI Master git-20.339.75073-e54708a
  - OpenMPTCProuter VPS version: debian10-x86_64
  - OpenMPTCProuter platform: Raspberry Pi 4 Model B Rev 1.2

Ysurac, 2021-03-29:
As it's a RPI, easier to save config, write again image on a SD card and restore backup.

PierreGranger, 2021-03-29:
Thanks for your help, that was one working solution ;)

closed 2021-03-29 by PierreGranger