Ysurac/openmptcprouter #1706: Need apk for smartphone

ssdeol84, 2021-03-31:
Hi sir thanks for a nice project. I've successfully installed a server which is not giving me access of mptcp with Shadowsocks on smartphone by using Shadowsocks app.

I'm successfully connected with server from client. But that doesn't aggregate my wifi and 4G network. Both my connection are enabled and also I've selected enabled mobile data always active in developers option in settings. But still my 4G data doesn't comes up while wifi is connected.

I request you to please make smartphone apk for android 11. Which can help to connect to mptcp server by both connections.

I've check one github in which there is a app multipath control apk. But that is very old and that is not working on android 11.

Please help me to fix it.

Ysurac, 2021-03-31:
I don't understand what you mean with your smartphone. OpenMPTCProuter is an OS and it can't work on smartphone directly.

ssdeol84, 2021-03-31:
Mptcp is already installed on a server and I'm connected to that same server with Shadowsocks client app in smartphone. But Shadowsocks is not having any feature to make both the wifi and 4G connection active. 

I need a patch or a app for smartphone that can get aggregate speed of both wifi and 4G

Ysurac, 2021-03-31:
This doesn't work that way. OpenMPTCProuter have a server part and a router part.
Linux kernel must be patched to support MPTCP, I don't know if there is any Android phone that have Multipath TCP patch, for iPhone some have multipath tcp support but I don't know if this can work that way.

ssdeol84, 2021-04-10:
Ok fine I understand bonding will not work on smartphone without patching the kernel.

But I use to connect my smartphone with Shadowsocks vpn server by using Shadowsocks client app.

I've put all the settings on client app but it's not connecting. Can you help me. I guess there is not problem with config file at server end.
This is my app snapshot

[image]

andykitcool, 2021-04-16:
I think ssdeol84 means that an app is needed to connect with openmptcprouter and aggregate 4G and WiFi signals of mobile phones. I found a similar application--> www.speedify.com , but the disadvantage is that they can only use their servers and need pay

ssdeol84, 2021-04-16:
Yes but Shadowsocks client app does the same but only need to implement that same way

Ysurac, 2021-04-16:
Shadowsocks doesn't do aggregation alone (without MPTCP).
For authentication you need to be able to enter a key, not a password. You app doesn't seems to support Shadowsocks key.

ssdeol84, 2021-04-16:
Can we make changes in script by which we can also give authorisation through password also? In shadowsocks

There are several features in shadowsocks client app by which we can call sever user details by QR code, file and manual.

[image]

closed 2021-07-20 by github-actions[bot]