Ysurac/openmptcprouter #1707: What is the difference between OpenMPTCProuter and Loud Balancing router like TL-R470T+

Gaou3, 2021-03-31:
Hello,
We need to aggregate multi connexions to have a big bandwith. We set up OpenMPTCProuter. 
But someone told me about "loud balancing router", and routers like TL-R470+. I saw on the internet that the aggregation method used by OpenMPTCProuter was "true aggregation", without understanding how "loud balancing" was "false" aggregation. What is the difference between both ?

Ysurac, 2021-03-31:
load balancing only balance connection on each WAN, so at output you see alternately each WAN public IP. Also you can only reach max speed of the fastest WAN on one connection (for example when you download a file in browser).
With OpenMPTCProuter, when VPS is up, you aggregate speed, so you have max speed of all WANs combined. You also have always same public IP (the VPS ip).
OpenMPTCProuter support load balancing when no VPS is set (or when VPS is down).

Gaou3, 2021-03-31:
Ok, thank's for your explanations.
In the case of load balancing router, if I have to upload a big file (10 Go for example), is it possible the connection balance from a WAN to an other WAN during the upload ? And change the IP ? And stop (few seconds) the connection during the balance ?
Because uploading big files is the major part of my use ... An other part is to make live stream videos.

You told me OpenMPTCProuter support load balancing. Is it possible to priorise a WAN ? Or better, assign a maximum number of data for each WAN (and is it possible to do the same with the VPS aggregation) ?

Ysurac, 2021-03-31:
Not it will not balance during an upload or change IP.
It's not possible in OpenMPTCProuter to customize WAN priority but WAN defined as master is prioritized by default.
It's possible to define a quota or to limit max used bandwidth on each WAN in all case.
OpenMPTCProuter also support in aggregation mode a transparent failover (if a WAN is down you should not notice it).

Gaou3, 2021-03-31:
Nice ! Could you told me the process to define a quota on each WAN please ?

Ysurac, 2021-03-31:
A monthly quota can be set in omr-quota via interface.

Gaou3, 2021-03-31:
Nice ! Thank you very much for your explanations.

closed 2021-03-31 by Gaou3