Ysurac/openmptcprouter #1708: Est-ce possible de prioriser des équipements ?

Merrick28, 2021-04-01:
Bonjour, 
en ces temps de télétravail, mes connexions à la maison souffrent quelquefois des ados qui consomment beaucoup de BP (netflix, youtube, etc...)
Est-ce possible de prioriser certains clients (typiquement mon ordi du boulot, idem pour ma femme) pour leur assurer un minimum de BP et que le reste des équipements soient dépriorisés ?

Merci d'avance.

Ysurac, 2021-04-01:
Il est possible de prioriser/déprioriser certains sites/ports/IP via Réseau->OMR-DSCP. Donc soit l'IP de vos ordinateurs, soit changer les priorités de certains sites gourmand dans l'onglet "DSCP domaines".
Pour un fonctionnement optimal il faut que SQM soit activé dans le wizard.

Merrick28, 2021-04-01:
Merci.

J'ai activé le SQM pour ma ligne vdsl2+ (pas pour la 4g, il est marqué que ce n'est pas conseillé pour le LTE), et j'ai passé les ordis concernés en CS2 - high priority, , on verra si ça suffit.

closed 2021-07-06 by github-actions[bot]