Ysurac/openmptcprouter #1709: A quoi sert le vpn dans openmptcprouter ?

kobbybest, 2021-04-01:
## Expected Behavior

Merci pour ce super logiciel et le travail effectué mais j'ai un petit soucis,
j'aimerais que mon ping soit stable

## Current Behavior

J'ai un soucis de ping avec openmptcprouter, en effet mon vps a un ping de 2ms par exemple vers www.google .fr,
J'ai 4 connections agrégées, (adsl + 3x 4g) , le débit est bien la avec des pointes a 250 mbits/s.
Lorsque je ping chaque connexion vers google pendant 2 min, tout est stable, ca ne monte jamais au dessus de 65 ms sur chaque connections
Par contre un fois agrégée, le ping est très bon (je pense qu'il prend en prio le maitre qui est sur l'adsl), j'ai environ 30ms mais de façon répétitives sur un test de 2 min cela monte à plus de 600 voir 1000 ms (j'ai ce résultat sur la console d'openmptcprouter et sur mon pc)
Après plusieurs tests, j'ai décidé de changer de VPN, j'ai donc commencé par couper le vpn glorytun TCP et à ma grande surprise, internet était toujours la et toujours agrégée, du coup je me pose la question : A quoi sert le VPN dans openmptcprouter.
De plus je n'ai plus le problème de ping qui monte de façon répété.
Est ce que je peux rester sans VPN ? ou bien c'est risqué de ne pas l'employer et si je doit le remettre, pourquoi le ping bug sous vpn ?

Cordialement

## Possible Solution

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: v0.57.3
  - OpenMPTCProuter VPS version: v0.1024
  - OpenMPTCProuter platform: x86_64 NUC intel

Ysurac, 2021-04-01:
Le VPN sert pour tout ce qui n'est pas TCP, donc notamment UDP et ICMP. Il n'est pas recommandé de l'arrêter.
Il y a plusieurs VPN qui peuvent mieux fonctionner selon le besoin. Et SQM peut aussi aider.
Après il y a finalement peu d'intérêt de chercher à avoir absolument un ping ICMP stable.

kobbybest, 2021-04-01:
Merci pour cette réponse, le problème est que pour jouer, avoir des hausses de ping à 600 ms est très compliqué, et je ne comprends pas pourquoi l'emploi du vpn provoque ces hausses, je vais essayer avec un autre VPN.

kobbybest, 2021-04-01:
J'ai essayé d'autres vpn mais je n'arrive pas a les faire fonctionner hormis glorytun UDP ou j'ai le même soucis que glorytun TCP. J'ai essayé d'activer SQM mais cela ne change rien du tout

Cordialement.

Kalimeiro, 2021-04-02:
Bonjour, il faudrait plus de détails sur vos connexions internet utilisé, le débit que vous pouvez avoir hors utilisation de OMR, le chiffrement utilisé depuis votre NUC (perso j'utilise des RPI)

closed 2021-06-24 by kobbybest