Ysurac/openmptcprouter #1711: I want 2 LAN with internet but without discussion between 2 networks

Jujuhwd, 2021-04-03:
455 / 5000
Résultats de traduction
## Expected behaviour
I want to have 2 different networks (LAN).
- 192.168.X.X
- 172.0.X.X

The 2 networks must be able to access the Internet but cannot discuss together.

## Current behavior
Currently, 2 interfaces for the 2 LANs. I have tried with the same LAN firewall and separated from it, but both networks can chat.

Both networks access the internet correctly.

## Characteristics

    openmptcprouter-v0.57.3-r0 +

Thank you

Kalimeiro, 2021-04-03:
use vlan.

Jujuhwd, 2021-04-03:
Surely there is another solution?

Routing rules?

Ysurac, 2021-04-03:
By creating a new firewall zone.

Jujuhwd, 2021-04-03:
That's what I did. I configured them identically.

Kalimeiro, 2021-04-04:
in same physical port in same switch you need vlan taggued and firewall zone.

from 2 separated ports on the same switch, you need firewall zone and vlan untaggued.

from 2 separated switch, you just need firewall zone.

Jujuhwd, 2021-04-04:
How to do this?

Kalimeiro, 2021-04-04:
in OMR choose an custom interface, exemple if your LAN1 is on ETH0 and you want VLAN 100 on ETH0 create a custom interface like eth0.100

Then, you need also a manageable switch to put VLAN TAG or UNTAG on ports.

Jujuhwd, 2021-04-04:
But soon I will transfer openmptcprouter to a vm, I would have 2 ports for this vm, one of which will be on a virtual interface on proxmox with other vm on it.

So my 2 networks will be physically separated. So he won't be able to discuss together? If he can't, that's fine with me.

Otherwise what to do without going through vlan?

Out of curiosity to create the vlan interface on openmptcp, you just have to choose the static address protocol and the concerned physical port, it's just the label by putting vlan which indicates that it is a vlan?

Thank you for your explanations.

closed 2021-07-08 by github-actions[bot]