Ysurac/openmptcprouter #1713: Configure vlans

jeandelamarquise, 2021-04-04:
## Expected Behavior
Hi,
I'm trying to configure vlans for iot projects on openmptcprouter but I have difficulties, can someone help me to configure one vlans so I can do it alone after ?

## Current Behavior
I think I found a way to have ip address different in network/interfaces. I create a new one that I affects for smart tv and iot (i called it "smartiot" like "lan", "omrvpn")...
I put static address in protocols. 192.168.40.1, network mask 255.255.255.0 gateway 192.168.10.1 (the original address of openMPTCProuter, maybe i made a mistake there).
I named the physical parameter eth0.40
I assigned a firewall zone : "smartios" (i would like to have help to configure that, i would like to know how i can communicate from zone "lan" to zone "smartiot" and from "smartiot" to "internet" (after I will do another vlan with no communication to the internet for iot that i don't trust to send on internet things).
I just start a dhcp server on this interface (i didn't do it before)

## Specifications

OpenMPTCProuter version 0.57.1

Version 0.1023 5.4.81-mptcp
Raspberry Pi 4 Model B Rev 1.2

Thanks for your help.

Ysurac, 2021-04-04:
Yes, you don't need a gateway.
For firewall, in Network->Firewall you can edit the zone and allow what you want between zones.

jeandelamarquise, 2021-04-04:
For now, maybe it's because I put that gateway but I can't ping from zone "lan" to that zone "smartiot".
I need to set firewall rules ?
Thanks Ysurac for all you do with that OpenMPTCProuter :)

Ysurac, 2021-04-04:
Yes, you can copy what is done for lan zone, where you can ping to wan or vpn zone.

jeandelamarquise, 2021-04-04:
Thanks I'll try that and come back after

closed 2021-07-09 by github-actions[bot]