Ysurac/openmptcprouter #1714: USB E3373 LTE USB Stick not found after restart

rsajadi63, 2021-04-04:


Ysurac, 2021-04-04:
You should check State->Kernel log

rsajadi63, 2021-04-04:
> You should check State->Kernel log

After reboot?

rsajadi63, 2021-04-04:
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.83 (runner@fv-az76-868) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r0+15225-bfc433efd4)) #0 SMP Tue Dec 15 19:35:00 2020
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000037400000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 242688
[    0.000000]   DMA zone: 2304 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 46080 pages, LIFO batch:15
[    0.000000] percpu: Embedded 15 pages/cpu s30860 r8192 d22388 u61440
[    0.000000] pcpu-alloc: s30860 r8192 d22388 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240384
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=squashfs,ext4 rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x2aff9000-0x2eff9000] (64MB)
[    0.000000] Memory: 813836K/970752K available (6416K kernel code, 232K rwdata, 1688K rodata, 2048K init, 400K bss, 91380K reserved, 65536K cma-reserved, 118784K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x2fc/0x430 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000006] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000020] Switching to timer-based delay loop, resolution 52ns
[    0.000184] Console: colour dummy device 80x30
[    0.000614] printk: console [tty1] enabled
[    0.000662] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000697] pid_max: default: 32768 minimum: 301
[    0.000892] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.000927] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.001800] Disabling memory control group subsystem
[    0.001842] CPU: Testing write buffer coherency: ok
[    0.002205] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.002681] Setting up static identity map for 0x200000 - 0x200060
[    0.002790] rcu: Hierarchical SRCU implementation.
[    0.003143] smp: Bringing up secondary CPUs ...
[    0.003807] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.004640] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.005356] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.005502] smp: Brought up 1 node, 4 CPUs
[    0.005566] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.005584] CPU: All CPU(s) started in HYP mode.
[    0.005599] CPU: Virtualization extensions available.
[    0.006194] devtmpfs: initialized
[    0.016556] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.016809] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.016851] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.019370] pinctrl core: initialized pinctrl subsystem
[    0.019757] reg-dummy reg-dummy: no of_node; not parsing pinctrl DT
[    0.020515] NET: Registered protocol family 16
[    0.023022] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.026925] cpuidle: using governor ladder
[    0.026965] cpuidle: using governor menu
[    0.027541] Serial: AMBA PL011 UART driver
[    0.028610] bcm2835-mbox 3f00b880.mailbox: mailbox enabled
[    0.050037] raspberrypi-firmware soc:firmware: Attached to firmware from 2020-04-15 11:43, variant start
[    0.060044] raspberrypi-firmware soc:firmware: Firmware hash is 82f9bb929ce2186eb1824178c1ae82902ad6275c
[    0.095654] bcm2835-dma 3f007000.dma: DMA legacy API manager, dmachans=0x1
[    0.096704] reg-fixed-voltage fixedregulator_3v3: GPIO lookup for consumer (null)
[    0.096714] reg-fixed-voltage fixedregulator_3v3: using device tree for GPIO lookup
[    0.096734] of_get_named_gpiod_flags: can't parse 'gpios' property of node '/fixedregulator_3v3[0]'
[... 777 lines cut ...]
[   14.504734] sd 0:0:0:0: no of_node; not parsing pinctrl DT
[... cut ...]
[  636.452515] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
[  639.451330] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
[  695.680179] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
[  698.657865] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
[  731.494753] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
[  734.486561] ieee80211 phy0

Ysurac, 2021-04-04:
I can see in eth1 and eth2, two devices with same MAC address. You know what they are ?

rsajadi63, 2021-04-04:
Yes huawei usb 4g dongle have same MAC address but I seen other products that use this dongle with openmptcprouter on raspberry pi 3 and work very well!!!!!

rsajadi63, 2021-04-04:
With same two USB modem that have same Mac address openmptcprouter work well until reboot not occurred

rsajadi63, 2021-04-04:
In the __>Status ->Routes 
[image]

rsajadi63, 2021-04-04:
[image]

rsajadi63, 2021-04-04:
[image]
[image]
[image]
# #

rsajadi63, 2021-04-04:
In the interface there are two wan in the System->openmptcprouter two wan connect but in the Status->Routes only there is one wan and eth1 not detected

Ysurac, 2021-04-04:
One interface need to be set as master. Check if it's the case.
Check also that macvlan is not enabled if you use a physical interface like eth1 in Network->Interface.
Also I can see "irancell", it's working in Iran without issues ? without the need to use some other DNS like in China ? Good to know :)

rsajadi63, 2021-04-05:
I set wan1 as master and set model for two wan normal all things ok and work very well but when I reboot one of USB modem not found

Ysurac, 2021-04-05:
Can you check system log, in State->System log just after reboot ?

rsajadi63, 2021-04-06:
Tanks for your help my problem solved for this problem I change 3 option 
System->OpenMPTCProuter ->Advanced Settings tab:
1- Master interface selection=On wizard change 
2-Minimum scaling CPU=600000
3-Scaling governor =ondemand

Ysurac, 2021-04-06:
Strange, but in this case seems to be related to too much power usage at boot, RPI3 have know issues about that.