Ysurac/openmptcprouter #1715: Question: How to make http request over specific interface

ByteArrayException, 2021-04-04:
Hi!

How can i make an HTTP request via the console on a certain interface (so that i only use the certain uplink and not the aggregated eth0)?

Many thanks in advance

Ysurac, 2021-04-04:
via the console ? by using curl --interface xxx http://xxx

ByteArrayException, 2021-04-04:
Hi!
Thanks for you help. But the return of the command is the following:
curl: (28) Failed to connect to icanhazip.com port 80: Operation timed out

(i tested this for ex. with icanhazip.com but an failure occures, maybe dns not working?)

Thanks in advance

ByteArrayException, 2021-04-06:
Hi!
I have now tried this with the interface IP instead of the interface name. Now this error no longer occurs, but it always returns the VPS IP. Bypass only works to a certain extent because the interface always changes (due to different registration times). Is there an easier way to send this request only via the interface and to serve only this one interface instead of sending it via the VPS?

Many greetings

Ysurac, 2021-04-06:
You can check what I do here: https://github.com/Ysurac/openmptcprouter-feeds/raw/develop/openmptcprouter/files/bin/omr-test-speed

ByteArrayException, 2021-04-06:
Thanks for the reply! I will check this today or tomorrow.

Kind regards

ByteArrayException, 2021-04-06:
This works fine! Many thanks! :)

closed 2021-04-06 by ByteArrayException