Ysurac/openmptcprouter #1716: OMR Bypass broken

sebaro11, 2021-04-04:
## Expected Behavior

Have some bypass rules setup some time ago, because the VPS-IP range is blocked on some sites.
Some domains and clients have direct access to WAN1 and WAN2.

## Current Behavior

The rules did not work anymore since today(?) When I try to browse the domains I just get "ERR_CONNECTION_RESET" from Chrome/Edge. When deactivating the rules the connection is possible, but I get these IP block messages again.
I set up icanhazip.com and ifconfig.co to different wan ports. If I curl those I get the right IP of the choosen wan port.
Since curling the domains and getting the right IPs seems to work, I wonder what the problem with the other domains is.

Even the rules for the clients are not working correct, all traffic is still routed through the VPS-IP instead of direct WAN1/2.

## Specifications

  - OpenMPTCProuter version: v0.57.3 r0+15225-bfc433efd4
  - OpenMPTCProuter VPS version: 5.4.81-mptcp 0.1024
  - OpenMPTCProuter platform: Raspberry Pi 4 Model B Rev 1.2 64 Bit

Kalimeiro, 2021-04-04:
and if you restart you'r rasp, the problem persist ?

sebaro11, 2021-04-04:
> and if you restart you'r rasp, the problem persist ?

Yes, actually I run the 0.57.2 for over 100 days straight with no problem. I thought that might be fixed with the newest firmware on both sides. But nothing changed at all.
Though I restarted everything before the upgrade with no effect.

Kalimeiro, 2021-04-04:
DNS IP of your client like Windows is the IP of your OMR, correct ?

sebaro11, 2021-04-04:
Yes, DNS is the OMR IP. DHCP is active and working.

sebaro11, 2021-04-05:
Fixed it. Problem was one DSCP rule for a WireGuard VPN which I wanted to be set as "latency sensitive". I did not specified the protocol as UDP, instead I used "all" which seems to broke the bypass rules.

before
[image]

after
[image]

Now every rule works fine again. Jesus was that a luck to switch that rule to udp xD

closed 2021-04-05 by sebaro11