Ysurac/openmptcprouter #1717: Port forwarding for shadowsocks

rsajadi63, 2021-04-05:
I have two openmptcprouter one I made it myself another buy from a company I set mine  like the one I bought from the company .the company use port forwardin and when I disable port forwarding check traffic throughout only one wan but when I enable port forwarding traffic throughout two wans so I configure mine similarly that but traffic throughout any wan when I disable port forward traffic throughout only one wan what should I do?i set two interfaces and shadowsocks lib manually

rsajadi63, 2021-04-05:
[image]

rsajadi63, 2021-04-05:
This is my project

Ysurac, 2021-04-05:
I don't see the image.

rsajadi63, 2021-04-05:
[image]

rsajadi63, 2021-04-05:
This is my project raspberry pi 4 act as sender for video live stream to server with public ip 77.36.23x.xx that on it openmptcprouter script run i connect Rpi4 to Rpi 3 act as router (openmptcprouter )when I send stream to port 1935 server traffic throughout only via one wan i seen on router that I buy used port forward for port 1935 when I disable check box port forward traffic throughout only one wan I set port forward on mine but stream stops!!!

rsajadi63, 2021-04-05:
[image]
[image]

rsajadi63, 2021-04-05:
[image]

rsajadi63, 2021-04-05:
[image]

rsajadi63, 2021-04-05:
This picture is mine

rsajadi63, 2021-04-05:
What should I do?

Ysurac, 2021-04-06:
I don't really understand what you try to do with the port forward.

rsajadi63, 2021-04-06:
I want send data by my client that connected by Ethernet port  to my router (openmptcprouter )that have two usb 4g modem toward my server on port 1935 and I want openmptcprouter router close all other port and ip except server’s IP (77.36.23x.xxx)and port 1935

Ysurac, 2021-04-06:
Your WAN firewall zone is empty in screenshots. Did you set any WAN ?

rsajadi63, 2021-04-06:
I set everything up as I bought it

rsajadi63, 2021-04-06:
Data send toward server only via one wan

rsajadi63, 2021-04-06:
[image]

rsajadi63, 2021-04-06:
If I send you kernel log and and system log you can see how it is set to work properly

Ysurac, 2021-04-06:
WANs are in WAN firewall zone ? because it's not the case in screenshots.
What is the bandwidth of each wan ? Because WAN are used based on latency priority. So both WANs are used when lowest latency wan is saturated.

rsajadi63, 2021-04-06:
For WANs are in WAN firewall zone I should where check?

Ysurac, 2021-04-06:
In Network->Firewall, by editing zone and checking what interfaces are covered.
In your tests you don't seems to reach 4Mb/s on a WAN, so this can be logic.

rsajadi63, 2021-04-06:
I set all the options like the one that works right and came to the conclusion that the problem is here
[image]

rsajadi63, 2021-04-06:
When I deactivate the checkboxes of both modems (mine and the one I bought) traffic throughout via on wan but when I active the checkboxes of both modem traffic divided between two wan for one I bought and work well but for mine traffic stop !!!!!!

Ysurac, 2021-04-06:
Your rules mean that you forward traffic that enter port 1935 to port 2003 of the router.
Same for port 5001.

rsajadi63, 2021-04-06:
The server on port 1935 is listening
I do not know port 5001 what it is and not important I need only port 1935

rsajadi63, 2021-04-06:
Traffic enters the modem via LAN and must reach port 1935 of the server via a dual SIM card.

rsajadi63, 2021-04-06:
They did not use the wizard and only set the shadowsocks

rsajadi63, 2021-04-06:
One point, they have used version 54.7 could it be the cause of the problem ???

rsajadi63, 2021-04-07:
Where is config file?

Ysurac, 2021-04-07:
In System->OpenMPTCProuter, tab "Show all settings"
or via SSH in /etc/config
or via System->Backup and "Generate archive"

rsajadi63, 2021-04-07:
Ok I compare them

rsajadi63, 2021-04-07:
Tanks for your help my problem solved

closed 2021-04-10 by rsajadi63