Ysurac/openmptcprouter #1718: IP V4 forwarding off by default ?

onemarcfifty, 2021-04-05:
## Expected Behavior

On a "naked" OpenMPTCPRouter install (i.e. no VPS configured so far) the expected behavior would be that the router acts like a "normal" OpenWrt Router, i.e. routes traffic from LAN to WAN

## Current Behavior

IP V4 forwarding is off, i.e. no internet connection from the LAN side

## Possible Solution

In a shell, type
`echo "1" >/proc/sys/net/ipv4/ip_forward
`
## Steps to Reproduce the Problem

  1. Install OpenMPTCPRouter, but don't configure VPS yet
  2. Connect a LAN client
  3. Try to ping a public IP, e.g. 8.8.8.8

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: v0.57.3 r0+15225-bfc433efd4 
  - OpenMPTCProuter VPS version: N/A
  - OpenMPTCProuter platform: WRT3200ACM

Kalimeiro, 2021-04-05:
in System -> OpenMPTCProuter -> Wizard, you need to "disable server" called VPS by default or delete it (even if you haven't setup a vps)

onemarcfifty, 2021-04-06:
Hi @Kalimeiro, many thanks - I had disabled the VPS - in fact that was the first thing I tried. I had not deleted it however. Still I am struggling to understand why IPV4 forwarding is off by default on a router.

Ysurac, 2021-04-06:
IPv4 forwarding is enabled by default. At least it should and seems to be on a fresh x86_64 image.

onemarcfifty, 2021-04-10:
I'll double check as soon as I can.

closed 2021-07-14 by github-actions[bot]