Ysurac/openmptcprouter #1721: Erreur de connexion lors de l'exécution du script serveur

MrTheSkream, 2021-04-06:
Bonjour,

Lors de l'exécution du script de l'installation de serveur je tombe souvent aujourd'hui sur des erreur de connexion HTTP.
Y a t'il une maintenance de votre côté ?

Merci d'avance.
Cordialement

Ysurac, 2021-04-06:
non aucune. Et je ne vois pas de soucis au niveau du CDN utilisé.
Il s'agit du script stable actuel ?
Dans ce cas les erreurs se produisent en interrogeant openmptcprouter.com ou au niveau de github.com ?

MrTheSkream, 2021-04-06:
[image]
Voici une capture de l'erreur ça devrait être plus compréhensible ainsi.

MrTheSkream, 2021-04-06:
Et oui c'est bien le script stable disponible dans la section wiki, install VPS sur debian 10

Ysurac, 2021-04-06:
Je ne vois pas d'erreurs de mon côté.
Le VPS est hébergé chez qui ?

MrTheSkream, 2021-04-06:
C'est un EC2 de chez AWS.

Ysurac, 2021-04-06:
Un soucis avec le CDN peut être, tu peux relancer le script avec un peu de chance ça passera.

MrTheSkream, 2021-04-06:
Le soucis c'est que je l'ai lancer sur 3 serveurs et cela marche une fois sur 15 essaies.

Ysurac, 2021-04-06:
Je n'arrive pas à reproduire l'erreur.
Que donne un ```openssl s_client  -servername www.openmptcprouter.com -connect www.openmptcprouter.com:443``` ?

MrTheSkream, 2021-04-06:
> 43
> CONNECTED(00000003)
> depth=2 O = Digital Signature Trust Co., CN = DST Root CA X3
> verify return:1
> depth=1 C = US, O = Let's Encrypt, CN = R3
> verify return:1
> depth=0 CN = www.openmptcprouter.com
> verify return:1
> ---
> Certificate chain
>  0 s:CN = www.openmptcprouter.com
>    i:C = US, O = Let's Encrypt, CN = R3
>  1 s:C = US, O = Let's Encrypt, CN = R3
>    i:O = Digital Signature Trust Co., CN = DST Root CA X3
> ---
> Server certificate
> -----BEGIN CERTIFICATE-----
> MIIFNDCCBBygAwIBAgISBNVK7T6mbXSTGoxhk2IE1buBMA0GCSqGSIb3DQEBCwUA
> MDIxCzAJBgNVBAYTAlVTMRYwFAYDVQQKEw1MZXQncyBFbmNyeXB0MQswCQYDVQQD
> EwJSMzAeFw0yMTAyMTcxNzE4NDRaFw0yMTA1MTgxNzE4NDRaMCIxIDAeBgNVBAMT
> F3d3dy5vcGVubXB0Y3Byb3V0ZXIuY29tMIIBIjANBgkqhkiG9w0BAQEFAAOCAQ8A
> MIIBCgKCAQEAyG0eba/6+pqqmdhI2Yesx6/P8R0cinoiVx6qfBfiiBq8qmRqNYHi
> Xb6VZ/9yPlcecnFUr/sp9P5N+D35lOqdC+bHm9ztbR6KSjECtU1BBO40BocaMMBR
> hftz10Gc7gorhdfC4/+sFQ24F9eLc8dCUI+unZYhoXBY0OnPp2CicwjV8B0bfeUU
> oVLl2DyWx7Ifhf+w3rvREKW0K9AyT7vBNJsQ0GM7CFM5CEA0MhQKFaJepA4AdQMO
> 2sC5hLPjP+j0Zs4MBm2l3+u9Mc4t0t9UqKQaF10fqH5MjiWwiOa79HJReAl8rjsC
> BfJk5GksW39DmpozjHFoNWMlDvbbt38euQIDAQABo4ICUjCCAk4wDgYDVR0PAQH/
> BAQDAgWgMB0GA1UdJQQWMBQGCCsGAQUFBwMBBggrBgEFBQcDAjAMBgNVHRMBAf8E
> AjAAMB0GA1UdDgQWBBSlQu39lHhaL+dFeHQ5/cSVFE5aCTAfBgNVHSMEGDAWgBQU
> LrMXt1hWy65QCUDmH6+dixTCxjBVBggrBgEFBQcBAQRJMEcwIQYIKwYBBQUHMAGG
> FWh0dHA6Ly9yMy5vLmxlbmNyLm9yZzAiBggrBgEFBQcwAoYWaHR0cDovL3IzLmku
> bGVuY3Iub3JnLzAiBgNVHREEGzAZghd3d3cub3Blbm1wdGNwcm91dGVyLmNvbTBM
> BgNVHSAERTBDMAgGBmeBDAECATA3BgsrBgEEAYLfEwEBATAoMCYGCCsGAQUFBwIB
> FhpodHRwOi8vY3BzLmxldHNlbmNyeXB0Lm9yZzCCAQQGCisGAQQB1nkCBAIEgfUE
> gfIA8AB2APZclC/RdzAiFFQYCDCUVo7jTRMZM7/fDC8gC8xO8WTjAAABd7E3QVcA
> AAQDAEcwRQIhAIElz6I4rShki4sZge1pC0/hff8wui20X7jrxuP/nTIAAiBpA3xf
> qlH+1r8kXWaeKwQxVgjE8Y2f1hb0jhTM6pF5AQB2AG9Tdqwx8DEZ2JkApFEV/3cV
> HBHZAsEAKQaNsgiaN9kTAAABd7E3QZgAAAQDAEcwRQIgLDvt27ZAhIBUXiTB0c/w
> +ehtCBnNLXnaEdFF/5JkBA4CIQCotiD0g/5aZRewkX04zFLBKs2YC96RTR1Q79RZ
> 3T/rdDANBgkqhkiG9w0BAQsFAAOCAQEAOf6yVG988OjH3sN8lu6haiL2uP8rFp07
> P0XicVPYipgGtlvD+IBOBem8XaGIEvhPdr8FNq7+gGzn2seGZZx82rFQqXpTi8fI
> afCY+Cn+CtlKYJ25aIFmrTX9MiJyGFJZmRbVBRofqY/tdle8V77cEPun27D72Kf7
> 3JRXTbEx8Cit96Br0CIHJd/xhvIKJ74E3vR11N+DrrkV2rTImp2pq4XAm+D1lh6q
> +xuPZotQiCqgWXV1TBSVD5eU7ZRa8bf2hru8Zs/Xi806BAcAzm3qAmbvPqNW7A7t
> ySqPzmJLczf31E1QoWPnlUaF7IlgO8NIBoGxAkSAsaftuUpfEAc1cw==
> -----END CERTIFICATE-----
> subject=CN = www.openmptcprouter.com
> 
> issuer=C = US, O = Let's Encrypt, CN = R3
> 
> ---
> No client certificate CA names sent
> Peer signing digest: SHA256
> Peer signature type: RSA-PSS
> Server Temp Key: X25519, 253 bits
> ---
> SSL handshake has read 3030 bytes and written 395 bytes
> Verification: OK
> ---
> New, TLSv1.3, Cipher is TLS_AES_256_GCM_SHA384
> Server public key is 2048 bit
> Secure Renegotiation IS NOT supported
> Compression: NONE
> Expansion: NONE
> No ALPN negotiated
> Early data was not sent
> Verify return code: 0 (ok)
> ---
> ---
> Post-Handshake New Session Ticket arrived:
[... 46 lines cut ...]
>     TLS session ticket:
>     0000 - 3c 81 89 72 75 07 b2 84-1a f0 2e f3 47 f2 48 39   <..ru.......G.H9
>     0010 - c7 99 1d 87 98 3c d5 da-31 3a cc 8a c8 69 4f e8   .....<..1:...iO.
>     0020 - 0c 19 9d 92 83 3d d4 06-7d b1 fa 55 62 f3 d3 8a   .....=..}..Ub...
>     0030 - c6 bb 7f dc f7 83 56 97-09 a2 03 34 c0 e6 7a d4   ......V....4..z.
>     0040 - fa c0 77 57 a5 9c 73 70-73 78 d1 41 e4 af 3a e8   ..wW..spsx.A..:.
>     0050 - a5 7a 16 9b f4 81 ba dd-eb 6c 27 75 d5 2f fa df   .z.......l'u./..
>     0060 - c3 b3 b7 4f 80 b0 75 4c-83 01 f5 3f 83 63 01 b4   ...O..uL...?.c..
>     0070 - b9 20 df 78 80 23 a5 9f-fd cc e0 90 b4 b7 73 24   . .x.#........s$
>     0080 - e8 af 65 ab 0f cc 3b bf-cb 12 f6 2a f9 21 88 af   ..e...;....*.!..
>     0090 - 45 26 b7 87 3a 5b 64 18-96 0b 01 85 b2 8e da c1   E&..:[d.........
>     00a0 - 1e 48 c7 99 62 70 45 43-b7 f3 68 51 be 2f 21 7d   .H..bpEC..hQ./!}
>     00b0 - 8c 2e 74 27 0d 6a d3 87-07 fa f2 10 66 ae e8 70   ..t'.j......f..p
>     00c0 - 5f ef e3 6d 8e 56 86 05-18 08 e3 4d 6b 63 91 3a   _..m.V.....Mkc.:
>     00d0 - cc 9f a1 a1 26 da 9d 98-d1 28 fb 22 d2 8e 6b ec   ....&....(."..k.
>     00e0 - 36 c7 75 79 be 99 a7 bf-f2 de 31 3b 3b fd 18 e1   6.uy......1;;...
> 
>     Start Time: 1617719871
>     Timeout   : 7200 (sec)
>     Verify return code: 0 (ok)
>     Extended master secret: no
>     Max Early Data: 0
> ---
> read R BLOCK
> closed

MrTheSkream, 2021-04-06:
J'exécuterai plusieurs fois le script a chaque installation si ça recommence, c'est pas grave

closed 2021-07-11 by github-actions[bot]