Ysurac/openmptcprouter #1723: Support for NanoPi R4S

starkovv, 2021-04-06:
Trying to get OMR running on NanoPi R4S.

Specification (see full spec at https://wiki.friendlyarm.com/wiki/index.php/NanoPi_R4S):
```
SoC: Rockchip RK3399
CPU: big.LITTLE，Dual-Core Cortex-A72(up to 2.0GHz) + Quad-Core Cortex-A53(up to 1.5GHz)
```

I try to build image for my R4S according to this manual https://github.com/Ysurac/openmptcprouter/wiki/Create-image-for-unsupported-platform and got this error:
```
OMR_TARGET="aarch64_cortex-a53" OMR_FEED_SRC="master" ./build.sh
[...]
[...]
[...]
Installing package 'luci-app-wireguard' from luci                                            
Installing package 'r8125' from openmptcprouter                                              
Installing package 'openmptcprouter-mini' from openmptcprouter                               
WARNING: Not overriding core package '6in4'; use -f to force                                 
WARNING: Not overriding core package 'umbim'; use -f to force                                
WARNING: Not overriding core package 'umbim'; use -f to force                                
Overriding core package 'umbim' with version from openmptcprouter                            
Done                                                                                         
Target aarch64_cortex-a53 not found ! You have to configure and compile your kernel manually.
```

Not sure where should I move from this point?

Ysurac, 2021-04-06:
As it's not a supported platform: 
```
cd aarch64_cortex-a53/source
make menuconfig 
make -j6
```

starkovv, 2021-04-06:
So I made:
```
cd aarch64_cortex-a53/source
make menuconfig
```

And for some reason the target system `Atheros ATH79` can't be changed:
```
     Target System (Atheros ATH79)  --->                  
     Subtarget (Generic)  --->                            
     Target Profile (8devices Carambola2)  --->           
     Target Devices  ----                                 
     Target Images  --->                                  
 [ ] Enable experimental features by default              
     Global build settings  --->                          
 [*] Advanced configuration options (for developers)  --->
 [ ] Build the OpenWrt Image Builder                      
 - -   Include package repositories                       
 [ ] Build the OpenWrt SDK                                
 [ ] Package the OpenWrt-based Toolchain                  
 [*] Image configuration  --->                            
     Base system  --->                                    
     Administration  --->                                 
     Boot Loaders  ----                                   
     Development  --->                                    
     Firmware  --->                                       
     Fonts  --->                                          
     IPv6  --->                                           
     Kernel modules  --->                                 
     Languages  --->                                      
     Libraries  --->                                      
     LuCI  --->                                           
     Mail  --->                                           
     Network  --->                                        
     OpenMPTCProuter  --->                                
     Utilities  --->                       
```

Ysurac, 2021-04-06:
This can be changed: Enter on it then choose and use spacebar.
But in any case, not sure that compilation will work and that image will work.

starkovv, 2021-04-06:
> This can be changed: Enter on it then choose and use spacebar.
Hitting Enter returns to the main menu.

I've solved it with `rm -rf aarch64_cortex-a53` and running `OMR_TARGET="aarch64_cortex-a53" OMR_FEED_SRC="master" ./build.sh` again.

> But in any case, not sure that compilation will work and that image will work.
Any chance to see prebuilt image any time soon?

Ysurac, 2021-04-06:
Possible, if someone send me a NanoPi R4S, send me money to buy one or if enough person are interested in the support I will buy it.

suyuan168, 2021-04-07:
R4S is expensive. The price is similar to that of X86. It's better to use X86

Kalimeiro, 2021-04-07:
WOW, i think 87.99$ with metalcase and 5v 3A usb-c power is not so expensive.

And, Rockchip RK3399 has AES working ????

triple-it, 2021-04-09:
@Ysurac where do you want to receive the R4S or the money?

Ysurac, 2021-04-09:
@triple-it 
For sending money, there is paypal: https://www.paypal.me/ycarus
For sending material, address is at bottom of this page: https://www.openmptcprouter.com/about
Thanks ;)

triple-it, 2021-04-09:
> @triple-it
> For sending money, there is paypal: https://www.paypal.me/ycarus
> For sending material, address is at bottom of this page: https://www.openmptcprouter.com/about
> Thanks ;)

Done! (I thought Donation would be simplest)

I have ordered from the following url:
https://www.friendlyarm.com/index.php?route=product/product&product_id=284

If I can help, let me know!
(Hoping to receive my R4S soon, but FriendlyArm is not always very fast, I guess)

Ysurac, 2021-04-09:
Thanks, I've ordered it.

Adorfer, 2021-04-09:
i ordered mine yesterday. it just seems to be too good to be true! (for openwrt it's already supported and runs great for wireguard)

triple-it, 2021-04-19:
Hi, I received mine R4S today. Setting up the build environment right know. I will start with a test-run for the R2S. What should be changed to support R4S? Only some kernel "make menuconfig" configuration? I see FriendlyElec also has a R4S repo, which I could try first.

Ysurac, 2021-04-19:
Mine is not sent yet...
Should be only make menuconfig config.

triple-it, 2021-04-19:
Should/could we support BBR2 on the r4s?

# Add BBR2 patch, only working on 64bits images for now
if [ "$OMR_TARGET" = "x86_64" ] || [ "$OMR_TARGET" = "bpi-r64" ] || [ "$OMR_TARGET" = "rpi4" ] || [ "$OMR_TARGET" = "espressobin" ] || [ "$OMR_TARGET" = "r2s" ] || [ "$OMR_TARGET" = "rpi3" ]; then

Ysurac, 2021-04-19:
I think it's 64bits, so no problem to add it.

triple-it, 2021-04-20:
Big build btw!

$ du -sh .
12G     .

Status update:
===========
Somehow, something, some build.sh changes the TARGET correctly to r4s, but in the process somewhere to r2s ..

$ more ../config-r4s
CONFIG_TARGET_rockchip=y
CONFIG_TARGET_rockchip_armv8=y
CONFIG_TARGET_rockchip_armv8_DEVICE_friendlyarm_nanopi-r4s=y
CONFIG_PACKAGE_kmod-6lowpan=y
CONFIG_KERNEL_ARM_MODULE_PLTS=y
CONFIG_KERNEL_TCP_CONG_BBR2=y

$ OMR_OPENWRT="master" OMR_TARGET="r4s" OMR_FEED_SRC="master"  sh -x ./build.sh -j`nproc` download world

But somehow the CONFIG_TARGET transforms during the build.sh proces to:

$ grep CONFIG_TARGET r4s/source/.config | grep -v "not set"
CONFIG_TARGET_rockchip=y
CONFIG_TARGET_rockchip_armv8=y
CONFIG_TARGET_rockchip_armv8_DEVICE_friendlyarm_nanopi-r2s=y
CONFIG_TARGET_BOARD="rockchip"
CONFIG_TARGET_SUBTARGET="armv8"
CONFIG_TARGET_PROFILE="DEVICE_friendlyarm_nanopi-r2s"
CONFIG_TARGET_ARCH_PACKAGES="aarch64_generic"

ending up in
~~
r4s/source/bin/targets/rockchip/armv8/openmptcprouter-v0.58beta7-r16575-4dcdc8249c-rockchip-armv8-friendlyarm_nanopi-r2s-squashfs-sysupgrade.img.gz
~~

starkovv, 2021-05-08:
UP!

Ysurac, 2021-05-08:
My R4S has just been sent today ...

starkovv, 2021-05-28:
May we close this issue as v0.58RC4 has been released?

Ysurac, 2021-05-28:
I need feedbacks on OpenMPTCProuter on R4S ;)

starkovv, 2021-06-10:
Not stable.

starkovv, 2021-06-12:
UPDATE5:

Resolved:
– the issue with constantly rebooting modems has been resolved by replacing the USB hub to the new one with strong external power
– the issue with non-working USB port on R4S disappeared after reflashing the SD-card

Still there:
– System –> OpenMPTCProuter –> Status returns "Network overview. No Data" message instead of network diagram. This has started happening after two or three system soft reboot.
– The Internet stopped working on my client

Ysurac, 2021-06-13:
For the diagram, check System log.
For DNS, by default root DNS servers are used, so this should always work. You can try via SSH on the router if it's working or not (if not this may be why the diagram is not show). And check DNS configuration in Network->DHCP&DNS and also in Services->Recursive DNS.

closed 2021-09-16 by github-actions[bot]