Ysurac/openmptcprouter #1726: Edgerouter configuration

kverkind, 2021-04-07:
Used the procedure explained in https://github.com/Ysurac/openmptcprouter/wiki/Router-install
openwrt-18.06-snapshot-r7911-f65330d27d-ramips-mt7621-ubnt-erx-initramfs-factory.tar
After that could connect using eth3
Next step I used:
openmptcprouter-v0.58beta5-r0+16336-b36068d35d-ramips-mt7621-ubnt_edgerouter-x-initramfs-kernel.bin
Now router was completely blocked. Could not use at all Had to do a complete re-install.
What did I do wrong? Tried all ports on 192.168.100.1

Ysurac, 2021-04-07:
You tried with an IP in 192.168.100.x network ?
I may need to test again...

kverkind, 2021-04-07:
I did. On all ports. But during install I had to use sysupgrade -n -F (force) because I got an error that requirements are not met.

Ysurac, 2021-04-07:
True, -F was required because of a arch name change.
I will make tests asap.

Ysurac, 2021-04-07:
I modified the doc to use 19.07 as base image.
I don't remember if I need "-p" option to sysupgrade or not... I think not.

kverkind, 2021-04-07:
Will try and let you know

kverkind, 2021-04-07:
Tried with new versions but no luck.
Network adapter is in 192.168.100 range:
enp3s0f1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.100.20  netmask 255.255.255.0  broadcast 192.168.100.255
        ether 38:d5:47:3d:ef:5d  txqueuelen 1000  (Ethernet)
        RX packets 7103  bytes 817456 (817.4 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 32690  bytes 45719374 (45.7 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
Tried all ports eth0 to eth3not able to ping 192.168.100.1

kverkind, 2021-04-08:
Anything I can do to help debug. Working on some nice use case where this solution would be excellent.

Ysurac, 2021-04-08:
I updated my ER-X with https://download.openmptcprouter.com/refs/tags/v0.58beta5/5.4/ubnt-erx/targets/ramips/mt7621/openmptcprouter-v0.58beta5-r0%2B16336-b36068d35d-ramips-mt7621-ubnt_edgerouter-x-squashfs-sysupgrade.bin and no problem (only a very slow boot).
You need to use port eth0 to connect.
I think sysupgrade.bin is the image needed even for update from OpenWrt 19.06. You can try ?

kverkind, 2021-04-08:
.success. I am able to login. Documentation needs to be updated. Will work on next steps now and let you know.

Ysurac, 2021-04-09:
Thanks, I modified the doc.

guillaumed24, 2021-04-15:
Hi
I have the same problem.

First step ok but on the second i have same error and now i can't ping any address on any port.

Can you help please ? Thank

Ysurac, 2021-04-15:
@guillaumed24 Should work, doc was modified.

guillaumed24, 2021-04-15:
> @guillaumed24 Should work, doc was modified.

I think i have brick my er-x.

No response from any port.....

Ysurac, 2021-04-15:
To unbrick it: https://help.ui.com/hc/en-us/articles/360018189493-EdgeRouter-Manual-TFTP-Recovery

kverkind, 2021-04-15:
Worked for me. Tested it with one WAN and was working OK. This week-end I will do real test with 2 WAN's.

kverkind, 2021-04-17:
Please note use eth2 port after installing openwrt.  Eth0 will not work

Ysurac, 2021-04-17:
On OpenWrt state before flashing OpenMPTCProuter ?

guillaumed24, 2021-04-17:
last step i have this error

Device ubnt-erx not supported by this image
Supported devices: ubnt,edgerouter-x ubnt-erx ubiquiti,edgerouterx - Image version mismatch: image 1.1, device 1.0. Please wipe config during upgrade (force required) or reinstall. Reason: Config cannot be migrated from swconfig to DSA
Image check failed but --force given - will update anyway!
Commencing upgrade. Closing all shell sessions.

guillaumed24, 2021-04-17:
It work, must wait
eth0 ok 192.168.100.1
slow boot yes

Ysurac, 2021-04-17:
Yes normal as indicated, openwrt 19 use swconfig, my image use DSA, it's why it need to be forced.
Boot is very slow yes.

guillaumed24, 2021-04-17:
perfect thank, 
what appen if i use this router with same parameter of other router already work on my vps ?

Ysurac, 2021-04-17:
You can't use multiples routers on same VPS at the same time.

kverkind, 2021-04-17:
> On OpenWrt state before flashing OpenMPTCProuter ?

yes

kverkind, 2021-04-19:
Resolved