Ysurac/openmptcprouter #1730: Design personal HTML pages

rsajadi63, 2021-04-07:
It is possible manipulation of HTML pages ?
Is 192.168.100.8/cgi-bin/luci/admin in access?

Adorfer, 2021-04-07:
Yes!

rsajadi63, 2021-04-08:
> Yes!

How can I ? Please describe the steps

Adorfer, 2021-04-08:
edit files in /www/ on the OMR.

rsajadi63, 2021-04-08:
> edit files in /www/ on the OMR.

Permission is denied!!

Adorfer, 2021-04-08:
please use the template.

rsajadi63, 2021-04-08:
> please use the template.

What is template?

Adorfer, 2021-04-08:
https://github.com/Ysurac/openmptcprouter/issues/new/choose

rsajadi63, 2021-04-09:
I use template please help me

rsajadi63, 2021-04-09:
How can I change ‘openMPTCProuter’ that displayed at top of all pages where background color is blue?

rsajadi63, 2021-04-09:
[image]

Ysurac, 2021-04-09:
Find yourself in openmptcprouter-feeds repo, I will not help on this as it's not related to OpenMPTCProuter operation.

closed 2021-04-09 by Ysurac