Ysurac/openmptcprouter #1731: USB Stick for my windows PC

dooley1981, 2021-04-08:
What is the best USB stick for my windows computer to install the software on? I have acheap one now that was $8 for 32GB. Is it a benifit getting a more powerful stick?

Ysurac, 2021-04-08:
32GB is not enough for x86_64 version.
But I don't understand when you say windows PC, OpenMPTCProuter is an OS so it can't run under windows, the only way to run it "under windows" is to use a VM.

dooley1981, 2021-04-08:
I already have this setup on my pc. It's booting from the USB stick, not windows. You didn't, I do have a 64GB stick but my question is, will I get better performance with a faster stick.

Ysurac, 2021-04-08:
No. OpenMPTCProuter doesn't read/write a lot to disk (and yes it's GB, I don't know why I was in MB).

dooley1981, 2021-04-08:
Okay thanks. What's the minimum space needed on the stick?

Ysurac, 2021-04-08:
About 1GB

dooley1981, 2021-04-08:
Okay thanks

closed 2021-04-08 by dooley1981