Ysurac/openmptcprouter #1732: When one of wans down and then up  it will have no effect!!!

rsajadi63, 2021-04-08:
I use 3 USB 4g modem when all three is connected and comes up I send stream toward server and traffic throughout 3 wans and all things is ok but when I unplug one of modems traffic throughout 2 wans and it is ok I plug USB modem again and it comes up but traffic throughout only two previous wan And does not pass through it why???

Adorfer, 2021-04-08:
can you please use the template. In order to understand the issue there would be a lot of guessing/assumptions, which are probably not correct and result to inefficient communication

rsajadi63, 2021-04-08:
> can you please use the template. In order to understand the issue there would be a lot of guessing/assumptions, which are probably not correct and result to inefficient communication

What is template??

Adorfer, 2021-04-08:
https://github.com/Ysurac/openmptcprouter/issues/new/choose
And perhaps adjust your keyboard's autorepeat to a hight threshold.

rsajadi63, 2021-04-08:
When I plug USB modem again I should stop sending stream and start again

Ysurac, 2021-04-09:
Check State->System log when you plug the modem.

rsajadi63, 2021-04-09:
i config two wan (WAN1 and WAN2)
system log when two usb pluged
Thu Apr  8 16:11:56 2021 daemon.info dnsmasq[1940]: using only locally-known addresses for domain use-application-dns.net
Thu Apr  8 16:11:56 2021 daemon.info dnsmasq[1940]: using only locally-known addresses for domain lan
Thu Apr  8 16:11:56 2021 daemon.info dnsmasq[1940]: using nameserver 127.0.0.1#5353
Thu Apr  8 16:11:56 2021 daemon.info dnsmasq[1940]: using 1 more local addresses
Thu Apr  8 16:11:56 2021 daemon.info dnsmasq[1940]: read /etc/hosts - 4 addresses
Thu Apr  8 16:11:56 2021 daemon.info dnsmasq[1940]: read /tmp/hosts/dhcp.cfg01411c - 0 addresses
Thu Apr  8 16:11:57 2021 authpriv.info dropbear[2037]: Not backgrounding
Thu Apr  8 16:11:57 2021 user.notice ModemManager: hotplug: add network interface ip6tnl0: event processed
Thu Apr  8 16:11:57 2021 user.notice ModemManager: hotplug: event reported: action=add, name=ip6tnl0, subsystem=net
Thu Apr  8 16:11:57 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Apr  8 16:11:57 2021 user.notice ModemManager: hotplug: add network interface sit0: event processed
Thu Apr  8 16:11:57 2021 user.notice ModemManager: hotplug: event reported: action=add, name=sit0, subsystem=net
Thu Apr  8 16:11:57 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Apr  8 16:11:57 2021 kern.info kernel: [   12.109559] usb 1-1.3: new high-speed USB device number 5 using dwc_otg
Thu Apr  8 16:11:57 2021 user.notice Shadowsocks: Rules UP
Thu Apr  8 16:11:57 2021 user.notice ModemManager: hotplug: add network interface gre0: event processed
Thu Apr  8 16:11:57 2021 user.notice ModemManager: hotplug: event reported: action=add, name=gre0, subsystem=net
Thu Apr  8 16:11:57 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Apr  8 16:11:57 2021 kern.info kernel: [   12.240270] usb 1-1.3: New USB device found, idVendor=12d1, idProduct=1f01, bcdDevice= 1.02
Thu Apr  8 16:11:57 2021 kern.info kernel: [   12.250953] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Thu Apr  8 16:11:57 2021 kern.info kernel: [   12.259451] usb 1-1.3: Product: HUAWEI_MOBILE
Thu Apr  8 16:11:57 2021 kern.info kernel: [   12.264957] usb 1-1.3: Manufacturer: HUAWEI_MOBILE
Thu Apr  8 16:11:57 2021 user.notice ModemManager: hotplug: add network interface gretap0: event processed
Thu Apr  8 16:11:57 2021 user.notice ModemManager: hotplug: event reported: action=add, name=gretap0, subsystem=net
Thu Apr  8 16:11:57 2021 kern.info kernel: [   12.270837] usb 1-1.3: SerialNumber: 0123456789ABCDEF
Thu Apr  8 16:11:57 2021 kern.debug kernel: [   12.277760] usb 1-1.3: no of_node; not parsing pinctrl DT
Thu Apr  8 16:11:57 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Apr  8 16:11:57 2021 kern.debug kernel: [   12.301804] uas 1-1.3:1.0: no of_node; not parsing pinctrl DT
Thu Apr  8 16:11:57 2021 kern.debug kernel: [   12.308699] usb-storage 1-1.3:1.0: no of_node; not parsing pinctrl DT
Thu Apr  8 16:11:57 2021 kern.info kernel: [   12.316379] usb-storage 1-1.3:1.0: USB Mass Storage device detected
Thu Apr  8 16:11:57 2021 kern.info kernel: [   12.324572] scsi host0: usb-storage 1-1.3:1.0
Thu Apr  8 16:11:57 2021 user.notice ModemManager: hotplug: add network interface erspan0: event processed
Thu Apr  8 16:11:57 2021 user.notice ModemManager: hotplug: event reported: action=add, name=erspan0, subsystem=net
Thu Apr  8 16:11:57 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Apr  8 16:11:57 2021 user.notice Shadowsocks: Reload omr-bypass rules
Thu Apr  8 16:11:57 2021 user.notice ModemManager: hotplug: add network interface ip6gre0: event processed
Thu Apr  8 16:11:57 2021 user.notice ModemManager: hotplug: event reported: action=add, name=ip6gre0, subsystem=net
Thu Apr  8 16:11:57 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Apr  8 16:11:58 2021 user.notice ModemManager: hotplug: add network interface bond0: event processed
Thu Apr  8 16:11:58 2021 user.notice ModemManager: hotplug: event reported: action=add, name=bond0, subsystem=net
Thu Apr  8 16:11:58 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Apr  8 16:11:58 2021 user.notice ModemManager: hotplug: add network interface teql0: event processed
Thu Apr  8 16:11:58 2021 user.notice ModemManager: hotplug: event reported: action=add, name=teql0, subsystem=net
Thu Apr  8 16:11:58 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Apr  8 16:11:58 2021 daemon.notice wpa_supplicant[2905]: Successfully initialized wpa_supplicant
Thu Apr  8 16:11:58 2021 user.notice : Added device handler type: 8021ad
Thu Apr  8 16:11:58 2021 user.notice : Added device handler type: 8021q
Thu Apr  8 16:11:58 2021 user.notice : Added device handler type: macvlan
Thu Apr  8 16:11:58 2021 user.notice : Added device handler type: veth
Thu Apr  8 16:11:58 2021 user.notice : Added device handler type: bridge
Thu Apr  8 16:11:58 2021 user.notice : Added device handler type: Network device
Thu Apr  8 16:11:58 2021 user.notice : Added device handler type: tunnel
Thu Apr  8 16:11:58 2021 user.notice ModemManager: hotplug: add network interface wlan0: event processed
Thu Apr  8 16:11:58 2021 user.notice ModemManager: hotplug: event reported: action=add, name=wlan0, subsystem=net
[... cut ...]
Fri Apr  9 12:42:06 2021 kern.err kernel: [  194.812831] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Apr  9 12:42:09 2021 user.notice omr-tracker: Launched
Fri Apr  9 12:42:09 2021 user.notice omr-tracker: Launching...
Fri Apr  9 12:42:09 2021 user.notice post-tracking-post-tracking: omrvpn down
Fri Apr  9 12:42:09 2021 kern.err kernel: [  197.836530] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Apr  9 12:42:09 2021 kern.err kernel: [  197.852599] brcmfmac: brcmf_cfg80211_dump_survey: chanspec failed (-512)

rsajadi63, 2021-04-09:
system log when one of usb modems unpluged
Thu Apr  8 16:12:02 2021 daemon.info dnsmasq[5928]: using only locally-known addresses for domain lan
Thu Apr  8 16:12:02 2021 daemon.info dnsmasq[5928]: using nameserver 127.0.0.1#5353
Thu Apr  8 16:12:02 2021 daemon.info dnsmasq[5928]: using 1 more local addresses
Thu Apr  8 16:12:02 2021 daemon.info dnsmasq[5928]: read /etc/hosts - 4 addresses
Thu Apr  8 16:12:02 2021 daemon.info dnsmasq[5928]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Thu Apr  8 16:12:02 2021 user.notice ucitrack: Setting up /etc/config/system reload dependency on /etc/config/luci_statistics
Thu Apr  8 16:12:02 2021 daemon.info dnsmasq-dhcp[5928]: read /etc/ethers - 0 addresses
Thu Apr  8 16:12:02 2021 daemon.info dnsmasq[5928]: read /etc/hosts - 4 addresses
Thu Apr  8 16:12:02 2021 daemon.info dnsmasq[5928]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Thu Apr  8 16:12:02 2021 daemon.info dnsmasq-dhcp[5928]: read /etc/ethers - 0 addresses
Thu Apr  8 16:12:02 2021 user.notice ucitrack: Setting up /etc/config/system reload dependency on /etc/config/dhcp
Thu Apr  8 16:12:02 2021 user.notice ucitrack: Setting up /etc/config/glorytun reload dependency on /etc/config/glorytun-udp
Thu Apr  8 16:12:02 2021 user.notice ucitrack: Setting up /etc/config/glorytun reload dependency on /etc/config/openmptcprouter-vps
Thu Apr  8 16:12:02 2021 daemon.notice netifd: Network device 'eth0' link is up
Thu Apr  8 16:12:02 2021 daemon.notice netifd: Interface 'lan' has link connectivity
Thu Apr  8 16:12:02 2021 kern.info kernel: [   17.362204] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xCDE1
Thu Apr  8 16:12:02 2021 user.notice ModemManager: hotplug: checking if ModemManager is available...
Thu Apr  8 16:12:02 2021 user.notice ucitrack: Setting up /etc/config/glorytun-udp reload dependency on /etc/config/openmptcprouter-vps
Thu Apr  8 16:12:02 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty58, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty58
Thu Apr  8 16:12:02 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty58, subsystem=tty
Thu Apr  8 16:12:02 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Apr  8 16:12:02 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty48, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty48
Thu Apr  8 16:12:02 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty48, subsystem=tty
Thu Apr  8 16:12:02 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Apr  8 16:12:03 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyAMA0, subsystem=tty, sysfspath=/sys/devices/platform/soc/3f201000.serial/tty/ttyAMA0
Thu Apr  8 16:12:03 2021 user.notice ModemManager: hotplug: event reported: action=add, name=ttyAMA0, subsystem=tty
Thu Apr  8 16:12:03 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Apr  8 16:12:03 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty1, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty1
Thu Apr  8 16:12:03 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty1, subsystem=tty
Thu Apr  8 16:12:03 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Apr  8 16:12:03 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty38, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty38
Thu Apr  8 16:12:03 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty38, subsystem=tty
Thu Apr  8 16:12:03 2021 user.notice ucitrack: Setting up /etc/config/shadowsocks-libev reload dependency on /etc/config/omr-tracker
Thu Apr  8 16:12:03 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Apr  8 16:12:03 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty28, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty28
Thu Apr  8 16:12:03 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty28, subsystem=tty
Thu Apr  8 16:12:03 2021 user.notice ucitrack: Setting up /etc/config/mlvpn reload trigger for non-procd /etc/init.d/mlvpn
Thu Apr  8 16:12:03 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Apr  8 16:12:03 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty56, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty56
Thu Apr  8 16:12:03 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty56, subsystem=tty
Thu Apr  8 16:12:03 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Apr  8 16:12:03 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty18, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty18
Thu Apr  8 16:12:03 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty18, subsystem=tty
Thu Apr  8 16:12:03 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Apr  8 16:12:03 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty46, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty46
Thu Apr  8 16:12:03 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty46, subsystem=tty
[... cut ...]
Fri Apr  9 12:46:32 2021 kern.err kernel: [  460.481588] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Apr  9 12:46:34 2021 user.notice post-tracking-post-tracking: Can't find omr-tracker-ss, restart omr-tracker...
Fri Apr  9 12:46:34 2021 user.notice omr-tracker: Launching...
Fri Apr  9 12:46:35 2021 kern.err kernel: [  463.474718] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Apr  9 12:46:38 2021 user.notice omr-tracker: Launched
Fri Apr  9 12:46:38 2021 user.notice post-tracking-post-tracking: omrvpn down

rsajadi63, 2021-04-09:
when usb modem pluged again 

Thu Apr  8 16:12:04 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty30, subsystem=tty
Thu Apr  8 16:12:04 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Apr  8 16:12:04 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty59, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty59
Thu Apr  8 16:12:04 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty59, subsystem=tty
Thu Apr  8 16:12:04 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Apr  8 16:12:04 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty20, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty20
Thu Apr  8 16:12:04 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty20, subsystem=tty
Thu Apr  8 16:12:04 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Apr  8 16:12:04 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty49, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty49
Thu Apr  8 16:12:04 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty49, subsystem=tty
Thu Apr  8 16:12:04 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Apr  8 16:12:04 2021 user.notice firewall: Reloading firewall due to ifup of wan1 (eth1)
Thu Apr  8 16:12:04 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty10, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty10
Thu Apr  8 16:12:04 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty10, subsystem=tty
Thu Apr  8 16:12:04 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Apr  8 16:12:04 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty2, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty2
Thu Apr  8 16:12:04 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty2, subsystem=tty
Thu Apr  8 16:12:04 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Apr  8 16:12:04 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty39, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty39
Thu Apr  8 16:12:04 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty39, subsystem=tty
Thu Apr  8 16:12:04 2021 user.notice unbound: default memory configuration
Thu Apr  8 16:12:04 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Apr  8 16:12:04 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty29, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty29
Thu Apr  8 16:12:04 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty29, subsystem=tty
Thu Apr  8 16:12:04 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Apr  8 16:12:04 2021 daemon.notice procd: /etc/rc.d/S90mptcp: Device 'bonding_masters' can't found!
Thu Apr  8 16:12:04 2021 daemon.notice procd: /etc/rc.d/S90mptcp: Use the hardware name like in ifconfig
Thu Apr  8 16:12:04 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty57, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty57
Thu Apr  8 16:12:04 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty57, subsystem=tty
Thu Apr  8 16:12:04 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Apr  8 16:12:04 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty19, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty19
Thu Apr  8 16:12:04 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty19, subsystem=tty
Thu Apr  8 16:12:04 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Apr  8 16:12:04 2021 user.notice MPTCP: No master multipath defined, setting it to eth1
Thu Apr  8 16:12:04 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty47, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty47
Thu Apr  8 16:12:04 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty47, subsystem=tty
Thu Apr  8 16:12:04 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Apr  8 16:12:05 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty0, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty0
Thu Apr  8 16:12:05 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty0, subsystem=tty
Thu Apr  8 16:12:05 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Apr  8 16:12:05 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty37, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty37
Thu Apr  8 16:12:05 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty37, subsystem=tty
Thu Apr  8 16:12:05 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Apr  8 16:12:05 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty27, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty27
Thu Apr  8 16:12:05 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty27, subsystem=tty
[... cut ...]
Fri Apr  9 12:48:55 2021 kern.err kernel: [  603.907782] brcmfmac: brcmf_cfg80211_dump_survey: Could not get noise (-110)
Fri Apr  9 12:48:57 2021 user.notice post-tracking-post-tracking: Can't find omr-tracker-ss, restart omr-tracker...
Fri Apr  9 12:48:57 2021 user.notice post-tracking-post-tracking: Can't find omr-tracker-ss, restart omr-tracker...
Fri Apr  9 12:48:57 2021 user.notice omr-tracker: Launching...
Fri Apr  9 12:49:00 2021 user.notice omr-tracker: Launched
Fri Apr  9 12:49:00 2021 user.notice omr-tracker: Launching...
Fri Apr  9 12:49:00 2021 user.notice post-tracking-post-tracking: omrvpn down

closed 2021-07-13 by github-actions[bot]