Ysurac/openmptcprouter #1733: Filesystem read only on Pi4

geeuk, 2021-04-08:
Hi,

I seem to have an issue with the PI4 running openmptcprouter that is goes in to file system read only when it powers off (it's used in the car).

I have to always run the command to fix this issue /usr/sbin/e2fsck -y -v -f /dev/mmcblk0p2 and reboot.

In the mount point I have ticked already the check mount point at startup option but despite this it still goes into read only system when power has been disconnected, is there a script to run at bootup to check if system read only and if positive to run the above command and reboot ?

Thanks,
Gokhan

rsajadi63, 2021-04-08:
I have this problem too 
I run openmptcprouter on raspberry pi 3

geeuk, 2021-04-08:
Ok I done a change the config I am going to test couple of times now and I will let you know if it fixed it. I will post back soon.

rsajadi63, 2021-04-08:
One question 
I use 3 USB 4g modem when all three is connected and comes up I send stream toward server and traffic throughout 3 wans and all things is ok but when I unplug one of modems traffic throughout 2 wans and it is ok I plug USB modem again and it comes up but traffic throughout only two previous wan And does not pass through it why???
I should stop sending and start again for solve this problem

geeuk, 2021-04-08:
okay.. after couple of hours of searching google it was a simple fix :) 👍 

Anyone experiencing the file system readonly issue when you power cut the pi running mptcprouter do the below

if already stuck in read only mode paste this and reboot:

`/usr/sbin/e2fsck -y -v -f /dev/mmcblk0p2`

and once system is writable than do this:

`In /boot/cmdline.txt, add rw just before rootwait`

@Ysurac maybe add this to the complied images in the future ? `In /boot/cmdline.txt, add rw just before rootwait` so this stops system going into read only mode when power is cut suddenly on pi's ? If possible.

closed 2021-04-08 by geeuk

Ysurac, 2021-04-09:
Done, this will be in next 0.58 beta.