Ysurac/openmptcprouter #1737: One USB modem connected but in status show two

rsajadi63, 2021-04-09:
Look at this picture
[image]
I connect only one USB modem dongle but state of wan2 is connected 
Operator is irancell !!!!!!

Ysurac, 2021-04-09:
You really need to use template, for that doesn't open issue as question.
wan2 have a gateway that is same as wan1, it's why there is same info.
One interface must be set as multipath master.

rsajadi63, 2021-04-09:
For use template I should delete all things are between <!   > and write mine question??

Ysurac, 2021-04-09:
When you open an issue, you don't need to delete  (it's comments, they are not displayed), only put what you want under the titles like "## Expected Behavior" where you put what you expect,...

rsajadi63, 2021-04-09:
Open this page for me when I want create a new issue 
[image]

Ysurac, 2021-04-09:
Because it's not the correct type. Question are for "Ask a general question, not related to an OpenMPTCProuter install" as indicated.

rsajadi63, 2021-04-09:
I should change type of request for example ‘help wanted’ ??

Ysurac, 2021-04-09:
to get the template yes.
For issues already open that you want to add template, you can copy the template and edit your already existing issues with that.

rsajadi63, 2021-04-09:
I done please help me for issues  #1732

closed 2021-07-13 by github-actions[bot]