Ysurac/openmptcprouter #1744: Change title openmptcprouter on all pages

rsajadi63, 2021-04-10:
## Expected Behavior
openMPTCProuter shows on all pages where background color is blue

## Current Behavior
How can I change it ??

## Specifications

  - OpenMPTCProuter version: 57.3
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter platform: RPI3 a

rsajadi63, 2021-04-10:
[image]

Ysurac, 2021-04-10:
Please stop opening issues on customization. It's not related to OpenMPTCProuter operation.

closed 2021-04-10 by Ysurac

rsajadi63, 2021-04-10:
what should I do for customization I need that

rsajadi63, 2021-04-10:
https://www.openmptcprouter.com/commercial
You mentioned it here

Adorfer, 2021-04-10:
This is not how it works. 
either you understand why the linked "commercial" page applies for your request and act accordingling or you have to figure it out yourself. 
in other words: This is NOT a valid issue for this project.