Ysurac/openmptcprouter #1746: Optimal Tweaks for solely 4G LTE Bonding

geeuk, 2021-04-11:
Hi Everyone,

I have been spending hours last couple of weekends trying to make an optimal bonding setup with only Lte 4G dongle setup.

Every step took hours and hours but eventually i have a working setup but I seem not be getting great speeds as in the LTE infrastructure is very weak when it comes to heavy load even basic speedtest after second run of speed test I can see the ping on each connection gives gateway ping response failed 1 second ago.

I know with dsl cable etc bonding will be more better as its more strong.

Has anyone had any optimal settings such and tweaking the ping check times and using a particular VPN method ???

Because the gateway gives up so quick on each interface max I am getting is like 30Mbps 4 LTe bonded, if I do a omr test speed each dongle gives about 15Mbps.. I just dont understand where the issue is.

Thanks
Gokhan

geeuk, 2021-04-11:
Below is example of what I mean

root@OpenMPTCProuter:~# omr-test-speed eth1tmp
Select best test server...
Best server is http://speedtest.tele2.net/1000GB.zip, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 1000G    0 43.3M    0     0   966k      0  12d 13h  0:00:45  12d 13h  945k^C

root@OpenMPTCProuter:~# omr-test-speed eth2
Select best test server...
Best server is http://ams.download.datapacket.com/10000mb.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 9536M    0 33.3M    0     0  1470k      0  1:50:41  0:00:23  1:50:18 1791k^C

root@OpenMPTCProuter:~# omr-test-speed eth2tmp
Select best test server...
http://par.download.datapacket.com/10000mb.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 9536M    0 25.2M    0     0  1134k      0  2:23:25  0:00:22  2:23:03 1027k^C

root@OpenMPTCProuter:~# omr-test-speed eth4
Select best test server...
Best server is http://mad.download.datapacket.com/10000mb.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 9536M    0 35.6M    0     0  1637k      0  1:39:22  0:00:22  1:39:00 1413k^C

root@OpenMPTCProuter:~# omr-test-speed
Select best test server...
Best server is http://proof.ovh.net/files/10Gb.dat, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  7 1192M    7 91.1M    0     0  1981k      0  0:10:16  0:00:47  0:09:29 2193k^C

Ysurac, 2021-04-11:
You can enable SQM.

geeuk, 2021-04-11:
Hi Yannick,

Thanks for your reply. 

I have now enabled SQM with up/down setting on 60000 Kb/s, should I left this on 0?

Speed has improved a bit with above setup.

Also Glorytun TCP works okay but when I switch to Glorytun UDP it always says VPN proxy Down, I rebooted the server and pi still like that, I forced the key update. I tried with two Cloud servers both I had no joy with Glorytun UDP.

Also which method of Congestion Control is happy with LTE ? I am currently using cubic.

geeuk, 2021-04-11:
Hi Again,

@Ysurac 

I am slowly giving up on this project unfortunately trying for litterally weeks and weeks, I belive this setup does not work with LTE connections.

My setup is all okay everything looks okay but It just seems to not be bonding well.

I have 12V 5A anker usb hub with pi4 B, 4 LTE dongle all healthy running without any issues.
I enabled sqm qos selected cake method but still getting very low bond please below:

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
 12 1192M   12  145M    0     0  1197k      0  0:16:59  0:02:04  0:14:55  846k^C

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  2 9536M    2  197M    0     0  1653k      0  1:38:27  0:02:02  1:36:25 2003k^C

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 1000G    0  191M    0     0  1612k      0   7d 12h  0:02:01   7d 12h 1081k^C

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  1 9536M    1  165M    0     0  1390k      0  1:57:00  0:02:01  1:54:59 1076k^C

Total : 5852k of all 4 eth connections.

Bonding I get::

root@OpenMPTCProuter:~# omr-test-speed
Select best test server...
Best server is http://speedtest.tele2.net/1000GB.zip, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 1000G    0  428M    0     0  3440k      0 84:39:37  0:02:07 84:37:30 2993k^C

Ysurac, 2021-04-12:
If all LTE connections are on same provider, you can reach provider antenna limits.
Also if connections have high latency or losing too much packets, you can't get a really good result.
Some parameters can be configured in Network->MPTCP to get a better speed in some case, like TCP Congestion Control, MPTCP scheduler or Multipath TCP SYN retries.

geeuk, 2021-04-13:
Hi,

Yes you are correct at 5.30AM when the network has less traffic, I get better results.

I think my issue is in fact the wifi settings on the PI, everytime I get 25Mbps I just did another test directly on the PI4B and my results are below:

[image]

geeuk, 2021-04-13:
The wifi on the raspberry pi4b I get max 25mbps down/up has anyone had any better results on pi4b ?

hairy-dog, 2021-04-16:
I wouldn't even think about using the wifi on the Raspberry Pi. I use the network port and proper access points. I'm surprised you get such good results.
For various reasons, I'm currently using just two 4G connections, though from different networks, different base locations. 
The speeds of the individual connections are nothing special, but the total is about the sum of the individuals:
down (peak) 16.15 Mbit/s + 35.44 Mbit/s => 49.94 Mbit/s
up (peak) 37.32.Mbit/s + 26.47 Mb/s => 63.79 Mbit/s

Ysurac, 2021-04-16:
@geeuk you can try to update RPI4 firmware: https://github.com/Ysurac/openmptcprouter/wiki/Update-RPI4-firmware

geeuk, 2021-04-16:
@hairy-dog 

You are 100% right, as I used it in the car as a mobile hotspot I never thought the wifi in a pi4 would be bad, but after plugging in my usb-c to ethernet to my phone I was shocked to see 70mbps speeds.

I got around the issue by adding a small access point to ethernet port.

My next issue was that when I do a speedtest with speedtest.net, omr-test-speed or curl download test files, the connection on each dongle would drop and cause the tun0 to stop meaning the bond breaks as tun goes off, so to be honest I just gave up.

I have a powerful usb hub 5amp 12v from anker so power was not an issue.

I tweaked the settings of sqm qos and used various settings tried for days and finally couldn't find a way of having a stable connection.

@Ysurac 

I might give it another try with fresh start and update the firmware on the pi.

3ricj, 2021-04-21:
I'm with you here, I've tried lots of different settings and can't find anything stable.  I suspect this project doesn't actually work at all with USB cell modems.  It claims to but it's really extremely unstable.  I'm rolling back to openWRT and normal load balancing -- it's working well for me on the same hardware.

Ysurac, 2021-04-21:
@3ricj not really a good way to get help on free software...

geeuk, 2021-04-24:
> I'm with you here, I've tried lots of different settings and can't find anything stable. I suspect this project doesn't actually work at all with USB cell modems. It claims to but it's really [...]

I kind of agree but I don't blame the software as I know if I had a stable line ie adsl on 1 Wan and say 6 LTE dongles this system would work excellent because the adsl line would always keep the service going.

I would like to thank @Ysurac as he has done an excellent work and would always be around to see if any changes etc.

hairy-dog, 2021-04-25:
I found that having two 4G connections was no faster than having two 4G connections (two different providers, three different base stations). And with FTTC VDSL, adding two 4G connections was no faster than one.
This is running OMR on a Pi4, which only has one ethernet port.
I just added an additional ethernet port (using a USB3 to ethernet adapter) and it has immediately made the speeds much better. It seems that the ethernet port had become a bottleneck.
I now have the wan connections on eth1: and the lan connection on eth0:

hairy-dog, 2021-04-28:
Update: after the performance increase from adding a second Ethernet port to my Pi4 I hadn't expected that it could get much better. I was wrong.
Last night the VPS was moved to newer hardware. I'm not sure if the details of what changed, but it involved the VPS being powered down, moved and started up again. 
My guess is a faster ethernet port, but that is only a guess. The performance improvement is significant, whatever caused it.

Ysurac, 2021-04-28:
Thanks for the feedback.
Maybe previous VPS host server was saturated.

closed 2021-08-02 by github-actions[bot]