Ysurac/openmptcprouter #1747: Erreur omr-tracker ; Command line is not complete

flimo44, 2021-04-11:
## Expected Behavior

hello,

In the system log I have continuous "Error  daemon.err omr-tracker". I can't see where it came from.

## Current Behavior
System log -->
Sun Apr 11 17:49:28 2021 daemon.err omr-tracker[19505]: Command line is not complete. Try option "help"
Sun Apr 11 17:49:29 2021 daemon.err omr-tracker[19506]: Command line is not complete. Try option "help"
Sun Apr 11 17:49:34 2021 daemon.err omr-tracker[19505]: Command line is not complete. Try option "help"
Sun Apr 11 17:49:35 2021 daemon.err omr-tracker[19506]: Command line is not complete. Try option "help"
Sun Apr 11 17:49:40 2021 daemon.err omr-tracker[19505]: Command line is not complete. Try option "help"
Sun Apr 11 17:49:41 2021 daemon.err omr-tracker[19506]: Command line is not complete. Try option "help"
Sun Apr 11 17:49:47 2021 daemon.err omr-tracker[19505]: Command line is not complete. Try option "help"
Sun Apr 11 17:49:48 2021 daemon.err omr-tracker[19506]: Command line is not complete. Try option "help"
Sun Apr 11 17:49:53 2021 daemon.err omr-tracker[19505]: Command line is not complete. Try option "help"
Sun Apr 11 17:49:54 2021 daemon.err omr-tracker[19506]: Command line is not complete. Try option "help"
Sun Apr 11 17:49:59 2021 daemon.err omr-tracker[19505]: Command line is not complete. Try option "help"
Sun Apr 11 17:50:01 2021 daemon.err omr-tracker[19506]: Command line is not complete. Try option "help"
Sun Apr 11 17:50:06 2021 daemon.err omr-tracker[19505]: Command line is not complete. Try option "help"
Sun Apr 11 17:50:08 2021 daemon.err omr-tracker[19506]: Command line is not complete. Try option "help"

## Specifications

  - OpenMPTCProuter version:  v0.58beta5 r0+16336-
  - OpenMPTCProuter VPS version: 0.1025-Test
  - OpenMPTCProuter platform: x86/x86_64

cayenne17, 2021-04-12:
I have the same issue.
yesterday, I switched from 0.57.3 to 0.58beta5 and noticed the same thing

Ysurac, 2021-04-12:
I hope to fix this in next beta, I added more check.

Ysurac, 2021-04-12:
I released beta 6, maybe it's fixed.

Ysurac, 2021-04-12:
cool :)

closed 2021-04-12 by Ysurac

flimo44, 2021-04-13:
Je confirme. Plus d'erreur en Beta6

Merci