Ysurac/openmptcprouter #1748: Cisco Anyconnect not working

ragefan, 2021-04-11:
## Expected Behavior
Connection to enterprise VPN with Cisco Any connect shall work Through openmptcprouter at home

Required config 

Protocol | Cisco AnyConnect Client Port
-- | --
TLS (SSL) | TCP 443
SSL Redirection | TCP 80
DTLS | UDP 443
IPsec/IKEv2 | UDP 500, UDP 4500

## Current Behavior

It fails to connect with v0.57.3 and latest beta of v.0.58

## Possible Solution

None? 
Was working with a successive version (intermediate build) to 0.57.3 and before 0.58 beta 1

## Steps to Reproduce the Problem

  1.start Cisco any connect client
  2.wait
  3.connection fails

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 0.57.3
  - OpenMPTCProuter VPS version: v. 0.1024
  - OpenMPTCProuter platform: rpi3

ragefan, 2021-04-20:
Seems to be due to MTU default mis-configuration.
Ping & connection is ok now with a lower MTU (default was 1500) but connection through IPSEC VPN is very very bad and not usuable.

ragefan, 2021-04-26:
Tested with 0.58rc1 same results.
Using omrbypass on one 4g connection only does not solve the issue so not linked to vps. Using same connection without omr through shared wifi is working fine.

docderwood, 2021-05-18:
Same issue for me.

Ysurac, 2021-05-18:
Difficult to tests for me as I have no VPN using Cisco Anyconnect.

ragefan, 2021-05-18:
How can we help?

Ysurac, 2021-05-18:
By providing a Cisco Anyconnect access ;)
Else, did you try to use V2Ray as proxy in System->OpenMPTCProuter, "advanced settings" checkbox ?

ragefan, 2021-05-18:
I will test that. Is there some logs we could collect?

ragefan, 2021-05-18:
V2ray is already activated by default on my side

ragefan, 2021-08-13:
Now Working with v.0.85.5...

closed 2021-08-13 by ragefan