Ysurac/openmptcprouter #1750: OMR-Bypass not working on a bridge interface as target

bhelm, 2021-04-12:
## Expected Behavior
I have setup a bridge interface on eth0 that is also used to talk to my wan2 router (so it has both the lan ip and the ip to talk to the wan2 router).
If i create an OMR-Bypass rule that should use the bridge interface, i expect the traffic to be routed using wan2.
[image]

## Current Behavior
The ORM-Bypass rules traffic gets routed using the "master" wan interface (wan1 in my case).
If i make wan2 the maser interface, i can create an rule on eth1 and it correctly forwards traffic to the non-master-interface. so my guess is that there is a problem with accepting the bridge interface.

## Steps to Reproduce the Problem

  1. Make the land interface a bridge and add eth0 to it
  2. make wan2 interface use the br-lan interface as device
 
[image]

  3. make wan1 the master for mptcp traffic
  4. add an OMR-Bypass rule to route traffic using wan2

## Context (Environment)
I Try to redirect all streaming traffic (disney, netflix) to the wan2 device, so i have the master wan1 free for the more important things like stream downloads and gaming. ;)

[image]

the status page shows "network interface duplicated". i dunno what it means.

Let me know if more information is required.

I tried to debug the iptables rules created, but i have to admit that i have no idea what is going on there with the xmarks. While debugging, i noticed that some rules getting duplicated every time i do save & apply in the omr bypass settings.

## Specifications

  - OpenMPTCProuter version: v0.57.3
  - OpenMPTCProuter VPS version: 0.1024
  - OpenMPTCProuter platform: Rpi4 b

Ysurac, 2021-04-12:
You have a really strange configuration, I don't know how this can work with aggregation. 
The network interface duplicated mean that more than one interface use same Mac address.

bhelm, 2021-04-12:
> The network interface duplicated mean that more than one interface use same Mac address.
ok, i guess that is normal for a bridge interface.

> You have a really strange configuration, I don't know how this can work with aggregation.
i dont know what you mean by aggregation. without an OMR-Bypass i can see both wans being utilized (in the bandwidth view). yes the configuration is special. i have created that bridge interface so i can have a full debian running in lxc using this bridge.

[image]

Kalimeiro, 2021-04-14:
if you want your lxc to use wan2, it's clearly not the right way ... you have a really strange configuration ...

a bridge has different mac adress than your first interface joined in that bridge ...

bhelm, 2021-05-06:
ok, i see, thanks.

closed 2021-05-06 by bhelm