Ysurac/openmptcprouter #1751: Balancing Network Traffic

nibirrayy, 2021-04-13:
I have currently 4 modem connected to my raspberry pi 4 which is running openmptcprouter. The problem is when I stress the network only one line is used to transfer most of the date. Is there is a way to balance this out.

Ysurac, 2021-04-13:
Usage of WANs is based on latency.

nibirrayy, 2021-04-13:
is there is a way to turn that off. I am using round robin ans still it uses one connection

Ysurac, 2021-04-13:
I will add an option to enable true round robin in next release so it will not fill the congestion window on all subflows.

pbapba512, 2021-04-24:
True round robin can be a really usefull feature, for me it can be a big help for the livestreaming usecases! :)
The another big wish from me is the better 4G USB modem handling, with the .56 it worked well, but with .57 its broken (at least for me and with my Huawei sticks). So I'm really exciting and waiting for the .58 finl release with the new features. 

Thank you for your work, Ysurac!

gsxryan, 2021-06-20:
Would that be 58/59 release?  will the current round robin option be changed, or will we look for a new option?

Ysurac, 2021-06-20:
The option is "Fill the congestion window on all subflows for round robin" that should set to false for true round robin, in Network->MPTCP in v0.58*

nibirrayy, 2021-09-19:
I am just commenting just keep this thread alive as the problem is still not completely fixed and I will like to give an update on the issue

currently round robin still heavily rely to latency from what I see on my day to day use. I have two LTE modem that I am combining and both have different providers.

Most of the time the provider which has less latency is still used to transfer more traffic instead of using pure round robin
I now have switched to zerotier to fixed this and from version 1.6.5 they also have multipathing support.

They have some really awesome way to force round robin by specifying parameters on asymmetric connections
check this link for more info about that
https://docs.zerotier.com/zerotier/multipath/#asymmetric-links

r2d2land, 2021-09-23:
> I now have switched to zerotier to fixed this and from version 1.6.5 they also have multipathing support.

Are you using ZT with OMR?

If so, where would you apply the configuration.

If not,

Ysurac…can the ZeroTier package be added to the current release thru the gui without breaking OMR?

thanks

closed 2021-12-27 by github-actions[bot]