Ysurac/openmptcprouter #1753: Pi4 router

It stopes when we change the usb ports or we shuffle the ports dongles 

    Also it get connected and disconnected by its own after few seconds ...it's keep connected and disconnected randomly

ssdeol84, 2021-04-15:
## Expected Behavior

It should keep connected with 4G dongles 3 usb dongles.
## Current Behavior

[image]
## Possible Solution

## Steps to Reproduce the Problem

  1. Tried reboot rpi4
  2. Tried to reboot VPS
  3. Shuffle usb 4G dongles and also change master randomly.

## Context (Environment)

Not getting constant speed it reduces upto 3-4 mbps download upload 2 mbps from 20 mbps

[image][image]

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter platform:

ssdeol84, 2021-04-15:
I'm on latest version of both vps and client on Debian 10

3ricj, 2021-04-18:
I don't know if this will solve your problem, but I'm working on setting up the same thing you are describing.  I also have a few cell usb dongles.  My first task was to try and stop the crazy renaming of interfaces; I finally learned a trick with the openWRT which doesn't seem to be documented anywhere:  If you rename an interface like this: 

ip link set eth1 name att0
ip link set eth2 name tmobile0

It remembers the name and sticks with it.  I had to fitz around with the logical interfaces (/etc/config/network) so everything lined up but this helped me keep my physical interfaces with the same name.  I'm still getting my VPS setup so I can't say if it actually helps your problem or not, but I thought I would share.  

Does anyone know if it's actually required to disable DHCP on the upstream modems?  It's not possible for me to disable this on all of my hotspots.  They just always have it.  I'm not sure if the requirement is a static address, or just not having DHCP servers around screwing things up.

3ricj, 2021-04-18:
In your configuration, I noticed that each dongle seems to be in the same subnet.  I'm not sure this will work. (192.168.4.1) because it's the same gateway address.  My rough understanding is that MPTCProuter uses different gateway addresses for separation of traffic (but I could be way off here, it's my first day).

ssdeol84, 2021-04-21:
Yes all dongles can use same gateway but during working it keeps disconnected and change the interface names by its own. Which keep me disconnected from tunnel..that leads poor bonding speed.

ImprezyLive, 2021-05-05:
First i suggest to change IP on modems (why? because if you have any web access to control modem, 1 address connect only with one device).
Second my modems i connect via HUB USB 3.0 or 3.1(gen2), and almost always after connect 4 modems, runs only 3 (some times diffrent, some times the sames). Disconnecting any modems from HUB VPN stop working for 5-15 sec.
And once small problems if modem run as master have bad quality signal or change from 4G to 3G or 2G VPN working not stable.

closed 2021-08-09 by github-actions[bot]