Ysurac/openmptcprouter #1754: Modem not connecting after reboot, log shows cannot find wwan0

jamesho451, 2021-04-17:
## Expected Behavior
Modem should automatically reconnect after rebooting the router

## Current Behavior
My Quectel modem won't connect after rebooting the router. I have to reflash the firmware every time I reboot to get it to work. Everything appears normal in the log until there's a line saying "Cannot find device "wwan0"". Also the log shows that it's a static ip device.

## Specifications

  - OpenMPTCProuter version: 5.4.86-mptcp 0.1024
  - OpenMPTCProuter VPS version: 5.4.86-mptcp 0.1024
  - OpenMPTCProuter platform: x86_64

Ysurac, 2021-04-17:
I don't understand what you mean by reflash firmware or by a static ip in log.

jamesho451, 2021-04-17:
> I don't understand what you mean by reflash firmware or by a static ip in log.

Reflash firmware means restore all router settings to default, I got the wording wrong, sorry about that. There is a line in the log like this "IPv4 connection setup required in interface wan1: static". Isn't the router supposed to get IP from my modem?

jamesho451, 2021-04-17:
I have an idea. Could it be that the Modem Manager's profile for the modem is tied to wwan0, but the physical interface of the modem changed after reboot? I've seen someone with this issue before, he has to change the physical interface of a connection after rebooting the modem.

Ysurac, 2021-04-17:
If you have only one modem, there is no reason.
For the message, what is the interface name you use for the modem ? wan1 ?

jamesho451, 2021-04-17:
> If you have only one modem, there is no reason.
> For the message, what is the interface name you use for the modem ? wan1 ?

My interface name is wan1, but when I plug in the modem the log says something like qmi-wwan0. I only have one modem going through ModemManager, the other interface, wan2, is set to dhcp client.

Ysurac, 2021-04-17:
qmi-wwan0 is the real interface name.
For ModemManager, you can user OpenWrt doc: https://openwrt.org/docs/guide-user/network/wan/wwan/modemmanager
You can try to use as device a path like in the doc.

jamesho451, 2021-04-17:
Turned out that it's a power issue. The modem physically crashed due to insufficient power every time I reboot. I swapped out my 1.5A charger to a 2.5A one and now the modem survives through reboot. Thank you so much for your help, I really appreciate your dedication to helping your users.

closed 2021-04-17 by jamesho451