Ysurac/openmptcprouter #1757: VPS Sapinet instabilité

artiflo, 2021-04-18:
Bonjour,

Je viens de souscrire a un VPS sapinet afin de remplacer mon VPS kimsufi dans le cadre de l'utilisation de la solution OpenMPTCPRouter.

Je rencontre toutefois un problème de stabilité la liaison entre mon routeur OMR et le VPS n'arrête pas de sauter. Alors que cela ne le faisait jamais avant avec la VPS Kimsufi et quand je rebascule sur la kimsufi j'ai pas du tout de déconnexion comme avec Sapinet.

Avec sapinet, OMR passe son temps de passer de vert à jaune en indiquant : can't contact Server Admin Script (no result - no api script answer). Version 0.1024 5.4.86-mptcp

OMR version : 0.57.3

Connaissez-vous un problème de compatibilité entre la solution OpenMPTCPRouter et les VPS sapinet ?

Ysurac, 2021-04-18:
Il faudrait alors voir dans Etat->Journal Système, si il y a vraiment un problème. Il faudrait aussi vérifier la latence de chaque connexion avec le VPS (normalement affiché dans la page status).
Si il y a un soucis, je peux tester Sapinet, pour 3€ ça ira :)

artiflo, 2021-04-18:
Merci pour ce retour rapide.

Voici ce que j'ai dans la page status : 

[image]

Voici un extrait du journal systeme :
Sun Apr 18 09:01:06 2021 daemon.err /usr/bin/ss-redir[19151]: remote recv: Host is unreachable
Sun Apr 18 09:01:06 2021 daemon.err /usr/bin/ss-redir[19150]: remote recv: Host is unreachable
Sun Apr 18 09:01:06 2021 user.notice OMR-VPS: Can't get sapin token, try later (can ping server sapin, no server API answer)
Sun Apr 18 09:01:07 2021 daemon.err /usr/bin/ss-redir[19151]: remote recv: Host is unreachable
Sun Apr 18 09:01:07 2021 daemon.err /usr/bin/ss-redir[19151]: remote recv: Host is unreachable
Sun Apr 18 09:01:08 2021 user.notice OMR-VPS: Can't get sapin token, try later
Sun Apr 18 09:01:08 2021 daemon.err /usr/bin/ss-redir[19152]: remote recv: Host is unreachable
Sun Apr 18 09:01:08 2021 daemon.err /usr/bin/ss-redir[19151]: remote recv: Host is unreachable
Sun Apr 18 09:01:10 2021 user.notice OMR-VPS: Can't get sapin token, try later (can ping server sapin, server API answer)
Sun Apr 18 09:01:13 2021 user.notice OMR-VPS: Can't get sapin token, try later
Sun Apr 18 09:01:13 2021 user.notice OMR-VPS: Can't get sapin token, try later
Sun Apr 18 09:01:20 2021 user.notice OMR-VPS: Can't get sapin token, try later
Sun Apr 18 09:01:20 2021 user.notice OMR-VPS: Can't get sapin token, try later
Sun Apr 18 09:01:23 2021 daemon.err /usr/bin/ss-redir[19153]: remote recv: Host is unreachable
Sun Apr 18 09:01:23 2021 user.notice OMR-VPS: Can't get sapin token, try later (can ping server sapin, no server API answer)
Sun Apr 18 09:01:25 2021 daemon.err /usr/bin/ss-local[19149]: getpeername: Socket not connected
Sun Apr 18 09:01:26 2021 user.notice OMR-VPS: Can't get sapin token, try later (can ping server sapin, no server API answer)
Sun Apr 18 09:01:28 2021 user.notice OMR-VPS: Can't get sapin token, try later (can ping server sapin, server API answer)
Sun Apr 18 09:01:36 2021 user.notice OMR-VPS: Can't get sapin token, try later
Sun Apr 18 09:01:37 2021 daemon.err /usr/bin/ss-redir[19150]: remote recv: Host is unreachable
Sun Apr 18 09:01:39 2021 user.notice OMR-VPS: Can't get sapin token, try later (can ping server sapin, no server API answer)
Sun Apr 18 09:01:40 2021 daemon.err /usr/bin/ss-redir[19153]: remote recv: Host is unreachable
Sun Apr 18 09:01:40 2021 user.notice OMR-VPS: Can't get sapin token, try later
Sun Apr 18 09:01:40 2021 user.notice OMR-VPS: Can't get sapin token, try later
Sun Apr 18 09:01:43 2021 daemon.err /usr/bin/ss-redir[19152]: remote recv: Host is unreachable
Sun Apr 18 09:01:43 2021 daemon.err /usr/bin/ss-redir[19152]: remote recv: Host is unreachable
Sun Apr 18 09:01:43 2021 user.notice OMR-VPS: Can't get sapin token, try later (can ping server sapin, no server API answer)
Sun Apr 18 09:01:46 2021 daemon.err /usr/bin/ss-local[19149]: getpeername: Socket not connected
Sun Apr 18 09:01:46 2021 daemon.info omr-tracker-ss: Shadowsocks is down (can't contact via http 74.82.42.42, 176.103.130.130, 1.1.1.1)
Sun Apr 18 09:01:46 2021 user.notice Shadowsocks: Rules DOWN
Sun Apr 18 09:01:52 2021 daemon.err /usr/bin/ss-local[19149]: getpeername: Socket not connected
Sun Apr 18 09:01:53 2021 user.notice OMR-VPS: Can't get sapin token, try later (can ping server sapin, no server API answer)
Sun Apr 18 09:01:56 2021 user.notice OMR-VPS: Can't get sapin token, try later (can ping server sapin, server API answer)
Sun Apr 18 09:01:57 2021 daemon.info omr-tracker-ss: Shadowsocks is up (can contact via http 212.27.48.10)
Sun Apr 18 09:01:57 2021 daemon.info omr-tracker-ss: Reload Shadowsocks rules
Sun Apr 18 09:01:57 2021 user.notice Shadowsocks: Rules UP
Sun Apr 18 09:01:58 2021 user.notice Shadowsocks: Reload omr-bypass rules
Sun Apr 18 09:01:58 2021 user.notice omr-bypass: Starting OMR-ByPass...
Sun Apr 18 09:01:59 2021 user.notice omr-bypass: Reload dnsmasq...
Sun Apr 18 09:01:59 2021 daemon.info dnsmasq[19438]: read /etc/hosts - 4 addresses
Sun Apr 18 09:01:59 2021 daemon.info dnsmasq[19438]: read /tmp/hosts/dhcp.cfg01411c - 9 addresses
Sun Apr 18 09:01:59 2021 daemon.info dnsmasq-dhcp[19438]: read /etc/ethers - 0 addresses
Sun Apr 18 09:01:59 2021 user.notice omr-bypass: OMR-ByPass is running
Sun Apr 18 09:02:00 2021 daemon.err /usr/bin/ss-local[19149]: getpeername: Socket not connected
Sun Apr 18 09:02:05 2021 daemon.err /usr/bin/ss-redir[19150]: remote recv: Host is unreachable
Sun Apr 18 09:02:05 2021 daemon.err /usr/bin/ss-local[19149]: getpeername: Socket not connected
Sun Apr 18 09:02:05 2021 daemon.err /usr/bin/ss-redir[19153]: remote recv: Host is unreachable
Sun Apr 18 09:02:05 2021 daemon.err /usr/bin/ss-redir[19153]: remote recv: Host is unreachable
Sun Apr 18 09:02:05 2021 daemon.err /usr/bin/ss-redir[19152]: remote recv: Host is unreachable
Sun Apr 18 09:02:05 2021 daemon.err /usr/bin/ss-redir[19152]: remote recv: Host is unreachable
Sun Apr 18 09:02:05 2021 daemon.err /usr/bin/ss-redir[19151]: remote recv: Host is unreachable
Sun Apr 18 09:02:05 2021 daemon.err /usr/bin/ss-redir[19151]: remote recv: Host is unreachable
Sun Apr 18 09:02:05 2021 daemon.err /usr/bin/ss-redir[19152]: remote recv: Host is unreachable
Sun Apr 18 09:02:05 2021 daemon.err /usr/bin/ss-redir[19153]: remote recv: Host is unreachable
Sun Apr 18 09:02:07 2021 user.notice OMR-VPS: Can't get sapin token, try later (can ping server sapin, no server API answer)
Sun Apr 18 09:02:07 2021 daemon.err /usr/bin/ss-redir[19151]: remote recv: Host is unreachable
[... cut ...]
Sun Apr 18 09:02:13 2021 daemon.err /usr/bin/ss-redir[19152]: remote recv: Host is unreachable
Sun Apr 18 09:02:13 2021 daemon.err /usr/bin/ss-redir[19150]: remote recv: Host is unreachable
Sun Apr 18 09:02:13 2021 daemon.err /usr/bin/ss-redir[19151]: remote recv: Host is unreachable
Sun Apr 18 09:02:13 2021 daemon.err /usr/bin/ss-redir[19152]: remote recv: Host is unreachable
Sun Apr 18 09:02:13 2021 daemon.err /usr/bin/ss-redir[19153]: remote recv: Host is unreachable
Sun Apr 18 09:02:14 2021 daemon.err /usr/bin/ss-redir[19152]: remote recv: Host is unreachable

Ysurac, 2021-04-18:
Alors la oui ça marche très mal.

Ysurac, 2021-04-18:
J'ai commandé un VPS, je vais tester dès que je l'aurais.

artiflo, 2021-04-18:
Merci, c'est super :)

Au passage la support dans OMR de plusieurs VPS est super pratique et surtout dans un cas comme ça.

Ysurac, 2021-04-18:
Tu as bien redémarrer le VPS après installation ?

artiflo, 2021-04-18:
Oui, je viens de le refaire au cas ou.

artiflo, 2021-04-18:
En rebasculant sur la kimsufi ce matin je constate maintenant que la connexion à le VPS kimsufi à le même problème à présent.

artiflo, 2021-04-18:
Désolé hier soir quand j'avais rebasculé sur kimsufi je n'avais pas vu que cela bagoté aussi avec kimsufi. 

Donc j'ai l'impression que mon problème c'est plutôt lié à l'ajout d'un second VPS dans OMR.

[image]

System log : 
Sun Apr 18 09:45:06 2021 daemon.info dnsmasq-dhcp[19349]: DHCPOFFER(eth0) 192.168.100.20 ac:9b:0a:f2:07:56
Sun Apr 18 09:45:06 2021 daemon.info dnsmasq-dhcp[19349]: DHCPREQUEST(eth0) 192.168.100.20 ac:9b:0a:f2:07:56
Sun Apr 18 09:45:06 2021 daemon.info dnsmasq-dhcp[19349]: DHCPACK(eth0) 192.168.100.20 ac:9b:0a:f2:07:56 TV-SONY
Sun Apr 18 09:45:07 2021 daemon.err /usr/bin/ss-redir[31928]: remote recv: Host is unreachable
Sun Apr 18 09:45:07 2021 daemon.err /usr/bin/ss-redir[31930]: remote recv: Host is unreachable
Sun Apr 18 09:45:08 2021 daemon.err /usr/bin/ss-redir[31928]: remote recv: Host is unreachable
Sun Apr 18 09:45:09 2021 daemon.err /usr/bin/ss-local[31927]: getpeername: Socket not connected
Sun Apr 18 09:45:09 2021 user.notice OMR-VPS: Can't get abricot token, try later (can ping server abricot, no server API answer)
Sun Apr 18 09:45:10 2021 daemon.err /usr/bin/ss-local[31927]: getpeername: Socket not connected
Sun Apr 18 09:45:10 2021 daemon.err /usr/bin/ss-local[31927]: getpeername: Socket not connected
Sun Apr 18 09:45:10 2021 user.notice OMR-VPS: Can't get abricot token, try later
Sun Apr 18 09:45:10 2021 user.notice OMR-VPS: Can't get abricot token, try later
Sun Apr 18 09:45:17 2021 user.notice OMR-VPS: Can't get abricot token, try later
Sun Apr 18 09:45:17 2021 user.notice OMR-VPS: Can't get abricot token, try later
Sun Apr 18 09:45:22 2021 user.notice OMR-VPS: Can't get abricot token, try later
Sun Apr 18 09:45:22 2021 user.notice OMR-VPS: Can't get abricot token, try later
Sun Apr 18 09:45:23 2021 daemon.err /usr/bin/ss-redir[31930]: remote recv: Host is unreachable
Sun Apr 18 09:45:23 2021 daemon.err /usr/bin/ss-redir[31929]: remote recv: Host is unreachable
Sun Apr 18 09:45:23 2021 daemon.err /usr/bin/ss-redir[31930]: remote recv: Host is unreachable
Sun Apr 18 09:45:23 2021 daemon.err /usr/bin/ss-redir[31928]: remote recv: Host is unreachable
Sun Apr 18 09:45:23 2021 daemon.err /usr/bin/ss-redir[31930]: remote recv: Host is unreachable
Sun Apr 18 09:45:23 2021 daemon.err /usr/bin/ss-redir[31931]: remote recv: Host is unreachable
Sun Apr 18 09:45:23 2021 daemon.err /usr/bin/ss-redir[31930]: remote recv: Host is unreachable
Sun Apr 18 09:45:23 2021 daemon.err /usr/bin/ss-redir[31928]: remote recv: Host is unreachable
Sun Apr 18 09:45:24 2021 daemon.err /usr/bin/ss-redir[31928]: remote recv: Host is unreachable
Sun Apr 18 09:45:24 2021 daemon.err /usr/bin/ss-redir[31929]: remote recv: Host is unreachable
Sun Apr 18 09:45:24 2021 daemon.info omr-tracker-ss: Shadowsocks is down (can't contact via http 104.16.1.1, 198.27.92.1, 88.191.250.176)
Sun Apr 18 09:45:24 2021 user.notice Shadowsocks: Rules DOWN
Sun Apr 18 09:45:25 2021 user.notice OMR-VPS: Can't get abricot token, try later (can ping server abricot, server API answer)
Sun Apr 18 09:45:26 2021 user.notice OMR-VPS: Can't get abricot token, try later
Sun Apr 18 09:45:26 2021 user.notice OMR-VPS: Can't get abricot token, try later
Sun Apr 18 09:45:31 2021 daemon.err /usr/bin/ss-local[31927]: getpeername: Socket not connected
Sun Apr 18 09:45:31 2021 user.notice OMR-VPS: Can't get abricot token, try later
Sun Apr 18 09:45:31 2021 user.notice OMR-VPS: Can't get abricot token, try later
Sun Apr 18 09:45:36 2021 daemon.err /usr/bin/ss-local[31927]: getpeername: Socket not connected
Sun Apr 18 09:45:39 2021 user.notice OMR-VPS: Can't get abricot token, try later (can ping server abricot, server API answer)
Sun Apr 18 09:45:41 2021 user.notice OMR-VPS: Can't get abricot token, try later
Sun Apr 18 09:45:41 2021 daemon.err /usr/bin/ss-local[31927]: getpeername: Socket not connected
Sun Apr 18 09:45:43 2021 user.notice OMR-VPS: Can't get abricot token, try later
Sun Apr 18 09:45:43 2021 user.notice OMR-VPS: Can't get abricot token, try later
Sun Apr 18 09:45:45 2021 user.notice OMR-VPS: Can't get abricot token, try later
Sun Apr 18 09:45:46 2021 daemon.err /usr/bin/ss-local[31927]: getpeername: Socket not connected
Sun Apr 18 09:45:51 2021 daemon.info omr-tracker-ss: Shadowsocks is up (can contact via http 1.0.0.1)
Sun Apr 18 09:45:51 2021 daemon.info omr-tracker-ss: Reload Shadowsocks rules
Sun Apr 18 09:45:51 2021 user.notice Shadowsocks: Rules UP
Sun Apr 18 09:45:51 2021 user.notice Shadowsocks: Reload omr-bypass rules
Sun Apr 18 09:45:51 2021 user.notice omr-bypass: Starting OMR-ByPass...
Sun Apr 18 09:45:52 2021 user.notice OMR-VPS: Can't get abricot token, try later
Sun Apr 18 09:45:53 2021 user.notice omr-bypass: Reload dnsmasq...
Sun Apr 18 09:45:53 2021 daemon.info dnsmasq[19349]: read /etc/hosts - 4 addresses
Sun Apr 18 09:45:53 2021 daemon.info dnsmasq[19349]: read /tmp/hosts/dhcp.cfg01411c - 9 addresses
Sun Apr 18 09:45:53 2021 daemon.info dnsmasq-dhcp[19349]: read /etc/ethers - 0 addresses
Sun Apr 18 09:45:53 2021 user.notice omr-bypass: OMR-ByPass is running
Sun Apr 18 09:45:53 2021 daemon.err /usr/bin/ss-redir[31931]: remote recv: Host is unreachable
Sun Apr 18 09:45:53 2021 daemon.err /usr/bin/ss-local[31927]: getpeername: Socket not connected
Sun Apr 18 09:45:55 2021 daemon.err /usr/bin/ss-redir[31929]: remote recv: Host is unreachable
Sun Apr 18 09:45:57 2021 user.notice OMR-VPS: Can't get abricot token, try later (can ping server abricot, no server API answer)
[... cut ...]
Sun Apr 18 09:47:24 2021 daemon.err /usr/bin/ss-redir[31930]: remote recv: Host is unreachable
Sun Apr 18 09:47:26 2021 user.notice OMR-VPS: Can't get abricot token, try later (can ping server abricot, no server API answer)
Sun Apr 18 09:47:28 2021 daemon.err /usr/bin/ss-local[31927]: getpeername: Socket not connected
Sun Apr 18 09:47:28 2021 daemon.info omr-tracker-ss: Shadowsocks is down (can't contact via http 198.27.92.1, 88.191.250.176, 151.101.129.164)
Sun Apr 18 09:47:28 2021 user.notice Shadowsocks: Rules DOWN

Ysurac, 2021-04-18:
Y'en a un en maitre et pas l'autre ou alors l'autre a été désactivé ?

artiflo, 2021-04-18:
Il y a un maitre (kimsufi actuellement) et le sapinet en désactivé.

Et ce matin quand j'ai posté c'était sapinet en maitre et kimsufi en désactivé.

Ysurac, 2021-04-18:
J'ai essayé la même configuration, et j'arrive pas à reproduire le soucis. Je suis cependant en v0.58 beta et j'ai peut être corrigé ce bug.

artiflo, 2021-04-18:
Je viens de supprime le VPS kimsufi afin d'avoir que le VPS sapinette sur OMR et je n'ai plus aucun problème de déconnexion la connection est stable.

Désolé pour la fausse piste, c'est donc plutôt un problème avec la version 0.57.3 et le multi VPS.

Tu veux que je test une beta pour confirmer/infirmer l'hypothèse ?

Ysurac, 2021-04-18:
J'ai testé sur la 0.58 beta en multi VPS, donc je dirais que c'est bon.
Après tu peux toujours tester aussi.

artiflo, 2021-04-18:
Si c'est bon pour toi, je vais attendre la version stable de la 0.58

Désolé pour dérangement, merci pour l'aide.

closed 2021-04-18 by artiflo