Ysurac/openmptcprouter #1759: not an issue just a question about IPV6 et Milkywan

vinceducat, 2021-04-18:
Hello from this week Hugues from milky wan send those information

_NAT64 - Passez à IPv6 !
 
J’ai mis en place une infra de NAT64. Ce protocole permet a des clients en IPv6 Only d’accéder a des sites en IPv4 sans avoir besoin d’IPv4. Du coup, pour nos VM sans IPv4, c’est bien pratique. Vous pouvez d’ores et déjà l’utiliser, il suffit de configurer nos DNS : 
- 130.117.11.11
- 2a0b:cbc0:42::42
 
Par ailleurs, si vous avez une VM OpenMPTCPRouter, depuis la migration, vous disposez d’une très grande plage d’IPv6 dédiée /48 qui est compatible avec OpenMPTCPRouter, je vous invite très fortement à activer IPv6, c’est quelque chose de très important chez nous !_

could give me some information to check if my OMR is well config

thanks

Ysurac, 2021-04-18:
You need to have IPv6 on all WANs to make this work. This will be supported in v0.58.

vinceducat, 2021-04-18:
ok thanks , i will wait ;-)

closed 2021-04-18 by vinceducat