Ysurac/openmptcprouter #1760: Continuous video stream

kverkind, 2021-04-18:
I have 2 WAN's. It is a given that both WAN's go down for a little time after a period of time. At all times there is one WAN available. Would it be possible to make use of the mptcprouter to accomplish a continuous stream?

Adorfer, 2021-04-18:
what is a video stream? rtmp? rtsp? You are serving the steream or the client on omr lan is a client watching? 
(hint: use a template...)

kverkind, 2021-04-19:
To be honest I have no idea if it is rtmp or rtsp.
Data is provided by external party (I am located in Belgium) examples are https://www.telenettv.be/en.html, https://www.vrt.be/vrtnu/livestream/ 
The data is casted to a chromecast device. Same challange is there for streaming radio (examples https://radioplayer.be/nl/discover/a-z/g and https://tunein.com/)
I use Edgerouter X in the beta version as documented by Ysurac. 
The router reports all to green and I can clearly see increase in speed. (I believe we can conclude all is working then).
My prime goal is not speed but redundancy knowing that each connection will go down for a limited period of time (but not at same time)

kverkind, 2021-04-25:
Closed. Redundant mode

closed 2021-04-25 by kverkind