Ysurac/openmptcprouter #1761: openmptcprouter support for 19.07 as a set of packages

wicadmin, 2021-04-19:
I build my own x86_64 OpenWRT as I have several customizations. Would it be possible to setup openmptcprouter to support this instead of master?

Ysurac, 2021-04-19:
No. OpenWrt 19.04 use a 4.14 kernel, OpenMPTCProuter/OpenWrt 21 and master branch are based on a 5.4 kernel.
OpenMPTCProuter also need some updated packages and arch support not available in 19.04.

wicadmin, 2021-04-19:
ok, got it. 

1) Are you working to incorporate this into upstream post 19.07?
2) Can those packages go into the OpenWRT extension build system and be built using their SDK and ImageBuilder?

If not, where would one plug into this build script custom compiles and adding pre-compiled packages?

Ysurac, 2021-04-19:
1. Not yet. I will wait that OpenWrt can use upstreamed kernel MPTCP (and that I can use upstreamed MPTCP), and in all case only part will be available I think.
2. no
OpenWrt packages are compiled for OpenMPTCProuter, for other packages you will need to add and compile them yourself, like for normal OpenWrt.

wicadmin, 2021-04-19:
@Ysurac  - how long does it take the build a x86_64 with the build script?

I've avoided building packages to make an images and used prepackaged ones to cut down on the build time.

Ysurac, 2021-04-19:
3h15 with github actions.

wicadmin, 2021-04-19:
ok, maybe you can provide some guidance on how I can go about leveraging OMR. I build ModemManager for OpenWRT nightly and would now like to do the same to work with OMR. Given the fast pace of modems and 5G modems coming to the market, MM is also understand constant development to accomodate them. I'll like to at least be able to have these build for OMR where the installation would be compatible.

https://gitlab.freedesktop.org/wicadmin/openwrt-modemmanager-packages

wicadmin, 2021-04-23:
@Ysurac - For OpenWRT snapshot, how can I correlate which snapshot version OMR used in its version build?

closed 2021-07-28 by github-actions[bot]