Ysurac/openmptcprouter #1762: port forwarding problem

dtony83, 2021-04-19:
Good day,

I followed the documentation at https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding to open successfully port 80 or 22 on the router but if I try to open for example port 21 or another custom one, tools like https://www.yougetsignal.com/tools/open-ports/  or a telnet command say that the port is closed.

Any suggestion? Thanks.

Ysurac, 2021-04-19:
All ports work the same way, so if you can open port 80 then this will work for port 21.
You can check Debug section of this wiki page.

dtony83, 2021-04-21:
> All ports work the same way, so if you can open port 80 then this will work for port 21.
> You can check Debug section of this wiki page.

thanks for the reply but the issue is still ongoing; [image]

port 80 open, port 3389 (or another one except 443) closed

i missed only the tcpdump step

closed 2021-07-25 by github-actions[bot]