Ysurac/openmptcprouter #1763: Can't contact Server Admin Script

ericguichard, 2021-04-19:
This error message pops from time to time. I have 3 wireless connections. Where should I start looking?

Ysurac, 2021-04-19:
Wireless connections doesn't have a stable latency and can loose packets. Status page only wait 1s to check API server script, so if the error is not always here then there is no problem.
You can still check State->System log about "OMR-VPS" if it say that it can't connect API.

ericguichard, 2021-04-19:
I get:
OMR-VPS: Can't get vps token, try later (can't ping server vps, server API answer)

Ysurac, 2021-04-19:
It look likes some packets are lost. When status page can't contact server API it ask OMR-VPS script to try himself. If all is working it's not a big problem.

ericguichard, 2021-04-19:
Thank you @Ysurac. For taking the time to respond to this query and above all for developing a great piece of software.
My connection is very slow today, but it is probably due to the saturation of the mobile networks.

closed 2021-04-19 by ericguichard