Ysurac/openmptcprouter #1764: Remote access to the OpenMPTCProuter interface not possible.

Ajowan84, 2021-04-19:
Hi, 

## Expected Behavior
Connect remotely to the OpenMPTCProuter interface 

## Current Behavior
Cannot connect to the OpenMPTCProuter interface.
On the other hand, I connect to the interface of my cameras, my NAS and my home automation without problem.
My forwarding ports are all configured the same ... 

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.57.3 r0+15225-bfc433efd4
  - OpenMPTCProuter VPS version: 0.1024 5.4.86-mptcp
  - OpenMPTCProuter platform: ESXi 

Thank

flimo44, 2021-04-19:
Hello ,
For me Problem is the certicate used  with openmptcp. I use reverse proxy... Solution change openMPTC certicate....

Ajowan84, 2021-04-19:
@Ysurac It's working good with 443 port.
Thank you for this quick response