Ysurac/openmptcprouter #1765: Makes the internet slower, unreliable.

3ricj, 2021-04-20:
## Expected Behavior

My expectation is that this would take 3 mostly reliable and fast internet connections  (Cell based) and turn them into a much faster, more reliable internet connection.  

## Current Behavior

When I use openMPTCPRouter on three usb cell modems (various brands, iphone, Novatel, etc) I have been able to make it connect and merge the net connections to each other.  However, the combined internet is SLOWER than any one of the cell devices, and often extremely slow to make new connections (new tcp connections take .. sometimes 5-6 seconds to establish).  

I am using 3 different cell providers on different frequencies.   They might share a tower, but not radios -- regardless, ,if I test them all at once I do not see any reduced speed (provided I keep them a few cm apart).

1.  Verizon UW 5g (normally:  200+mbit/second, peaks at 1000+)
2. ATT/iphone 4g LTE:  20-50mbit
3. Tmobile 5G hotspot (usb):  50-300mbit)

Combined I get... maybe 15-20mbit.  Never more.  I'm using a PI 4b, powered usb hub, and the latest build.  I've tried a number of settings (turning off encryption, UDP/tcp/etc) and can't seem to move above 20mbit no matter what I do.  

My VPS is 80ms away.   I can download from my VPS at the above speeds on each modem.   If I run a speed test on each cell device at the same time, they operate at full speed -- so the hardware (pi 4/USB) seems capable of this speed.  

My uplink is gig ethernet to my test workstation (no wifi, yet).  My VPS is hosted by google cloud, debian 10 using your script. That part was easy (once I got the firewall open).

## Possible Solution

I'm not sure what else I can try.  This produce so far has turned fast reliable internet connections into slow unreliable internet connections.  

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Context (Environment)

I am planning a trip to rural america, where only one of these cell devices may work at any given time.   Right now just testing in a lab. 

## Specifications

  - OpenMPTCProuter version: 0.57.3
  - OpenMPTCProuter VPS version: 0.1024 5.4.86 mptcp
  - OpenMPTCProuter platform: RPi4b 8gb

Ysurac, 2021-04-20:
Status page is green ? Can I get a screenshot ?
Check also MPTCP support of each WAN in Network->MPTCP.

3ricj, 2021-04-20:
[image]

I'm still screwing around with settings; here I disabled one of the interfaces to make testing easier, and i disabled the VPN.  The MPTCP test reports "BAD GATEWAY" on one of my links (verizon0).  I'm not sure what that means, traffic is working..

Ysurac, 2021-04-20:
One interface must be set as master.

3ricj, 2021-04-20:
So one "master" and .. what should the others be?

3ricj, 2021-04-21:
I've gotten my connections a bit more stable -- new image to clear out old wan stuff which I think was causing confusion.  However, my one core problem remains:  When using openmptcprouter, new tcp connections take forever (sometimes not working at all). . For example, going to speedtest.net it will sometimes never start a test telling me my firewall is misconfigured.  Clicking refresh and restarting the router a few times normally makes it work again, without even changing any settings.  I'm not sure what else to try or do.   There are some sites on the internet I just can't reach at all.  It's pretty frustrating.   Here's my whole configuration, minus hopefully all of the passwords.  

All router settings
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.d/resolv.conf.auto'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].ednspacket_max='1232'
dhcp.@dnsmasq[0].server='127.0.0.1#5353' '/lan/' '/use-application-dns.net/'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.ra_slaac='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.lan.force='1'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].RootPasswordAuth='on'
dropbear.@dropbear[0].Port='22'
dscp.@classify[0]=classify
dscp.@classify[0].direction='both'
dscp.@classify[0].proto='icmp'
dscp.@classify[0].class='cs5'
dscp.@classify[0].comment='ICMP'
dscp.@classify[1]=classify
dscp.@classify[1].direction='both'
dscp.@classify[1].proto='udp'
dscp.@classify[1].class='cs5'
dscp.@classify[1].src_port='53,123,5353'
dscp.@classify[1].comment='DNS udp and NTP'
dscp.@classify[2]=classify
dscp.@classify[2].direction='both'
dscp.@classify[2].proto='tcp'
dscp.@classify[2].class='cs5'
dscp.@classify[2].src_port='53,5353'
dscp.@classify[2].comment='DNS tcp'
dscp.@classify[3]=classify
dscp.@classify[3].direction='both'
dscp.@classify[3].proto='tcp'
dscp.@classify[3].class='cs6'
dscp.@classify[3].dest_port='65001,65301,65011'
dscp.@classify[3].comment='OMR vpn'
dscp.@classify[4]=classify
dscp.@classify[4].direction='both'
[... 1607 lines cut ...]
v2ray.omrroutingo=routing_rule
v2ray.omrroutingo.type='field'
v2ray.omrroutingo.inbound_tag='omrbridge'
v2ray.omrroutingo.outbound_tag='out'
v2ray.main_routing=routing
v2ray.main_routing.enabled='1'
v2ray.main_routing.rules='omrrouting' 'omrroutingo'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='wan1' 'wan2' 'wlan0'
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.channel='36'
wireless.radio0.hwmode='11a'
wireless.radio0.path='platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio0.htmode='VHT80'
wireless.radio0.cell_density='0'
wireless.radio0.country='US'
wireless.wifinet2=wifi-iface
wireless.wifinet2.device='radio0'
wireless.wifinet2.mode='sta'
wireless.wifinet2.network='wan3'
wireless.wifinet2.encryption='psk2'
wireless.wifinet2.key='77499ee3'
wireless.wifinet2.ssid='Verizon-M2100-3884_disable'
Powered by openmptcprouter v0.57.3 r0+15225-bfc433efd4

3ricj, 2021-04-21:
Now, if I run the "MPTCP support check", it thinks for a long time and finally says: 

"Bad Gateway
The process did not produce any response"

Ysurac, 2021-04-21:
You have only one interface with multipath enabled : wan4 that is set as master.
You seems to use v2ray and no VPN.
You seems to use some USB modem, do you use an USB powered hub ?
Did you check State->System log ?

3ricj, 2021-04-21:
I have wan2 and wan4 currently enabled -- they are both fast and reliable internet connections.  One is master, one is backup. 
I tried changing to v2ray, but it doesn't seem to make any difference.  I disabled VPN because? I don't think I need it? you tell me?  I don't need encryption. Will it help openmptcp? Then I will use it. But I was worried that it was causing overhead.  
Yes, they are USB modems - self powered with battery, but going via a powered hub makes no difference . Again, the hardware works great because I can run other software on the same system and get exception performance.  (over 1gb/second using for example  bittorrent on load balancing openwrt.  Same PI4b, same usb modems). 

To clarify:  The same hardware:  PI4b, using normal openwrt works extremely fast and is reliable.   However, it doesn't bind multiple net connections.  

There are no errors on any logs on the system.  Just lots of this: 

"
Wed Apr 21 13:46:26 2021 user.notice post-tracking-post-tracking: Set backup ip route replace default scope global  nexthop via 192.168.1.1 dev tmobile0 weight 1"

Ysurac, 2021-04-21:
The interface set as backup is not used while other interface(s) is(are) up.
Proxy are used for TCP (and UDP for v2ray), and VPN for anything else (ICMP, UDP,...). So in normal usage, both are needed.

3ricj, 2021-04-21:
.. well, it was using the "backup" wan for data.. 

I have turned on the proxy again.  The documentation on this project is extremely poor.  An example of a working configuration would be extremely helpful.

Ysurac, 2021-04-21:
Project is still in beta, and for now I prefer to work on code.
Default configuration is working.

3ricj, 2021-04-21:
Well, the default configuration isn't working for me.  I understand this is in beta, but projects without documentation or working code isn't very helpful.  I see this bug tracker is full of other people who can't seem to make it work either.  I'm willing to spend some time trying to make this work, and even write documentation if we make some progress: i'm happy to be a contributor.  But if your perspective is that the only thing you want to do is write code: Then it's hopeless.  You have actual customers who want to use your stuff, that's a good problem to have!

hikey, 2021-04-21:
3ricj, relax a bit man.
Projects are there to explore and make it better, you talk about customers you did not buy anything right ?
So explore and see what work for you and what does not, and if you don't like it there are always other possibilities and even commercial ones where you really become a customer.

hairy-dog, 2021-04-21:
I agree that the documentation is not good (remember what you paid for this), but the default configuration seems fine. 
You need one wan to be master, the others to be enabled. 
Don't switch anything else off or on: leave everything at default until you get things working and you get a feel for how things work.
And of course, remember the golden rule: only change one thing at a time.

3ricj, 2021-04-21:
> 3ricj, relax a bit man.
> Projects are there to explore and make it better, you talk about customers you did not buy anything right ?
> So explore and see what work for you and what does not, and if you [...]

I am relaxed. :)  I'm just trying to make it actually operate.  And yes, I'm a customer, that's how open source works.  I'm paying with my time to try and make a thing work.  We'll get there.   If there was a commercial solution here I would have paid for it already.

Ysurac, 2021-04-21:
You are not a customer and it's not how open source works.
There is already some commercial solution and I think you should use them : OverTheBox, speedify,...

closed 2021-04-21 by Ysurac

3ricj, 2021-04-21:
Well, what's clear is that I'm no longer a customer.  Best of luck to you.

3ricj, 2021-04-21:
It's neat how when someone has trouble with his bad code, he just gets rude and closes a ticket.   Ah well, this is the way of open source, he's got the right idea to build community!  :)

hairy-dog, 2021-04-21:
Erm, I think you'll find its the other way round.

hikey, 2021-04-21:
I really think you are not the type of person who contribute on a open source project , like you say: you want something what is completly documentated, and where you can behave like a customer. Here the admin spent hundred of hours to make a working solution where you can even adjust it to your needs and can adjust the code.
He own us nothing, we only can be glad he put so much time in it to make it work.
For me it work without any issues, the Wiki is a good starting point.
https://www.youtube.com/watch?v=mYYoIDCWszo

3ricj, 2021-04-21:
I am extremely appreciative of open source software, I use, contribute and write much of it myself.  Running OSS projects is hard work. It's true. But an attitude of "no I'm not going to help you or write documentation or tell you how to make my stuff work because I'd rather just write code" isn't very helpful.  good bye!