Ysurac/openmptcprouter #1766: Run openMPTCProuter on a laptop for testing?

oettinger31, 2021-04-21:
## Expected Behavior
Installation of openmptcprouter on a HP 635 Notebook PC with AMD E-350 (2x1,6GHz), single LAN port for testing.

## Current Behavior
I have not yet done anything with openmptcprouter. Can I install openmptcprouter on my above mentioned laptop for testing before buying dedicated hardware for that? Is the single Ethernet port an issue?

## Specifications

  - OpenMPTCProuter version: none
  - OpenMPTCProuter VPS version: none
  - OpenMPTCProuter platform: x86_64

Ysurac, 2021-04-21:
You can install it on a laptop. You need to dd the image to hard disk or to an USB key.
Single Ethernet port is not an issue, MacVLAN is used to create some "virtual" interfaces.

vinceducat, 2021-04-21:
Hello i have OMR with laptop and usb key work nice

oettinger31, 2021-04-22:
Thank, got the laptop working as OMR :)

closed 2021-04-22 by oettinger31