Ysurac/openmptcprouter #1767: openmptcprouter and Starlink

oettinger31, 2021-04-21:
I have recently started using Starlink. Parallely I have a Telekom Magenta Hybrid connection running. I want to combine both with openmptcprouter.
1. The Starlink modem in dishy has a fixed IP 192.168.100.1 that conflicts with the openmptcprouter IP. The Starlink modem IP cannot be changed. Can the openmptcprouter IP be changed?
2. The Starlink router has a fixed IP 192.168.1.1 that cannot be changed. The Wiki says that the Router IPs shall be changed to 192.168.10.1 and 192.168.11.1. That is not possible with Starlink. Can I instead configure Starlink with 192.168.1.1 and insert that IP in the wizard for WAN1 or will that cause problems?
3. Can I instead remove the Starlink router from the setup and connect dishy (with the modem) directly to openmptcprouter and configure WAN1 with the modem IP 192.168.100.1 as gateway? Problem 1 would remain.

Ysurac, 2021-04-21:
1. Yes OpenMPTCProuter IP can be changed.
2. Routers IPs in doc are example, so no problem
3. No idea, I don't know how starlink work and if the provided router is needed or not.

oettinger31, 2021-04-21:
Thank you. Is there a guide on how to change openmptcp ip? Or will I intuitively find it in the wizard?
Starlink router is not needed, you can attach any router so I will just try whats working best.

Ysurac, 2021-04-21:
You can change OpenMPTCProuter IP via the wizard or via Network->Interfaces, edit "LAN" interface.

hairy-dog, 2021-04-21:
The wiki examples are just that: examples. I have the OMR IP as 192.168.0.201 and my wan connections are 192.168.8.1, 192.168.10.1, 192.168.12.1 and 192.168.11.1 
All you need to do is to connect to each of the wan routers in turn and give each a fixed IP address and switch off their DHCP service.
Then on the OMR settings, add the WAN connections, for example putting 192.168.10.2 in the address box, and 192.168.10.1 in the gateway box. I don't think the docs make that clear.

docderwood, 2021-04-21:
> * Yes OpenMPTCProuter IP can be changed.
> * Routers IPs in doc are example, so no problem
> * No idea, I don't know how starlink work and if the provided router is needed or not.

If I want to use two Starlinks but their routers only allow them to have 192.168.100.1 as their address, will OpenMPTCP allow me use them together?

hairy-dog, 2021-04-22:
If the starlinks both have to have that up address, you won't be able to connect both to the same network. This is not an openmptcproter restriction. 
I think you will need to add an additional router between the second Starlink and the rest of the network.
I decided not to order one starlink no because the cost was too high. Two would cost twice as much!

oettinger31, 2021-04-22:
[image]
@Ysurac 
I have trouble with the Starlink.
If I don't use a VPS and connect both ISPs to OMR, always Hybrid is the outgoing connection. If I disconnect Hybrid, Starlink takes over. I wish it was the other way round.
If I connect the VPS and have both ISPs the speed is like only Hybrid is active. So I disconnect Hybrid and have only Starlink attached and then I have no internet at all.
I tried both using Starlink router or connect dishy with the modem directly (of course I changed the IPs accordingly) but the behavior is the same.

@docderwood This would be my idea for you, if you have Starlink running with OMR: Connect one dishy directly with IP 192.168.100.1 and the other Starlink with the router with 192.168.1.1

Ysurac, 2021-04-22:
You can check in State->System log what is happening when there is only starlink.

oettinger31, 2021-04-22:
I did the following:
- Start OMR Laptop
- Enable VPS
- Enable VPN (I enabled VPN in the session before manually, so the screenshot in above comment is not correct).
- Internet Test positive, DL speed looks like hybrid only
- Systemstate Screenshot
- Disconnect Hybrid, Starlink-only connected
- Screenshot
- Internet test, negative, no internet at all
- Copy syslog

[systemlog_starlink-only.txt](https://github.com/Ysurac/openmptcprouter/files/6357633/systemlog_starlink-only.txt)
[image]
[image]

Ysurac, 2021-04-22:
Internet seems to be working from the router (else it can't get VPS IP, load average,...)
IPv6 is disabled for hybrid connection ? DHCP too ? Are you sure your client computer use OMR as gateway and DNS ?

oettinger31, 2021-04-22:
DHCP at hybrid router is off, IPv6 is enabled on Hybrid, I will disable it.
I have a static ip at the client computer with the OMR as gateway.
With the Starlink router it's not possible to disable DHCP but the direct connection to the modem gives the same behavior. 
I will check my public IP if it is from the VPS or from Hybrid IPS.

docderwood, 2021-04-22:
> [image]
> @Ysurac
> I have trouble with the Starlink.
> If I don't use a VPS and connect both ISPs to OMR, always Hybrid is the outgoing connection. If I disconnect Hybrid, Starlink takes over. I wish it was the other way round.
> If I connect the VPS and have both ISPs the speed is like only Hybrid is active. So I disconnect Hybrid and have only Starlink attached and then I have no internet at all. You see it where Ethernet in Task Manager goes down to 0 in the attached screenshot, there I disconnected the Hybrid line.
> I tried both using Starlink router or connect dishy with the modem directly (of course I changed the IPs accordingly) but the behavior is the same.
> 
> @docderwood This would be my idea for you, if you have Starlink running with OMR: Connect one dishy directly with IP 192.168.100.1 and the other Starlink with the router with 192.168.1.1

Thank you for the tip!!. 
...my Starlink will be here 4/26, I will be unable to install for a couple of weeks :(

bdruth, 2021-04-22:
fwiw - 192.168.100.1 is _**Dishy's**_ address. If you connect _directly_ to Dishy, you will get a CGNAT address from the Starlink network - that address will **not** be in the 192.168.100.x subnet - e.g. mine is currently `100.68.0.x`. If you want to hook up two Starlink dishes, you can do as @docderwood suggests, but you won't be using `192.168.100.1` ... also, unless you do some magical routing, you'll only be able to connect to the status page of _one_ Dishy ... `192.168.100.1` is static to the dish, every dish uses the same IP, irrespective of what router sits in front of it. And even then, you'll need to hard-code a route in OMR to which Dishy you want `192.168.100.1` to route to.

Fwiw, I'm pretty sure you can change the LAN subnet the Starlink router is offering up to be whatever you want it to be (that doesn't impact the `192.168.100.1`, as it's upstream of the router. e.g. you could setup the Starlink router on one dish to use `192.168.1.0/24`, a 2nd to be `192.168.2.0/24`, and your OMR can use `192.168.0.0/24`.

oettinger31, 2021-04-23:
@bdruth
Does that mean I have to put the 192.168.100.1 as gateway (direct connection OMR to dishy w/o Starlink router) and put the 100.73.0.x that's shown in the Starlink App as IPv4 Adress for the WAN1?

oettinger31, 2021-04-23:
Ok, did some further testing. Got Starlink finally running, Starlink router is connected (191.168.1.1 with DHCP because no configuration possible in Starlink router). So i connected the VPS, it works and my IP in the internet is the VPS IP.

Connected to Hybrid and again only the speed of hybrid. So I deactivated IPv6 on my client PC and TADAAAAA, it works now with bonded speed (down only 100Mbps max due to my Laptop running as OMR with only 100M Ethernet port + Client PC on powerline is braking) but upload is clearly more or less both speeds bonded.

Now my problem is only, that IPv6 cannot be deactivated in the Hybrid router and using another router for Telekom Hybrid is definitely not possible (only the two available Telekom router can handle Hybrid).
On a PC this is not an issue, there I can deactivate IPv6 in the adapter settings. But Smarthones or my TV don't have the option :(

Ok, therefore I can say, that the issue can be closed, OMR is working as intended! Thank you for your support!

Shitty hybrid router is out of scope here.

[image]

[image]

Ysurac, 2021-04-23:
To disable IPv6 you can use a router with multiple ports and use a port for each wan router, so no IPv6 conflict on network.
You can also use another router between OpenMPTCProuter and the Hybrid router, like an OpenWrt on a RPI and disable IPv6 support.

oettinger31, 2021-04-23:
OMR is currently a laptop:
- disconnect Hybrid router ethernet from network
- connect laptop in OMR with Hybrid Wifi

Possible? Than the hybrid has no chance to give IPv6 to my other network devices

Ysurac, 2021-04-23:
Using Wifi as source is not a good idea: some packets are lost and OpenMPTCProuter doesn't like that.
You can use an USB to Ethernet adapter ;)

oettinger31, 2021-04-23:
regarding @docderwood question with two starlink routers with the same IP maybe its a possibility to use 2 different ethernet ports, than the same IPs shouldn't matter?

docderwood, 2021-04-23:
> regarding @docderwood question with two starlink routers with the same IP maybe its a possibility to use 2 different ethernet ports, than the same IPs shouldn't matter?

This is my plan, once I get it installed: Two separate physical WAN ethernet ports for Starlink along with two separate LTE connections. 

The only issue I see is accessing the Starlink app for each individual dish, since the OPenMPTC router will have the same IP address that can't be changed (yet, maybe starling will allow in the future)

oettinger31, 2021-05-10:
Last update from my side:
- Added two Usb2Eth devices for the WANs. So now no physical connection between my LAN and WANs. IPv6 of Hybrid therefore isn't an issue anymore, I can leave all client devices on Auto istead of setting everything manually.
- As USB 2.0 of my laptop was the bottleneck in the whole system (for the additional ethernet ports), I changed to a suitable Mini PC (Intel Pentium Silver J5005, 4 GB RAM) with USB 3
- If for Starlink WAN DHCP client is set, both the connection to the Starlink router or the Dish work. With direct connection to dish and static routes, the Starlink app with status page work.
- I changed the VPS from vultr to Hetzner. IPv6 is working better now and Hetzner is not blacklisted as Proxy/VPN for Amazon Prime video in Germany ;)

Thank's for the great support!