Ysurac/openmptcprouter #1769: Help with OMR install

juavos00, 2021-04-22:
Yes first timer here and I’m having an issue getting OMR installed on a qotom Q190G4N-S07 I picked up from Amazon. Long story short we used a usb thumb drive to load up Linux and downloaded the OMR image. Used etcher on the qotom to flash it over the ssd and now every time it boots it freezes after running the startup script. I cannot access the web GUI using the address 192.168.100.1 as it says network host unreachable. Please tell me what I need to do in order to resolve this! Or is this unit a hunk of junk and I need to send it back?

thanks

Ysurac, 2021-04-22:
Please use the template when opening an issue.
Are you sure it freeze ? Press enter and you should see login prompt.
To access the GUI you should get a DHCP address from OMR (if there is no DHCP server on the network) or you can set an IP in 192.168.100.x subnet on your client.

If it's still not working, try to add a screen and keyboard and after login (root user with no password) use ```ip a``` to show network config.

juavos00, 2021-04-22:
Yes, I have a keyboard and monitor on the qotom and it blinks a cursor for a few seconds then stops. Can’t type anything afterwards which would tell me it is frozen. Today I’m going to try erasing the ssd and reinstall OMR image directly

Ysurac, 2021-04-22:
What is the used image ?

juavos00, 2021-04-22:
https://download.openmptcprouter.com/release/v0.57.3/x86_64/targets/x86/64/openmptcprouter-v0.57.3-r0+15225-bfc433efd4-x86-64-generic-ext4-combined-efi.img.gz

Ysurac, 2021-04-22:
Why EFI ? It allow only UEFI boot ?

juavos00, 2021-04-22:
I tried this one via usb and it did the same thing

https://download.openmptcprouter.com/release/v0.57.3/x86_64/targets/x86/64/openmptcprouter-v0.57.3-r0+15225-bfc433efd4-x86-64-generic-ext4-combined.img.gz

Ysurac, 2021-04-22:
Should work.
You dd the decompressed image to the whole disk/key ?

juavos00, 2021-04-22:
I used etcher to flash the usb with the supplied image. Was there a step I missed in the installation? I’m getting an msata to USB adapter today so that I can plug the ssd into my PC to erase and start over

juavos00, 2021-04-22:
I did notice that when I used etcher on my pc to create the usb stick, etcher created a smaller partition on the drive. Is there a setting or something to write the full disk of the image?

juavos00, 2021-04-23:
Switch hardware to that of a raspberry pi 4 using your image. Works great!

closed 2021-04-23 by juavos00

hi10bha, 2021-08-17:
Hi, can u elaborate I need to do the same thing, I have sata ssd and I need to install my our on it, can u pls help