Ysurac/openmptcprouter #1770: Port forwarding not working

jamesho451, 2021-04-22:
## Expected Behavior
I set the port forwarding rule in OMR and the port should be open

## Current Behavior
The port is not open even after I go to /etc/shorewall/rules to add the following line: ACCEPT          net             $FW(also tried vpn for this parameter)             tcp     54666 and do shorewall reload after that. I was actually trying to open udp port 5060 for VOIP, I used tcp port 54666 to test because I can easily check it with online port checker, and when using port checker port 65222 for ssh is open. I configure the port forwarding on OMR with source zone of wan(also tried vpn). 

## Specifications

  - OpenMPTCProuter version: 5.4.86-mptcp 0.1024
  - OpenMPTCProuter VPS version: 5.4.86-mptcp 0.1024
  - OpenMPTCProuter platform: x86_64

Ysurac, 2021-04-22:
To open port there is a doc: https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding and if you have issue you can check debug section of the page to help you find where is the issue.

jamesho451, 2021-04-22:
Packets are flowing in when I set the source zone of my port forwarding rule to vpn. However, the online port checker still says port closed on port 54666. On 65222 the port checker says the post is open. How can I make 54666 open?

jamesho451, 2021-04-22:
> To open port there is a doc: https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding and if you have issue you can check debug section of the page to help you find where is the issue.

I was following that before, I can see that the rule is added to /etc/shorewall/rules and everything in debug checks out including the tcpdump.

Ysurac, 2021-04-22:
In this case this should be working. Do you have something that listen to port 54666 ? Online scanner only detect open ports when something listen to it (you can test with telnet also).

jamesho451, 2021-04-22:
> In this case this should be working. Do you have something that listen to port 54666 ? Online scanner only detect open ports when something listen to it (you can test with telnet also).

I don't have anything running on that port. I'm not very knowledgeable about these stuff, thank you for the information. With the packets flowing in I'd assume it's working now.

closed 2021-04-22 by jamesho451