Ysurac/openmptcprouter #1772: Bonding doesn't work as expected

SgtSilvergreen, 2021-04-23:
## Expected Behavior
My Ubiquiti EdgeRouter X, which has OMR installed, should be able to connect to the VPS. I connect a DSL and an LTE connection, both have about 50 Mbit/s. So I should be able to achieve a bonding speed of about 90 - 100 Mbit/s (already proven by Speedify).

## Current Behavior

The ER-X connects to the VPS as desired, but I only achieve a speed of about 20 - 40 Mbit/s.
A speed test shows the following results:

root@OpenMPTCProuter:~# omr-test-speed
Select best test server...
host: speedtest.frankfurt.linode.com - ping: 20
host: speedtest.tokyo2.linode.com - ping: 272
host: speedtest.singapore.linode.com - ping: 191
host: speedtest.newark.linode.com - ping: 105
host: speedtest.atlanta.linode.com - ping: 120
host: speedtest.dallas.linode.com - ping: 142
host: speedtest.fremont.linode.com - ping: 169
host: speedtest.tele2.net - ping: 19
host: proof.ovh.net - ping: 27
host: speed.hetzner.de - ping: 15
host: ipv4.bouygues.testdebit.info - ping: 29
host: par.download.datapacket.com - ping: 28
host: nyc.download.datapacket.com - ping: 109
host: ams.download.datapacket.com - ping: 34
host: fra.download.datapacket.com - ping: 19
host: lon.download.datapacket.com - ping: 32
host: mad.download.datapacket.com - ping: 46
host: prg.download.datapacket.com - ping: 26
host: sto.download.datapacket.com - ping: 42
host: vie.download.datapacket.com - ping: 27
host: war.download.datapacket.com - ping: 40
host: atl.download.datapacket.com - ping: 124
host: chi.download.datapacket.com - ping: 121
host: lax.download.datapacket.com - ping: 164
host: mia.download.datapacket.com - ping: 139
host: nyc.download.datapacket.com - ping: 108
Best server is https://speed.hetzner.de/10GB.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                                    Dload  Upload   Total   Spent    Left  Speed
  1  9.7G    1  118M    0     0         2121k      0  1:20:27  0:00:57  1:19:30 2213k^C
root@OpenMPTCProuter:~# omr-test-speed eth1
Select best test server...
host: speedtest.frankfurt.linode.com - ping: 14
host: speedtest.tokyo2.linode.com - ping: 279
host: speedtest.singapore.linode.com - ping: 270
host: speedtest.newark.linode.com - ping: 101
host: speedtest.atlanta.linode.com - ping: 121
host: speedtest.dallas.linode.com - ping: 137
host: speedtest.fremont.linode.com - ping: 164
host: speedtest.tele2.net - ping: 13
host: proof.ovh.net - ping: 20
host: speed.hetzner.de - ping: 16
host: ipv4.bouygues.testdebit.info - ping: 22
host: par.download.datapacket.com - ping: 25
host: nyc.download.datapacket.com - ping: 96
host: ams.download.datapacket.com - ping: 18
host: fra.download.datapacket.com - ping: 12
host: lon.download.datapacket.com - ping: 27
host: mad.download.datapacket.com - ping: 42
host: prg.download.datapacket.com - ping: 20
host: sto.download.datapacket.com - ping: 36
host: vie.download.datapacket.com - ping: 23
host: war.download.datapacket.com - ping: 37
host: atl.download.datapacket.com - ping: 132
host: chi.download.datapacket.com - ping: 134
host: lax.download.datapacket.com - ping: 159
host: mia.download.datapacket.com - ping: 126
host: nyc.download.datapacket.com - ping: 96
Best server is http://fra.download.datapacket.com/10000mb.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                                     Dload  Upload   Total   Spent    Left  Speed
  3 9536M    3  314M    0     0        5739k      0  0:28:21  0:00:56  0:27:25 3484k^C
root@OpenMPTCProuter:~# omr-test-speed eth2
Select best test server...
host: speedtest.frankfurt.linode.com - ping: 229
host: speedtest.tokyo2.linode.com - ping: 330
host: speedtest.singapore.linode.com - ping: 335
host: speedtest.newark.linode.com - ping: 138
host: speedtest.atlanta.linode.com - ping: 160
host: speedtest.dallas.linode.com - ping: 174
host: speedtest.fremont.linode.com - ping: 249
host: speedtest.tele2.net - ping: 132
host: proof.ovh.net - ping: 53
host: speed.hetzner.de - ping: 45
host: ipv4.bouygues.testdebit.info - ping: 59
host: par.download.datapacket.com - ping: 75
host: nyc.download.datapacket.com - ping: 149
host: ams.download.datapacket.com - ping: 53
host: fra.download.datapacket.com - ping: 41
host: lon.download.datapacket.com - ping: 66
host: mad.download.datapacket.com - ping: 69
host: prg.download.datapacket.com - ping: 65
host: sto.download.datapacket.com - ping: 64
host: vie.download.datapacket.com - ping: 69
host: war.download.datapacket.com - ping: 79
host: atl.download.datapacket.com - ping: 157
host: chi.download.datapacket.com - ping: 139
host: lax.download.datapacket.com - ping: 182
host: mia.download.datapacket.com - ping: 164
host: nyc.download.datapacket.com - ping: 136
Best server is http://fra.download.datapacket.com/10000mb.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                                      Dload  Upload   Total   Spent    Left  Speed
  0 9536M    0 22.4M    0     0        1187k      0  2:17:06  0:00:19  2:16:47 1022k^C

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.58beta7 r0+16336-b36068d35d
  - OpenMPTCProuter VPS version: 0.1024 5.4.86-mptcp - on Ubuntu 20
  - OpenMPTCProuter platform: Ubiquiti EdgeRouter X

I haven't actually changed any settings, it's a fresh install. The VPS reaches speeds up to 150Mbps, so that shouldn't be the bottleneck.

Ysurac, 2021-04-23:
OpenMPTCProuter VPS should be updated too.
Check load average on ER-X when doing a speed test.
Speed from each interfaces doesn't seems to be really great, you should test for at lest 2 minutes. To have test on always same server, use ```omr-test-speed ovh eth2``` (this will use ovh server)

I didn't tested ER-X speed yet, not sure it's really great. Maybe encryption can be disabled in the wizard.

SgtSilvergreen, 2021-04-23:
> OpenMPTCProuter VPS should be updated too.

I set up the VPS a few days ago with the command. Shouldn't this be the latest version?
```
wget https://www.openmptcprouter.com/server/debian10-x86_64.sh
sh debian10-x86_64.sh
```
> Check load average on ER-X when doing a speed test.

Is similar to the current speed

> Speed from each interfaces doesn't seems to be really great, you should test for at lest 2 minutes. To have test on always same server, use `omr-test-speed ovh eth2` (this will use ovh server)

BusyBox v1.33.0 () built-in shell (ash)

   ___                 __  __ ___ _____ ___ ___              _
  / _ \ _ __  ___ _ _ |  \/  | _ \_   _/ __| _ \_ _ ___ _  _| |_ ___ _ _
 | (_) | '_ \/ -_) ' \| |\/| |  _/ | || (__|  _/ '_/ _ \ || |  _/ -_) '_|
  \___/| .__/\___|_||_|_|  |_|_|   |_| \___|_| |_| \___/\_,_|\__\___|_|
       |_|
------------------------------------------------------------------------------
   (r0+16336-b36068d35d)
------------------------------------------------------------------------------
-----------------------------------------------------
 PACKAGE:     openmptcprouter
 VERSION:     v0.58beta7

 BUILD REPO:  https://github.com/ysurac/openmptcprouter
 BUILD DATE:  Fri Apr 16 20:04:10 UTC 2021
-----------------------------------------------------
root@OpenMPTCProuter:~# omr-test-speed ovh eth2
Best server is http://proof.ovh.net/files/10Gio.dat, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  4 10.0G    4  454M    0     0  1559k      0  1:52:05  0:04:58  1:47:07 1647k
root@OpenMPTCProuter:~# omr-test-speed ovh eth1
Best server is http://proof.ovh.net/files/10Gio.dat, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
 11 10.0G   11 1137M    0     0  6025k      0  0:29:00  0:03:13  0:25:47 6049k
root@OpenMPTCProuter:~# omr-test-speed ovh
Best server is http://proof.ovh.net/files/10Gio.dat, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
 26 10.0G   26 2690M    0     0  3716k      0  0:47:01  0:12:21  0:34:40 3607k

> I didn't tested ER-X speed yet, not sure it's really great. Maybe encryption can be disabled in the wizard.

I do not know either, I got the router yesterday. Have no idea where I could find such a setting.

Ysurac, 2021-04-23:
You use an unstable release (v0.58rc1), you have to use unstable release of VPS too: ```wget -O - http://www.openmptcprouter.com/server-test/debian10-x86_64.sh | sh```

When I say load average, I mean CPU load average, you can see them in System->OpenMPTCProuter status page. Check also that all is green in status page.

Encryption can be disabled in System->OpenMPTCProuter, "advanced settings" checkbox.

SgtSilvergreen, 2021-04-23:
> You use an unstable release (v0.58rc1), you have to use unstable release of VPS too: `wget -O - http://www.openmptcprouter.com/server-test/debian10-x86_64.sh | sh`

Alright, done

> When I say load average, I mean CPU load average, you can see them in System->OpenMPTCProuter status page. Check also that all is green in status page.

[image]

> Encryption can be disabled in System->OpenMPTCProuter, "advanced settings" checkbox.

Ok, done

Ysurac, 2021-04-23:
I see a high load average. CPU of ER-X is really slow...
And what is the result with disabled encryption ? A little better speed ?

SgtSilvergreen, 2021-04-23:
Yes, a lot better. Around 70 Mbit/s on the OMR but when i do a speedtest on my PC its around 40-50 Mbit/s

Ysurac, 2021-04-23:
Strange to get a lower speed on your PC...

Ysurac, 2021-04-23:
You must set an interface as Master in System->OpenMPTCProuter, wizard page. Else you can get some strange results.

SgtSilvergreen, 2021-04-23:
Yea, i just read something about hardware offloading, would that help the router?
Master I have now set, but gave no speed difference

Ysurac, 2021-04-23:
You can try, this may help.

SgtSilvergreen, 2021-04-23:
I think I enabled offload in Network -> Firewall -> Routing/NAT, but it has little effect on performance. When I do a speed test it is as if the LTE is not added at all, since bonding speed and speed of DSl are almost identical.

Ysurac, 2021-04-23:
You can check in Network->MPTCP, Bandwidth tab if aggregation is working or not.

SgtSilvergreen, 2021-04-26:
The bonding works and it also shows me reasonably good values in the bandwidth interface. Unfortunately, he can not really process and forward them to my computer, so it is not worth it for me. Nevertheless, thank you for your help. Really good work

closed 2021-04-26 by SgtSilvergreen

Ysurac, 2021-04-26:
I think ER-X is too slow to give any real good result. A RPI4 give a far better result.