Ysurac/openmptcprouter #1774: ShadowsSoks is not running

gianlucabarani, 2021-04-23:
Good morning,
i setup VPS and router, internet navigation working but i have ShadowsSoks is not running . My hardware is

Modello | Raspberry Pi 3 Model B Rev 1.2
Architecture | ARMv7 Processor rev 4 (v7l)
Versione Firmware | openmptcprouter v0.57.3 r0+15225-bfc433efd4 / LuCI Master git-20.339.75073-e54708a
Versione del Kernel | 5.4.83

Only one question in Menu-shadowsocks-libev ----->under Remote Server there is a label Password and Key(Base64), but in file configuration under VPS /root/openmptcprouter_config.txt ther is  Key
Thanks
best regards

Ysurac, 2021-04-23:
Did you reboot the VPS ? The kernel must be 5.4.86-mptcp
Only the VPS key must be set in System->OpenMPTCProuter, wizard all other keys are retrieved via the VPS API.

gianlucabarani, 2021-04-24:
Hi, yes VPS it was rebooted several time. Sorry how i can upgrade Kernel? I must flash another image? At this link i don't found https://www.openmptcprouter.com/download 
I 've downloded and flashed this image ext4 image: openmptcprouter-v0.57.3-r0+15225-bfc433efd4-bcm27xx-bcm2710-rpi-3-ext4-factory.img.gz (2021/01/05 - 65.88Mo - sha256sum)
If log into ssh openmptcp router can i upgrading it?
Sorry.

Ysurac, 2021-04-24:
You must install VPS script. If you can change the kernel then you must have a MPTCP kernel.
Make sure you can use a custom kernel on your VPS provider.

gianlucabarani, 2021-04-24:
Sorry i have installed VPS script on VPS provider side on Ubuntu 20. The VPS script must be installed on the raspberry (mptcp router) side? is correct?

Ysurac, 2021-04-24:
No, VPS script must be installed on VPS side. Debian 10 is recommended.
When you reboot, you must have with ```uname -a``` a kernel with 5.4.86-mptcp. If it's not the case it's possible that VPS provider doesn't allow custom Kernel.

gianlucabarani, 2021-04-24:
uname -a it' ok return  5.4.86-mptcp.

gianlucabarani, 2021-04-24:
now seem working but unstable, sometimes in state tab there is a VPS connected with public IP with green check. After random minuts green check change in Direct output with public IP local Wan (not VPS). In the OpenMPTCProuter figure there is x check witch "ShadowSocks not running" but i thinks that is correct because VPN is not configure.

Ysurac, 2021-04-24:
Check State->System log
If you correctly configured key, "shadowsocks not running" should not be displayed.

gianlucabarani, 2021-04-25:
[image]
I upload image ' status . I 've tried several configuration all key are set (server key. shadowsockskey, etcc) but ther is someting wrong that i don't understanding it. The kernel in Rapsberry Pi3 is 5.4.83 . Kernel in VPS is 5.4.86-mptcp. 
reflash image on Raspberry? Sorry, best regards

Ysurac, 2021-04-25:
There is a configuration problem, both interface seems to have same gateway and share same interface name.
You must use macvlan if you want to use same interface for multiples sources.

gianlucabarani, 2021-04-26:
Ok reflash image seem all ok now. Tanks .
[image]

closed 2021-04-26 by gianlucabarani