Ysurac/openmptcprouter #1777: Need to add ffmpeg and nginx rtmp lib so that we can add encoding part in router by add usb 3.0 capture card

ssdeol84, 2021-04-24:
**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**

Need bonding encoder by which we can Aggregation bandwidth with rtmp stream.
H.264

**Describe alternatives you've considered**

Enhance the bonding feature
Like other bonding encoders work like liveU solo,tvu and other solutions

**Additional context**

Please add this feature.

Ysurac, 2021-04-24:
I don't know what you want exactly, but many packages are available in System->Software.

suyuan168, 2021-04-26:
Currently, no consideration is given to embedding the video coding part in OMR.
OMR should only be a router.
If necessary, I will consider it later.

Adorfer, 2021-04-26:
please do not add too many bells and whistles aside "beeing a mptcprouter".

closed 2021-07-31 by github-actions[bot]