Ysurac/openmptcprouter #1778: Build failing

wicadmin, 2021-04-24:
The instructions need to be updated to included `wget`. But build still fails nonetheless.

[image]

Ysurac, 2021-04-24:
And what is the result when you re run it with -j1 V=s as indicated ?

wicadmin, 2021-04-24:
```
make[8]: Nothing to be done for 'install-data-am'.
make[8]: Leaving directory '/builds/wicadmin/openmptcprouter/openmptcprouter/x86_64/source/build_dir/host/patch-2.7.6/lib'
make[7]: Leaving directory '/builds/wicadmin/openmptcprouter/openmptcprouter/x86_64/source/build_dir/host/patch-2.7.6/lib'
make[6]: Leaving directory '/builds/wicadmin/openmptcprouter/openmptcprouter/x86_64/source/build_dir/host/patch-2.7.6/lib'
Making install in src
make[6]: Entering directory '/builds/wicadmin/openmptcprouter/openmptcprouter/x86_64/source/build_dir/host/patch-2.7.6/src'
make[7]: Entering directory '/builds/wicadmin/openmptcprouter/openmptcprouter/x86_64/source/build_dir/host/patch-2.7.6/src'
make[7]: Nothing to be done for 'install-data-am'.
 /bin/mkdir -p '/builds/wicadmin/openmptcprouter/openmptcprouter/x86_64/source/staging_dir/host/bin'
  /builds/wicadmin/openmptcprouter/openmptcprouter/x86_64/source/staging_dir/host/bin/install -c patch '/builds/wicadmin/openmptcprouter/openmptcprouter/x86_64/source/staging_dir/host/bin'
make[7]: Leaving directory '/builds/wicadmin/openmptcprouter/openmptcprouter/x86_64/source/build_dir/host/patch-2.7.6/src'
make[6]: Leaving directory '/builds/wicadmin/openmptcprouter/openmptcprouter/x86_64/source/build_dir/host/patch-2.7.6/src'
Making install in tests
make[6]: Entering directory '/builds/wicadmin/openmptcprouter/openmptcprouter/x86_64/source/build_dir/host/patch-2.7.6/tests'
make[7]: Entering directory '/builds/wicadmin/openmptcprouter/openmptcprouter/x86_64/source/build_dir/host/patch-2.7.6/tests'
make[7]: Nothing to be done for 'install-exec-am'.
make[7]: Nothing to be done for 'install-data-am'.
make[7]: Leaving directory '/builds/wicadmin/openmptcprouter/openmptcprouter/x86_64/source/build_dir/host/patch-2.7.6/tests'
make[6]: Leaving directory '/builds/wicadmin/openmptcprouter/openmptcprouter/x86_64/source/build_dir/host/patch-2.7.6/tests'
make[6]: Entering directory '/builds/wicadmin/openmptcprouter/openmptcprouter/x86_64/source/build_dir/host/patch-2.7.6'
make[7]: Entering directory '/builds/wicadmin/openmptcprouter/openmptcprouter/x86_64/source/build_dir/host/patch-2.7.6'
make[7]: Nothing to be done for 'install-exec-am'.
 /bin/mkdir -p '/builds/wicadmin/openmptcprouter/openmptcprouter/x86_64/source/staging_dir/host/share/man/man1'
 /builds/wicadmin/openmptcprouter/openmptcprouter/x86_64/source/staging_dir/host/bin/install -c -m 644 'patch.man' '/builds/wicadmin/openmptcprouter/openmptcprouter/x86_64/source/staging_dir/host/share/man/man1/patch.1'
make[7]: Leaving directory '/builds/wicadmin/openmptcprouter/openmptcprouter/x86_64/source/build_dir/host/patch-2.7.6'
make[6]: Leaving directory '/builds/wicadmin/openmptcprouter/openmptcprouter/x86_64/source/build_dir/host/patch-2.7.6'
make[5]: Leaving directory '/builds/wicadmin/openmptcprouter/openmptcprouter/x86_64/source/build_dir/host/patch-2.7.6'
make[4]: Leaving directory '/builds/wicadmin/openmptcprouter/openmptcprouter/x86_64/source/build_dir/host/patch-2.7.6'
mkdir -p /builds/wicadmin/openmptcprouter/openmptcprouter/x86_64/source/staging_dir/host/stamp
touch /builds/wicadmin/openmptcprouter/openmptcprouter/x86_64/source/build_dir/host/patch-2.7.6/.built
touch /builds/wicadmin/openmptcprouter/openmptcprouter/x86_64/source/staging_dir/host/stamp/.patch_installed
find /builds/wicadmin/openmptcprouter/openmptcprouter/x86_64/source/build_dir/host/patch-2.7.6 -mindepth 1 -maxdepth 1 -not '(' -type f -and -name '.*' -and -size 0 ')' | xargs -r rm -rf
make[3]: Leaving directory '/builds/wicadmin/openmptcprouter/openmptcprouter/x86_64/source/tools/patch'
time: tools/patch/compile#29.05#13.99#37.51
make[2]: Leaving directory '/builds/wicadmin/openmptcprouter/openmptcprouter/x86_64/source'
make[1]: *** [tools/Makefile:154: /builds/wicadmin/openmptcprouter/openmptcprouter/x86_64/source/staging_dir/host/stamp/.tools_compile_yyyyyynnyyyynyyyyyynyynnyyynyyyyyyyyyyyyyyyynynnyyyyyyyy] Error 2
make[1]: Leaving directory '/builds/wicadmin/openmptcprouter/openmptcprouter/x86_64/source'
make: *** [/builds/wicadmin/openmptcprouter/openmptcprouter/x86_64/source/include/toplevel.mk:230: world] Error 2
```

Ysurac, 2021-04-26:
No idea. You can check what is done by Github Actions to compile all images: https://github.com/Ysurac/openmptcprouter-feeds/blob/develop/.github/workflows/main.yml

wicadmin, 2021-04-27:
Looks like it gives this error if you build as root user. Used a non-root user instead.

closed 2021-04-27 by wicadmin