Ysurac/openmptcprouter #1779: EdgeRouter X crashes constantly

milankragujevic, 2021-04-24:
## Expected Behavior
EdgeRouter X should not crash. 

## Current Behavior
EdgeRouter X crashes every few minutes. 

Description: 

It crashes in two possible ways:

1. It stays turned on but all ports go down and up. This is a bug with OpenWRT apparently https://bugs.openwrt.org/index.php?do=details&task_id=972
2. It reboots, but quickly (doesn't take more than a minute to recover, compared to regular 2-5 minutes to turn on)

## Possible Solution
No idea.

## Steps to Reproduce the Problem

  1. Install OMR v0.58rc1 on EdgeRouter X
  2. Setup with VPS, turn off encryption, add 3 WANs
  3. Run a Speedtest or leave it as is, it will inevitably crash

## Context (Environment)
Not applicable. 

## Specifications

  - OpenMPTCProuter version: v0.58rc1
  - OpenMPTCProuter VPS version: v0.1025-test 5.4.100-mptcp
  - OpenMPTCProuter platform: ubnt-erx, raips

milankragujevic, 2021-04-24:
Okay, the problem is definitely extremely high CPU usage (load avg. 2.56, 1.88, 1.39). I'm just not sure what I'm doing wrong. I tried it with 2 WANs, same. Tried disabling features, same. Extreme CPU load. What can I do, @Ysurac? I bought this ER-X specifically for OMR, would be a real shame if it didn't work out in the end.

### top
```
root@OpenMPTCProuter:~# top
Mem: 105712K used, 146360K free, 832K shrd, 13072K buff, 37744K cached
CPU:  10% usr  17% sys   0% nic  72% idle   0% io   0% irq   0% sirq
Load average: 1.31 1.54 1.32 2/100 26717
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
26651 15053 root     S     1660   1%   6% /bin/sh /bin/omr-tracker wan2
26707 26032 root     R     1312   1%   2% top
22609     1 root     S     4620   2%   0% /usr/bin/ttyd -i eth0 /bin/login
 4207     1 root     S     3744   1%   0% /usr/sbin/uhttpd -f -h /www -r OpenMPTCProuter -x /cgi-bin -u /ubus -t 240 -T 240 -k 0 -A 1 -n 3 -N 100 -R -p 0.0.0.0:80 -p [::]:80 -C /etc/uhttpd.crt -K /etc/uhttpd.key
  834     1 root     S     3472   1%   0% /sbin/rngd -f -r /dev/urandom
11223 11192 root     S     3408   1%   0% adb -P 5037 fork-server server
 3744     1 root     S     3196   1%   0% /usr/sbin/snmptrapd -Lf /dev/null -f
11192 11189 root     S     3120   1%   0% adb wait-for-device
26608 15055 root     S     3068   1%   0% curl -s -4 -m 3 http://ip.openmptcprouter.com
13813     1 dnsmasq  S     2832   1%   0% /usr/sbin/dnsmasq -C /var/etc/dnsmasq.conf.cfg01411c -k -x /var/run/dnsmasq/dnsmasq.cfg01411c.pid
 5112     1 root     S     2440   1%   0% /usr/sbin/vnstatd --nodaemon
12467     1 root     S     2392   1%   0% /usr/bin/ss-local -c /var/etc/shadowsocks-libev/ss_local.tracker_sss0.json
12468     1 root     S     2320   1%   0% /usr/bin/ss-redir -c /var/etc/shadowsocks-libev/ss_redir.hi1.json
12469     1 root     S     2320   1%   0% /usr/bin/ss-redir -c /var/etc/shadowsocks-libev/ss_redir.hi3.json
 2860     1 root     S     2228   1%   0% /sbin/rpcd -s /var/run/ubus/ubus.sock -t 120
 3302     1 root     S     1896   1%   0% /usr/sbin/hostapd -s -g /var/run/hostapd/global
 3303     1 root     S     1896   1%   0% /usr/sbin/wpa_supplicant -n -s -g /var/run/wpa_supplicant/global
    1     0 root     S     1768   1%   0% /sbin/procd
15027     1 root     S     1732   1%   0% /sbin/netifd
 4681     1 root     S     1544   1%   0% /usr/bin/dbus-daemon --system
13370     1 root     S     1504   1%   0% /usr/sbin/glorytun keyfile /tmp/glorytun-vpn.key port 65001 host 185.119.90.12 dev tun0 mptcp retry count -1 const 500000 timeout 10000 keepalive count 5 idle 20 interva
 5487     1 root     S     1488   1%   0% /sbin/blockd
15053     1 root     S     1432   1%   0% /bin/sh /bin/omr-tracker wan2
15054     1 root     S     1432   1%   0% /bin/sh /bin/omr-tracker omrvpn
15052     1 root     S     1432   1%   0% /bin/sh /bin/omr-tracker wan1
11189     1 root     S     1368   1%   0% /bin/sh /etc/rc.common /etc/rc.d/S99adb-enablemodem boot
 2808     1 logd     S     1364   1%   0% /sbin/logd -S 64
15055     1 root     S     1316   1%   0% /bin/sh /bin/omr-tracker-ss tracker_sss0
26032 25986 root     S     1312   1%   0% -ash
 3632     1 root     S     1308   1%   0% /usr/sbin/crond -f -c /etc/crontabs -l 5
11133     1 root     S<    1304   1%   0% /usr/sbin/ntpd -n -N -l -S /usr/sbin/ntpd-hotplug -p 0.openwrt.pool.ntp.org -p 1.openwrt.pool.ntp.org -p 2.openwrt.pool.ntp.org -p 3.openwrt.pool.ntp.org
26717 26651 root     R     1304   1%   0% [pgrep]
  790     1 ubus     S     1280   1%   0% /sbin/ubusd
25986  3073 root     S     1172   0%   0% /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22 -K 300 -T 3
 3073     1 root     S     1148   0%   0% /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22 -K 300 -T 3
  953     1 root     S     1044   0%   0% /sbin/urngd
26645 15052 root     S      980   0%   0% ping -B -I eth1 -w 2 -c 2 -Q 184 192.168.1.1
26634 15054 root     S      980   0%   0% ping -B -I tun0 -w 10 -c 2 -Q 184 10.255.255.1
  796     1 root     S      940   0%   0% /sbin/askfirst /usr/libexec/login.sh
   10     2 root     IW       0   0%   0% [rcu_sched]
   40     2 root     IW       0   0%   0% [kworker/2:1-mm_]
   32     2 root     IW       0   0%   0% [kworker/0:1-eve]
  200     2 root     IW       0   0%   0% [kworker/3:1-mm_]
   26     2 root     IW       0   0%   0% [kworker/3:0-eve]
    9     2 root     SW       0   0%   0% [ksoftirqd/0]
   15     2 root     SW       0   0%   0% [ksoftirqd/1]
   14     2 root     SW       0   0%   0% [migration/1]
   20     2 root     SW       0   0%   0% [ksoftirqd/2]
   24     2 root     SW       0   0%   0% [migration/3]
   19     2 root     SW       0   0%   0% [migration/2]
   11     2 root     SW       0   0%   0% [migration/0]
   25     2 root     SW       0   0%   0% [ksoftirqd/3]
11596     2 root     IW       0   0%   0% [kworker/1:0-eve]
  700     2 root     SW       0   0%   0% [ubifs_bgt0_1]
 8906     2 root     IW       0   0%   0% [kworker/u8:3-ev]
 4177     2 root     IW       0   0%   0% [kworker/0:2-eve]
31243     2 root     IW       0   0%   0% [kworker/u8:0-ev]
^C496     2 root     SW       0   0%   0% [ubi_bgt0d]
root@OpenMPTCProuter:~#
```

[... 1343 lines cut ...]
Thu Mar 25 06:50:10 2021 daemon.info dnsmasq[13813]: using only locally-known addresses for domain onion
Thu Mar 25 06:50:10 2021 daemon.info dnsmasq[13813]: using only locally-known addresses for domain localhost
[... cut ...]
Thu Mar 25 06:50:17 2021 user.notice omr-bypass: OMR-ByPass is running
Thu Mar 25 06:50:18 2021 user.notice omr-tracker: Launched
Thu Mar 25 06:50:19 2021 daemon.info glorytun: starting glorytun vpn instance vpn
Thu Mar 25 06:52:31 2021 authpriv.info dropbear[25986]: Child connection from 192.168.10.100:3225
Thu Mar 25 06:52:32 2021 authpriv.notice dropbear[25986]: Pubkey auth succeeded for 'root' with key sha1!! a1:2e:e1:d5:b2:e5:82:4f:47:81:71:ad:61:44:86:17:90:39:87:72 from 192.168.10.100:3225
```

Kalimeiro, 2021-04-25:
you need to choose another crypto for glorytun : 
Thu Mar 25 06:28:01 2021 daemon.err glorytun[13370]: AES-256-GCM is not available on your platform

the default is chach20.

milankragujevic, 2021-04-25:
I selected "none".

milankragujevic, 2021-04-26:
I tried it again, reflashed the router. It is really slow. Unbearably slow. Load (from loadavg) doesn't go above 2.5, and the CPU should have 4 cores, but the web UI is extremely slow, it cannot serve even CSS properly every time. 

It is unclear to me if it should work like this or not. 

I didn't give any logs for now as this is just being slow, nothing in logs about that.

@Ysurac please advise.

Ysurac, 2021-04-26:
Check via SSH using htop what is using CPU or/and memory.
It's slow even when connections are not used ?

milankragujevic, 2021-04-26:
Nothing special. RAM is never more than half used. 

[image]

This is with all ports except eth0 disconnected. Factory reset router, reflashed again. 

The Web UI is incredibly slow.

Adorfer, 2021-04-26:
i do not see the link to https://bugs.openwrt.org/index.php?do=details&task_id=972
since that but was resolved in OpenWRT18.x
1) are you really sure about this regression in current OpenWRT?
2) please try to provide crash/reboot-trigger log, ideally from the serial console.

milankragujevic, 2021-04-26:
@Adorfer 

1. I am not sure, I did not do enough research, this was just an example.
2. I cannot access the serial console as the device is under warranty with a warranty void if removed seal. Warranty covers hardware if the device is not tampered with. Hardware warranty cannot be voided by modifying the software, only software warranty. The device comes with no software warranty as part of it's EULA. Therefore I cannot open the device and cannot access the serial console.

The problem is mostly related to the device being incredibly slow. I am not sure if anyone else with the same router has the same problems?

Adorfer, 2021-04-27:
- please point out the significant lines from "logread -f" during a crash. (if there is such info)
- what is the exact symptom of a crash?

milankragujevic, 2021-04-27:
@Adorfer 

1. The devices reboots, I cannot access a log. I will leave it running with logread and see what it last said.
2. it reboots (uptime is reset)

Please pay more attention to the fact that it is unbearably slow and has idle CPU usage when reset and all ports except eth0 (LAN) disconnected of about 50% per core. Doing nothing, with no internet or configuration. This probably causes a crash -- high load.

Something is wrong, somewhere, if @Ysurac claims to support it but I cannot get it configured properly given how slow it is.

Ysurac, 2021-04-27:
@milankragujevic ER-X is not supported in current stable release, so I claim nothing, I only compile an image to allow tests on a beta image. I didn't have yet the time to test how it perform.

Adorfer, 2021-04-27:
reboot, understood. ("crash" could have been "freeze" as well, that's why i am asking)

>     1. The devices reboots, I cannot access a log. I will leave it running with logread and see what it last said.

please try a "logread -f" with console open. with some look, perhaps with several tries, you get some chunks of the problem before the device restarts.

milankragujevic, 2021-04-27:
@Ysurac Aha! Very good, now I understand you did not test it. I was under the impression it worked for everyone except me.

@Adorfer It's slow even with no cables connected, no configuration (right after flashing). That is a problem, it shouldn't be that slow. 

I don't need speeds above 100/50 Mbps, without encryption. I have a stable VDSL link with 6ms ping to server but only 20/2 Mbps speed, variable 85-95/8 Mbps link over 5GHz MikroTik wireless (DOCSIS on the other end, 2.8km) and LTE-A 150/50 Mbps but small data cap. I combine it but limit downloda speed of LTE, so I use it for upload only, and on my MikroTik router behind OMR I limit WAN speeds to 100/50 in any case. 

But it is irrelevant as I did not connect anything to ER-X and it gets 50% CPU usage on all cores.

milankragujevic, 2021-04-30:
Can anyone who **has** an ER-X please **install** OMR v0.58rc1 on it and **test if it is unbearably slow**? Thanks!

igor095, 2021-05-11:
@milankragujevic

I have ER-X running OMR since Jan'21. Works fine. Yesterday I upgraded my ER-X with latest image from @Ysurac v0.58rc2. Works fine too.

I also saw lacking LUCI on ER-X and it was connected with:
1. "Misconfiguration."

Sysupgrade -n 'img'. Then disable everything first. All interfaces but LAN0, MPTCP, rules in shadowsocks, dnsmasq. And then enable necessary components step-by-step. This works with RPI platform too.

2. "Ethernet". 

It's an Ubiquiti issue. Unstable work of Ethernet at 1Gb speed. If at least one interface has a connectivity problem -> all other ports begins to loose packets -> CPU overload. In other words, even if you connect your modem with 1 meter patch cable -> But THIS cable cannot guarantee stable work at 1Gbs with ER-X -> You will get problems. Also with current linux drivers for MT chipset you cannot enable "full manual" mode. On the boot it still will be trying "autonegotiation", then changes the speed, then "link up", then "link down" etc. Also you won't see errors on interface because it's not an L2 problem. 

It doesn't mean that you cannot use ER-X with 1Gbs link - of course you can, but keep in mind all written above. I use 1Gb for LAN0 and 100Mbs for WAN0 and WAN1.

milankragujevic, 2021-05-11:
@igor095 But how does it work with stock configuration?

igor095, 2021-05-12:
@milankragujevic Actually I didn't use it on hi speed link with stock firmware. But original firmware IMHO based on OpenWRT and uses LUCI and I cannot say that it was more reliable than OMR.

This screenshot I did this morning:
[image]

milankragujevic, 2021-05-27:
So it's been a while, but I did as I was instructed by @Adorfer, and it doesn't even get to showing anything, it just restarts:

[image]

It keeps rebooting constantly, as soon as it can boot up and start all the stuff, it will reboot. 

Symptoms are that it reboots (lights do the reboot thing), and the uptime resets obviously. You **cannot** get logs out of it because it crashes instantly, not in stages one by one like dominos.

@Ysurac I strongly advise you to put a warning in the RC notes about ER-X, as I bought this device specifically for OMR, and although you did not claim any support, it was implied by it being announced as when you supported any other device, but this device is absolutely incapable of running OMR at all. 

I managed one speedtest between the billion reboots, and it's absolutely terrible. This is with encryption turned **off**.

[image]

igor095, 2021-05-28:
The only thing I can imagine - we are talking about completely different devices. 

Also I suppose that there is no problem with edge-x,  but may be with PSU? Try to use another one for 1A. For example Netgear says that if you use POE out then the standard PSU MAY NOT provide enough power for device to work normally. You should use more powerful PSU. 

Any way my EDGE never rebooted occasionally. Max uptime 46 days.

[image]
[image]

milankragujevic, 2021-05-28:
@igor095 It is the same device and I don't use PoE. Do you only use it with MacVLAN or dis you try the Ethernet ports?

igor095, 2021-05-28:
I just wanted to say that the only difference from original is the PSU for 1A (but it is not obligate, I think). Now I'm using 802.11Q (not MacVLAN) only for economy of 1Gb port quantity. 

Before v0.58rc2 it worked with common configuration: Port 0 -> Lan0, Port 1 -> WAN1, Port 2 -> WAN2.

milankragujevic, 2021-05-28:
@igor095  I can test with 12V 1A PSU (and 12V 2A if needed). I didn't think the PSU was so weak, never looked at it, but I see the spec now is 0.5A for the original one :( We'll see, I'll post here my results. Could be power problem, as it crashes instantly. But could also not be, will see...

milankragujevic, 2021-05-31:
@Ysurac I did as instructed by @igor095 and tried the ER-X with 12V 1.5A PSU that is known to be working (I have 20 different adapters, I can try them all but they're all fine) and I recorded a handy video of it's behavior (spoiler: it's the same as before). Didn't try accessing it as it's obviously constantly crashing - it can't power on fully most times, so all LEDs shortly blink then it tries to power on and crashes, etc. It then powers on fully and crashes after a few seconds. I can't get logs out of it as it crashes before it can log what happened to it. Doesn't happen on factory firmware. 

https://user-images.githubusercontent.com/7611163/120199289-5b84b600-c223-11eb-87b1-2f48ac2edc52.mp4

milankragujevic, 2021-06-01:
@igor095 Just another update: I tried with the firmware you emailed me, and it works in that it doesn't reboot, at all, ever, but it's really buggy and the connection is pretty bad (~50/50 Mbps with 3 bonded instead of ~120/50 Mbps, didn't try with 2). It's a step forward at least. 

Additionally, @Ysurac, I coordinated a check with some friends who also have ER-X's, it seems the problems are not related to OMR but to OpenWrt itself being unstable and very unpredictable on this device. They encountered the same problem I described in the video above. Weirdly, it works for me with some firmware image compiled by @igor095 in that it doesn't reboot all the time. 

This might be something "out of your hands" @Ysurac, so I will investigate with OpenWrt snapshot, and if it happens still, I will move my discussion to their forums, and if ever there is a solution, I will notify you all here. If it doesn't happen on OpenWrt, I will also notify.

Thanks for your time in any case in helping me follow through with this issue.

milankragujevic, 2021-08-30:
Not stale.

kverkind, 2021-11-16:
Not an answer but I can confirm Edgerouter-X is very unstable (not usable). I moved to RPI 4 and there it runs flawless (only problems after power cut).

simontretter, 2021-11-29:
Maybe this helps: OpenWRT master constantly crashes with the current default kernel version (5.4), but is reasonably stable when switching to the testing kernel (5.10).

kverkind, 2021-11-29:
> Maybe this helps: OpenWRT master on the ER-X constantly crashes for me with the current default kernel version (5.4), but is reasonably stable when switching to the testing kernel (5.10).

Where can I find this?

Ysurac, 2021-11-29:
Kernel 5.10 will not be supported by OpenMPTCProuter. For now next supported kernel will be 5.15 but I didn't ported ER-X yet.

simontretter, 2021-12-03:
@Ysurac Why is that? Won't 5.10 be the choice for the next OpenWRT 22.XX stable? 

@kverkind You would need to build OpenWRT yourself and set CONFIG_TESTING_KERNEL=y (as ramips is still on 5.4 on OpenWRT master).

Ysurac, 2021-12-03:
@simontretter Because kernel 5.10 have a version of MPTCP (mptcp-next) in kernel that I can't use because fullmesh is not supported.
The first usable version of MPTCP upstreamed is 5.14.
So, as I don't want to port current out kernel MPTCP version, and because MPTCP in kernel is more tested, I work on a next release based on 5.15.

milankragujevic, 2022-03-04:
Not stale.

milankragujevic, 2022-06-02:
Not stale.

milankragujevic, 2022-09-01:
Not stale.

Ysurac, 2022-09-01:
OpenMPTCProuter 0.59.1 is also available with a 5.15 release to test, did you test it ?

milankragujevic, 2022-09-01:
@Ysurac No, I got rid of the router. The issue can be closed with "pending further testing", but not by the bot's comment :)

ccmks, 2022-09-19:
I am willing to test this. Before I decide to buy the ER-X, where can I see the documentation to flash OMR to the ER-X?

When flashing OMR, should I use the 5.15 or 5.4 kernel? Can the kernel be mixed with VPS, like OMR is using 5.15 but VPS is using 5.4, and vice versa?

Do you think it would be worth to buy this to test, or it is not worth because there might be a lot of bugs that have not been addressed yet?

Thank you

harryparato, 2022-10-12:
> I am willing to test this. Before I decide to buy the ER-X, where can I see the documentation to flash OMR to the ER-X?
>
> When flashing OMR, should I use the 5.15 or 5.4 kernel? Can the kernel be [...]

Hi, have you tested the firmware on Edgerouter X? It works fine?

Thank You

ccmks, 2022-10-13:
No, someone opened new post, about Edgerouter X. It seemed to be working, but due to low power CPU, it can barely get around 40 Mbps aggregated connection. Since then, I decided not to get one because it is not worth it.

closed 2023-01-11 by milankragujevic

rewulff, 2024-06-18:
Are there any updates about hardware offload and cpu performance in current openmptcp version on er-x?