Ysurac/openmptcprouter #1780: Limiter le débit sur certains IP

JeedHome44, 2021-04-25:
## Expected Behavior

Je constate que mes appareils sont lent si Netflix est en route sur la Tv ou autre...

## Current Behavior

Est-ce possible de limiter le débit en download d'un ou plusieurs IP du LAN?

## Specifications

  - OpenMPTCProuter version:  57.3
  - OpenMPTCProuter VPS version:  last
  - OpenMPTCProuter platform:  Pi 4B 2Go

Kalimeiro, 2021-04-25:
Pour limiter le débit ou dans la plupart des cas mieux le gérer, il faut utiliser SQM.

Pour limiter le débit de ton LAN, par exemple tes équipements TV, il faut que ceux-ci soit dans un VLAN séparé de tes autres "appareils".

Connaissant ta configuration réseau (tout dans un même LAN, plusieurs routeurs, plusieurs CPL, plusieurs switchs en cascade), tu n'arriveras pas à équilibrer tes flux réseaux correctement.

JeedHome44, 2021-04-25:
> Pour limiter le débit ou dans la plupart des cas mieux le gérer, il faut utiliser SQM.
>
>
>
> Pour limiter le débit de ton LAN, par exemple tes équipements TV, il faut que ceux-ci soit dans un VLAN [...]

Salut!

Je pense que tu n'as pas vu mes réponses dans mon autre issue que j'ai clôturé. J'ai viré les CPL depuis plus de 10 jours, j'ai vérifié chaque AP et certains étaient sur auto avec le DHCP au lieu de désactivé.
J'ai donc forcé sur désactivé et depuis, je n'ai enfin plus mon problème de LAN.

Et oui, OMR doit fonctionner sur un seul LAN. C'est le but de OMR non?

Donc mon LAN va bien mieux depuis que j'ai fait ça. J'ai pu remettre la 4G en maître et tout est au top!

Ça n'a donc plus rien à voir avec cette issue.

La, je veux pas limiter le débit d'un wan. Je veux juste limiter le débit d'un appareil.

On peut bien faire un byPass pour dire que telle ou telle IP soit que sur tel ou tel wan.

Donc, n'est-il pas possible de faire pareil mais dire à OMR de limiter le débit sur tel IP.

Par exemple, autoriser la TV à télécharger en 2Mbps Max. La Mi Box en 2Mbps aussi...?

Kalimeiro, 2021-04-25:
> Salut!
>
> Je pense que tu n'as pas vu mes réponses dans mon autre issue que j'ai clôturé. J'ai viré les CPL depuis plus de 10 jours, j'ai vérifié chaque AP et certains étaient sur auto avec le DHCP au [...]

Ce n'est pas ce qui a été recommandé.

> 
> Et oui, OMR doit fonctionner sur un seul LAN. C'est le but de OMR non?

Non.

> 
> Donc mon LAN va bien mieux depuis que j'ai fait ça. J'ai pu remettre la 4G en maître et tout est au top!
> 
> Ça n'a donc plus rien à voir avec cette issue.
> 
> La, je veux pas limiter le débit d'un wan. Je veux juste limiter le débit d'un appareil.

J'ai bien compris, ne t'inquiète pas là dessus.

> 
> On peut bien faire un byPass pour dire que telle ou telle IP soit que sur tel ou tel wan.

Oui, c'est le but d'un "bypass".

> 
> Donc, n'est-il pas possible de faire pareil mais dire à OMR de limiter le débit sur tel IP.
> 
> Par exemple, autoriser la TV à télécharger en 2Mbps Max. La Mi Box en 2Mbps aussi...?
> 

Oui, avec des VLAN et SQM.

JeedHome44, 2021-04-25:
Et donc, OMR doit être paramétré comment en multi LAN?

Depuis le début, on m'a dit de brancher mon Pi OMR sur la Freebox et mon routeur 4G sur la Freebox également...

Et comme ça, OMR est seul DHCP sur le LAN.

Première fois que j'entends que ce n'est pas recommandé...

Donc tu fais comment pour faire un VLAN avec un seul appareil et ensuite paramétrer SQM stp?

Kalimeiro, 2021-04-25:
> 
> 
> Et donc, OMR doit être paramétré comment en multi LAN?

Avec des VLAN !

> 
> Depuis le début, on m'a dit de brancher mon Pi OMR sur la Freebox et mon routeur 4G sur la Freebox également...
> 
> Et comme ça, OMR est seul DHCP sur le LAN.
> 
> Première fois que j'entends que ce n'est pas recommandé...

OK mais ce n'est pas DU TOUT ce que je t'ai donné comme conseille sur l'architecture ! tu avais plusieurs routeurs, plusieurs CPL, bornes WIFI, le tout brancher d'une manière qui ne peut pas bien fonctionner.

Je t'ai dit de tout concentrer en un seul point sur un même switch et de créer des VLAN pour répondre précisément à ce que tu souhaite.

> 
> Donc tu fais comment pour faire un VLAN avec un seul appareil et ensuite paramétrer SQM stp?

Tu mélange vraiment tout et tes compétences sont limités alors je te réexplique tout une dernière fois et tu cherche sur internet pour configurer des VLAN etc... car ce n'est pas le but de parler de ça ton problème n'est pas lié à OMR :

- un VLAN (192.168.10.x/24) avec DHCP, pour tes TV/BOX TV/machin truc TV.
- un VLAN (192.168.20.x/24) avec DHCP, pour internet sur tes telephones/ordinateurs fixes/ordinateurs portables.
- un VLAN (192.168.30.x/24) avec DHCP, pour ta domotique/alarme/vidéosurveillance.

Tu configure SQM pour tes WAN évidemment, puis, tu configure SQM pour chacun de tes VLAN et tu mes le débit que tu souhaite leur attribuer ! soit 2Mbps pour le VLAN TV, et "admettons" 5Mbps pour le VLAN domotique et 30Mbps pour le VLAN informatiques.

De ce fait : 
1°) tu limite les paquets broadcast à chaque sous réseau (qui doivent être à un nombre affreusement élevé et qui te créer un goulot d'étranglement dans ton réseau).
2°) tu évite que les trucs domotiques chinois puisse avoir accès à tes ordinateurs et téléphones.
3°) tu isole mieux ton réseau...
...

De rien !

JeedHome44, 2021-04-25:
Désolé mais la tu me perds complètement.

Hani fixe les IP de tous les appareils sur 192.168.100.X
Ce n'est pas pour tout refaire!

De plus, tu parles de plusieurs DHCP distincts alors qu'Ysarc m'a dit à plusieurs reprises qu'il ne faut qu'un seul DHCP d'activé.

La tu me parles de mettre un seul point Wifi mais les objets dans les extrémités de captent pas bien donc c'est le but d'un répéteur d'étendre le wifi...

Mon réseau n'est pas si complexe que ça...

Donc la, vu ton explication, il faudrait donc un Routeur par VLAN avec le DHCP d'activé...

Tout le contraire de ce qu'on me dit depuis le début...

Et tu dis que je suis hors sujet.

Non, je demande une fonction qui permette de limiter le débit comme le fait le byPass qui oriente l'appareil vers un seul WAN. Ça serait juste intéressant d'avoir dans ce byPass une option supplémentaire permettant de limiter à X Mbps le débit du byPass de l'appareil ou de l'IP en question.

La c'est bien trop complexe ce que tu me demandes de faire.

Et tu me dis que j'ai trop d'appareils... c'est interdit d'avoir une maison domotisée? 😄

J'ai l'impression que tu as quelque chose contre moi...

C'est juste que je n'ai pas compris pourquoi j'avais des problèmes ces derniers mois.

C'est juste que TP Link, quand on met leurs appareils en mode AP, mettent le DHCP sur auto au lieu de la désactivé.

J'ai découvert ça récemment et depuis, tout est rentré dans l'ordre sur mon LAN et j'ai un bon Ping.

J'ai juste un Switch aux fesses de la Freebox pour aller vers le Pi, le Pi Octoprint, le thermostat Qivivo... et un Switch plus loin pour mettre Pi Jeedom, Pi Home assistant.
Ces 2 switch sont sur la Freebox en direct donc je ne vois pas ce qu'il y a de mal.

Je vais bientôt recevoir un Xiaomi AX3600 pour voir s'il couvre toute la maison et peut être passer à 2 SSID uniquement mais j'y crois pas trop avec les contrôleurs Magic Home dehors qui sont très capricieux.

Et perso, je n'ai pas envi de faire un VLAN par appareil. Car c'est seulement 3-4 que je veux limiter avec des limites différentes. Je pensais donc qu'il y avait quelque chose de simple de prévu dans OMR.

Vu que tu me dis non, tant pis, je resterai comme ça pour l'instant. Ça arrivera peut être un jour 😄

Ysurac, 2021-04-25:
Qhelque chose comme "luci-app-nft-qos" peut peut-être marcher, mais je ne sais pas si ça ne pose pas de soucis avec OpenMPTCProuter, je n'ai jamais essayé ni regardé.

JeedHome44, 2021-04-25:
Je viens de demander sur un groupe Domotique Facebook et il me disent aussi QoS.

Je vois l'onglet QoS sur OMR avec SQM mais il n'y a rien qui parle de QoS après.

Ce n'est pas activé c'est ça?

Ysurac, 2021-04-25:
Ce n'est pas installé. Il faut voir dans Système->Logiciels.

JeedHome44, 2021-04-25:
Vu que je viens enfin de retrouver un OMR stable, et vu que tu me dis ne jamais avoir testé, ça me fait un peu peur de faire quelque chose qui n'a pas été testé lol.
QoS permet de limiter un seul IP c'est ça?

JeedHome44, 2021-04-25:
Je me suis renseigné un peu plus sur QoS et ça a l'air vraiment d'être ce dont j'ai besoin. En plus, ça permettrait aussi de prioritiser la sonnette video wifi par exemple, les smartphones... ca serait un très bon complément dans OMR.

Y a-t-il un risque que je le teste pour OMR ou est-ce déconseillé?

J'ai été voir dans Système > Logiciel et ça ne me met rien dans la liste.

JeedHome44, 2021-04-25:
Je n'avais pas mis à jour les listes.
J'ai donc fait un backup pour être sur et j'ai installé le packet "luci-app-nft-qos"

Par contre, je ne le vois pas dans le menu SQM QoS.
Y a-t-il quelque chose de particulier à faire?

Merci

JeedHome44, 2021-05-06:
Bonjour,

Je pense avoir réussi à mettre QoS sur OMR.

[image]

Par contre, certains menus, si on les active, font bug le menu QoS si on veut revenir dessus.

Et si je coche la case pour activer la priorité, ça fait aussi buguer le menu QoS.

Est-ce possible de résoudre ce problème dans la prochaine mise à jour?

Pour l'instant, j'ai juste mis mes 2 Mi Box et ma TV du salon dans ce menu en mettant 1 Mbps pour chacun.

Maintenant, je peux consulter internet même si Netflix est en cours sur l'un de ces 3 appareils.

C'était donc bien QoS qu'il me fallait.

Par contre, j'aimerai bien maintenant, établir des priorités QoS.

Par exemple, mettre en priorité les smartphones, la sonnette wifi...

Je pense que c'est bugué pour le moment.

closed 2021-08-11 by github-actions[bot]