Ysurac/openmptcprouter #1781: Vpn ne repond pas

montana013, 2021-04-25:
Bonjour 

J'ai une installation avec une freebox mini 4K et un routeur 4g-5g  huawei CPE pro 1 sous freemobile.
Le tout avec raspberry 3b+ et un vps ovh.
Cela fait deux jours que j'essaye tout,hier j'ai réussi à tout faire marcher par miracle (et encore que dans un sens eth0 freemobile et eth1 freebox dans l'autre sens impossible) mais depuis aujourd'hui plus rien.
J'essaye du coup juste avec la box 4g sans la freebox . Bien sur le raspberry connecté dessus mais dès que je le branche j'ai plus internet et impossible d'accéder à l'interface 192.168.100.1 et  quand je me connecte soit il me dit vpn ne repond pas ou Dns impossible de résoudre le domaine.
Le dhcp est désactivé sur la BOX 4G , quand je passe part un switch des que je branche le raspberry j'ai plus internet.

Merci

Ysurac, 2021-04-25:
Il faudrait utiliser le template.
Ensuite j'ai pas tout compris avec eth0 eth1, car où est le lan ? Et il y a du Ethernet over USB ?

montana013, 2021-04-25:
Bonjour,
J'ai deux ports sur mon routeur 4G  un port WAN et un port LAN .
J'ai connecté le port WAN  sur l'entré 1 du switch , sur l'entré deux du switch j'ai mis le raspberry , le port 3 le PC .
[image]

[image]

[image]<br /><a target='_blank' href='https://usefulwebtool.com/fr/clavier-georgien'>traduction francais georgien gratuit</a><br />

Kalimeiro, 2021-04-25:
il faut brancher le câble Ethernet sur le port LAN de ta BOX 4G.

le port WAN sert pour mettre une connexion ADSL pour faire du failover, c'est à dire en cas de panne de l'ADSL la 4G prend le relai automatiquement.

La tu souhaite utiliser la connexion 4G avec OMR et le VPS OVH, donc tu dois brancher le LAN de ta box 4G sur le switch.

Je te conseille de mettre en DHCP ton PC qui n'est pas dans le même sous réseau que le raspberry.

montana013, 2021-04-25:
il y a du mieux , mais pourquoi le raspberry m'affiche toujours les messages d'erreurs VPN et SHADOWSocks ? 
J'ose pas redemarrer ou éteindre le raspebrry j'ai plusieurs messages "read only" j'ai du re installer le soft.
[image]

Ysurac, 2021-04-25:
La clef du VPS est bien configuré dans Système->OpenMPTCProuter ?
Si oui, le VPS a été redémarré et le noyau est bien 5.4.86-mptcp (```uname -a``` sur le VPS pour vérifier)

montana013, 2021-04-25:
c'est un mystere ca marche , pendant 20s  le message "VPN ne réponds pas" et d'un coup tout est vert !Merci infiniment de ton aide ! par contre quand j'avais atteint cette etape hier mon objectifs finale c'est de steamer mon PC à l’extérieur via Moonlight streaming (dispo sur github) ce que je ne peux pas faire en ADSL avec 1mega en UP.
Hier quand j'avais réussi (j'avais mis en plus le WAN 2 la freebox)  a me connecter à distance à mon pc mais j'avais pas du tout les débits attendue en 4G j,'étais resté étrangement sur mon debits adsl. donc cetait trés trés lent.
SI tu as une idée je suis preneurs.

[image]

montana013, 2021-04-26:
Après d'autres soucis ça remarche ,tu sais si c'est possible que j'ouvre les ports ci-dessous pour moonlight streaming  ils sont nécessaires  de base les ports sont bloque par free mobile mais j'imagine le passage par le vps et ton outil permet de contourner cette limitation ? Si oui comment merci par avance 
[image]

Ysurac, 2021-04-26:
C'est dans la doc: https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding

montana013, 2021-04-27:
Bonjour  J'ai réussi mais j'ai une latence énorme pourtant je ne fait d'agrégation de réseau, je suis juste connecté au routeur 4g.
J'ai raspberry 3b+ est ce que cela peux être causé par sa vitesse de traitement ? Il y a une solution pour réduire la latence ? Merci

Ysurac, 2021-04-27:
Il faut regarder la charge du routeur dans la page Système->OpenMPTCProuter, "Etat".
La latence entre la connexion et le VPS est également affichée dans cette page.

montana013, 2021-04-27:
J'ai entre 35 et 48ms en sachant que quand je faisais du streaming de ma ligne ftth  vers la 4G j'avais de bon résultats.

[image]

montana013, 2021-04-27:
Quand je passe vpn glorytun UDP au lieu de TCP il y a du mieux je suppose que le vpn + raspberry + la 4g des deux côtés  c'est trop en demander niveau latence.
Dommage niveau débit je suis bon  j'imagine pas de solution pour améliorer la chose ?

[image]
[image]<br /><a target='_blank' href='https://fr.imgbb.com/'>prendre une photo avec ma cam en ligne</a><br />

closed 2021-08-02 by github-actions[bot]